INTI-CMNB/KiBot #680: [HOW] Avoid (W043) Missing component

oliv3r, 2024-10-04:
Currently my [pipeline](https://gitlab.com/olliver/ochime/-/jobs/7955124909#L48) is triggering
```
WARNING:(W043) Missing component `LCSC:TC5050RGBF08-3CJH-AF53A` (kibot - v6_sch.py:2416)
```

I traced the error to https://github.com/INTI-CMNB/KiBot/blob/c495efb1c834585840a8df8785a5fee7188168ab/kibot/kicad/v6_sch.py#L2420 but don't quite understand what the problem is.

Locally it's fine, the entire kibot pipeline runs fine (albeit with that warning). Why is it complaining? I think this is one of those 'ignore during CI' so that's good; but still would like to know if I can reasonably solve this warning. I've excluded 'missing library items' already from kicad itself, which is why the ERC isn't complaining. So what causes this and how would I resolve this, without putting this specific component into a component library and make that available to kibot/the docker image?

Is this a kicad6 warning that's irrelevant for 7 or 8? If so, maybe hide the warning on kicad 8 projects?

set-soft, 2024-10-05:
> Locally it's fine, the entire kibot pipeline runs fine (albeit with that warning). Why is it complaining?

Is the LCSC lib available in your repo? Doing a clone and switching to *change/upgrade_ochime* I don't even see the lib tables.

This warning is informing that KiBot couldn't find a component named TC5050RGBF08-3CJH-AF53A from a lib named LCSC

oliv3r, 2024-10-09:
No, the library is not in the repo, the symbol and footprint exist only in the project specific kicad files. I thought this was a 'kicad feature', where all symbols and footprints are always copied into the project. ERC/DRC are set to 'ignore' library issues for exactly this purpose. I'm "ok" with this behavior, these are LCSC generates parts, that I don't want to keep a library of anyway.

So why is kibot producing the error, when kicad is honoring 'missing libraries'.

Note, that I am still on 1.7.0

set-soft, 2024-10-09:
> No, the library is not in the repo, the symbol and footprint exist only in the project specific kicad files. I thought this was a 'kicad feature', where all symbols and footprints are always copied [...]

Then ignore the warning. As I explained in other thread: KiBot doesn't know if you did it on purpose or not.
BTW: providing the library is more convenient for derivative work.

> So why is kibot producing the error, when kicad is honoring 'missing libraries'.

Warnings are not errors. KiBot can't know if you are lazy or forgot something.
 
> BTW, while good for me, the warning is not triggering an `exit !0` so the pipieline doesn't even care :p that might be a bug.
> 
> Note, that I am still on 1.7.0

Warnings doesn't stop the flow, unless you explicitly ask for it

closed 2024-10-09 by set-soft