INTI-CMNB/KiBot #682: [BUG] Strange copper (un)fills happening near mousebites in PDF generation

oliv3r, 2024-10-09:
So this is a bit of a complex situation I agree. I have generated my boards and panels via CI pipelines, that I'm very happy with.

However, there is no sane way to take individual boards, and stick them together with mousebites for FAB houses. These boards are completely unrelated, so I don't want to stick them into the same folder, or make it a kicad variant. So I end up having to make a pcb manually, copy/pasting the (panelize) outputs. This is cumbersome but ok.

Given however the following piece
[image]
we see from left to right, pin 8, a trace, and a copper fill, then the mouse bite, followed by another copper fill on the other side. 

Next, exporting this via the JLCPCB plugin I get the following gerber for F.Cu
[image]
Everything is fine.

Now, when generating the PDF however, using the recommended kibot way, I get the following.
[image]
Here we see the copper fill removed, a new copper fill added in the mousebite shape on both sides.

Is kibot re-filling the boards somehow during PDF printing?

Attached are the files in question.

[unfill.tar.gz](https://github.com/user-attachments/files/17304580/unfill.tar.gz)

set-soft, 2024-10-09:
Your files doesn't include the KiBot configuration. There are preflights to refill zones, so it might be the case.

In these cases the best is to create a very simple example where the problem is visible. In fact you usually realize the reason while doing it. So please try to create a small example of SCH+PCB+KiBot config where I can see the problem. Otherwise I could waste more than hour trying to guess what you do.

oliv3r, 2024-10-11:
I thought I added everything; but the stupid thing is, if you look at the PDF, the first page the infills are not touched. So it must come from the 'other' layers.

```yaml
kibot:
  version: 1

global:
  environment:
    models_3d: '${KIPRJMOD}/packages3D'
  restore_project: true

preflight:
  set_text_variables:
    - name: 'VCS_PCB_DATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Idate || echo "Unknown"'
    - name: 'VCS_PCB_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Iseconds || echo "Unknown"'
    - name: 'VCS_PCB_HASH'
      command: 'echo $(git log -1 --format="%H" "${KIBOT_PCB_NAME}" || echo "Unknown")'
      before: 'Git <'
      after: '>'
    - name: 'VCS_SCH_TIMEDATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_SCH_NAME:-.}") -Iseconds || echo "Unknown"'
    - name: 'VCS_SCH_HASH'
      command: 'echo $(git log -1 --format="%H" "${KIBOT_SCH_NAME}" || echo "Unknown")'
      before: 'Git <'
      after: '>'
    - name: 'VCS_TAG'
      command: 'echo $(git describe --always --dirty || echo "Unknown")'
    - name: 'SHEETPATH' # Hack to avoid W082 Unknown sheetpath
      text: ''

outputs:
  - name: 'pdf_sch_print'
    run_by_default: @RUN_BY_DEFAULT@
    comment: 'Exports the schematic to the most common exchange format. Suitable for printing.'
    type: 'pdf_sch_print'
    dir: 'outputs/pdf'
    options:
      all_pages: true
      background_color: false
      frame: true
      monochrome: false
  - name: 'pdf_pcb_print'
    run_by_default: @RUN_BY_DEFAULT@
    comment: 'Exports the PCB to the most common exchange format. Suitable for printing.'
    type: 'pcb_print'
    dir: 'outputs/pdf'
    options:
      force_edge_cuts: true
      forced_edge_cuts_use_for_center: false
      invert_use_for_center: true
      format: 'PDF'
      individual_page_scaling: false
      keep_temporal_files: false
      scaling: @KICI_PDF_SCALING@
      pages:
        - sheet: 'PCB top'
          layers:
          - layer: 'F.Cu'
          - layer: 'F.Mask'
            color: '@KICI_PDF_COLOR_PCB_MASK@'
          - layer: 'F.SilkS'
            color: '@KICI_PDF_COLOR_PCB_SILK@'
          - layer: 'User.Comments'
        - sheet: 'Fabrication top'
          layers:
          - layer: 'F.Fab'
          - layer: 'User.Comments'
        - sheet: 'Silkscreen top'
          layers:
          - layer: 'F.SilkS'
        - sheet: 'F.Cu'
          layers:
          - layer: 'F.Cu'
        - sheet: '%ln layer'
          repeat_for_layer: 'F.Cu'
          repeat_layers: 'inners'
          layers:
          - layer: 'F.Cu'
        - sheet: 'B.Cu'
          layers:
          - layer: 'B.Cu'
        - sheet: 'Silkscreen bottom'
          layers:
          - layer: 'B.SilkS'
          mirror: true
        - sheet: 'Fabrication bottom'
          layers:
          - layer: 'B.Fab'
          - layer: 'User.Comments'
          mirror: true
        - sheet: 'PCB bottom'
          layers:
          - layer: 'B.Cu'
          - layer: 'B.Mask'
            color: '@KICI_PDF_COLOR_PCB_MASK@'
          - layer: 'B.SilkS'
            color: '@KICI_PDF_COLOR_PCB_SILK@'
          - layer: 'User.Comments'
          mirror: true
      plot_sheet_reference: true

...
definitions:
  RUN_BY_DEFAULT: true
  KICI_PDF_SCALING: 1.5
  KICI_PDF_COLOR_PCB_MASK: '#14332440'

outputs:
  - name: 'pdf_sch_print'
    run_by_default: @RUN_BY_DEFAULT@
    comment: 'Exports the schematic to the most common exchange format. Suitable for printing.'
    type: 'pdf_sch_print'
    dir: 'outputs/pdf'
    options:
      all_pages: true
      background_color: false
      frame: true
      monochrome: false
  - name: 'pdf_pcb_print'
    run_by_default: @RUN_BY_DEFAULT@
    comment: 'Exports the PCB to the most common exchange format. Suitable for printing.'
    type: 'pcb_print'
    dir: 'outputs/pdf'
    options:
      force_edge_cuts: true
      forced_edge_cuts_use_for_center: false
      invert_use_for_center: true
      format: 'PDF'
      individual_page_scaling: false
      keep_temporal_files: false
      scaling: @KICI_PDF_SCALING@
      pages:
        - sheet: 'PCB top'
          layers:
          - layer: 'F.Cu'
          - layer: 'F.Mask'
            color: '@KICI_PDF_COLOR_PCB_MASK@'
          - layer: 'F.SilkS'
            color: '@KICI_PDF_COLOR_PCB_SILK@'
          - layer: 'User.Comments'
        - sheet: 'Fabrication top'
          layers:
          - layer: 'F.Fab'
          - layer: 'User.Comments'
        - sheet: 'Silkscreen top'
          layers:
          - layer: 'F.SilkS'
        - sheet: 'F.Cu'
          layers:
          - layer: 'F.Cu'
        - sheet: '%ln layer'
          repeat_for_layer: 'F.Cu'
          repeat_layers: 'inners'
          layers:
          - layer: 'F.Cu'
        - sheet: 'B.Cu'
          layers:
          - layer: 'B.Cu'
        - sheet: 'Silkscreen bottom'
          layers:
          - layer: 'B.SilkS'
          mirror: true
        - sheet: 'Fabrication bottom'
          layers:
          - layer: 'B.Fab'
          - layer: 'User.Comments'
          mirror: true
        - sheet: 'PCB bottom'
          layers:
          - layer: 'B.Cu'
          - layer: 'B.Mask'
            color: '@KICI_PDF_COLOR_PCB_MASK@'
          - layer: 'B.SilkS'
            color: '@KICI_PDF_COLOR_PCB_SILK@'
          - layer: 'User.Comments'
          mirror: true
      plot_sheet_reference: true

...
definitions:
  RUN_BY_DEFAULT: true
  KICI_PDF_SCALING: 1.5
  KICI_PDF_COLOR_PCB_MASK: '#14332440'
```

set-soft, 2024-10-12:
Please provide:

1. A valid configuration file
2. How to invoke KiBot

This one isn't valid, and needs an schematic, which wasn't provided

set-soft, 2024-10-14:
Ok, you should disable the `colored_pads` and `colored_vias` options.
Now is documented.

closed 2024-10-14 by set-soft

oliv3r, 2024-10-18:
I'll go read the documentation, but how are colored pads/via's triggering a refill? Wouldn't we want an option instead 'never refill' or something?

set-soft, 2024-10-18:
Lamentably KiCad's plot API doesn't have a mechanism to generate "colored" items.
So we must do it manually.
I can't find a mechanism to just draw the pads/vias without drawing the rest of the things, zones included.
So what we are currently doing is moving away all but the pads/vias, plot the layer and then put the stuff again.
With zones the current approach (and I tried another approach that has problems) is to unfill them, plot and fill them again.
If you know a better approach let me know.
So, yes, ideally we should avoid it, is just that it looks like this is impossible.
KiCad's Python API is very limited, full of bugs and changes all the time. Ideally it shouldn't be like this, but this is what we have.

oliv3r, 2024-10-18:
I guess we should keep this as an open big then until we can avoid redraws. Ideally we'd be able to rely on kicad to generate the pdfs better nativly throigh the API :(

commit f9abab27 (2024-10-14): [DOCs][Added] Note about `colored_pads` and `colored_vias` side effects

Closes #682