INTI-CMNB/KiBot #684: [BUG] panelize process error

Uvis-Acurable, 2024-10-09:
When running panelize to generate the panel for the board it returns error: "An error occurred: argument 3: TypeError: wrong type", that essentially comes from subprocess from kiplot.py, however I can't see anything wrong with the arguments.

This used to work, but there have been changes to multiple things, OS upgrades docker updates, different version of KiBot

**Environment:
- Mac M2 Pro, Sonoma 14.4
- Docker v4.34.2 (updated from 4.32)
- KiCad v8.0.4
- Tried 2 images: kicad8_auto:1.8.1-1_k8.0.4_d12.1 and kicad8_auto:1.7.0

Debug output:

`DEBUG:{
    "page": {
        "type": "A0"
    },
    "layout": {
        "hspace": "2mm",
        "vspace": "2mm",
        "rows": 5,
        "cols": 5,
        "vbackbone": "2mm",
        "hbackbone": "2mm",
        "vbonecut": false,
        "hbonecut": false
    },
    "tabs": {
        "type": "annotation"
    },
    "cuts": {
        "type": "mousebites",
        "offset": "-0.2mm"
    },
    "framing": {
        "type": "tightframe",
        "width": "8mm"
    },
    "tooling": {
        "type": "4hole",
        "hoffset": "5.0mm",
        "voffset": "5.0mm",
        "size": "3.65mm"
    },
    "fiducials": {
        "type": "3fid",
        "hoffset": "14.5mm",
        "voffset": "3.5mm",
        "opening": "2.5mm"
    },
    "text": {
        "type": "simple",
        "text": "${boardTitle} ${boardRevision}",
        "voffset": "3.5mm",
        "width": "2.0mm",
        "height": "2.8mm",
        "thickness": "0.4mm",
        "layer": "F.SilkS"
    },
    "text2": {
        "type": "simple",
        "text": "This side up for fixture interface",
        "voffset": "3.5mm",
        "width": "2.0mm",
        "height": "2.8mm",
        "thickness": "0.4mm",
        "layer": "B.SilkS"
    },
    "copperfill": {
        "type": "hatched",
        "clearance": "2mm",
        "width": "0.5mm",
        "spacing": "0.5mm"
    },
    "post": {
        "millradius": "1mm",
        "refillzones": true
    },
    "source": {
        "tolerance": "15mm"
    }
} (kibot - out_panelize.py:766)
DEBUG:Writing panel config to /tmp/kibot_panel_cfgvmkaz8tb.json (kibot - gs.py:982)
DEBUG:- Executing: /usr/bin/kikit panelize --preset /tmp/kibot_panel_cfgvmkaz8tb.json /workdir/project/kibot_f1hahgxt.kicad_pcb /workdir/outputs/P02/Panel/acupebble-oximeter-board_P02-panel.kicad_pcb (kibot - kiplot.py:154)
ERROR:Trace stack: (kibot.gs - gs.py:831)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 164, in run_command
    res = _run_command(command, change_to)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 150, in _run_command
    return run(command, check=True, stdout=PIPE, stderr=STDOUT, cwd=change_to)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
ERROR:Running ['/usr/bin/kikit', 'panelize', '--preset', '/tmp/kibot_panel_cfgvmkaz8tb.json', '/workdir/project/kibot_f1hahgxt.kicad_pcb', '/workdir/outputs/P02/Panel/acupebble-oximeter-board_P02-panel.kicad_pcb'] returned 1 (kibot.gs - gs.py:848)
DEBUG:- Output from command: An error occurred: argument 3: TypeError: wrong type
No output files produced
 (kibot.gs - gs.py:851)
WARNING:(W139) Keeping temporal files: ['/workdir/project/kibot_f1hahgxt.kicad_pcb', '/workdir/project/kibot_f1hahgxt.kicad_pro', '/workdir/project/kibot_f1hahgxt.kicad_pro-bak', '/workdir/project/kibot_f1hahgxt.kicad_prl', '/workdir/project/kibot_f1hahgxt.kicad_prl-bak', '/workdir/project/kibot_f1hahgxt.kicad_dru', '/workdir/project/kibot_f1hahgxt.kicad_dru-bak', '/tmp/kibot_panel_cfgvmkaz8tb.json'] (kibot - out_panelize.py:821)`

set-soft, 2024-10-09:
> DEBUG:- Output from command: An error occurred: argument 3: TypeError: wrong type

Note that this is an error from KiKit. When you find an issue related to KiKit you should add:

```json
    "debug": {
        "trace": true
    }
```

To your KiKit config. For some reason that I don't understand this isn't the default. Once you do it you'll find the real place where the problem is:

```
An error occurred: argument 3: TypeError: wrong type
No output files produced
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui.py", line 217, in panelize
    doPanelization(input, output, preset, plugin)
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui.py", line 279, in doPanelization
    frameCuts = ki.buildFraming(preset, panel)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui_impl.py", line 485, in buildFraming
    panel.makeTightFrame(framingPreset["width"], framingPreset["slotwidth"],
  File "/usr/lib/python3/dist-packages/kikit/panelize.py", line 1430, in makeTightFrame
    boardSlot = boardSlot.buffer(slotwidth, join_style="mitre")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/shapely/geometry/base.py", line 632, in buffer
    self._lgeos.GEOSBufferParams_setJoinStyle(params, join_style)
ctypes.ArgumentError: argument 3: TypeError: wrong type
```

So this seems to be a compatibility issue with the Shapely installed in the docker image, triggered by the use of the *tightframe*. I'll try to find the exact solution.

set-soft, 2024-10-09:
I don't see a simple solution, you could try to just use *pip* to install shapely.
Lamentably this is a very problematic package and the feature you want to use needs a version that isn't available for Debian stable.

set-soft, 2024-10-09:
Hi @Uvis-Acurable !

I added a really ugly installation of Shapely to the current images, please try pulling a fresh *ghcr.io/inti-cmnb/kicad_auto_full:dev_k8* and tell me if it works for your PCB.

Uvis-Acurable, 2024-10-10:
Hi @set-soft I will try this and let you know

Uvis-Acurable, 2024-10-10:
I am still getting the same error. Now when I enabled the trace I am getting the same output as you

set-soft, 2024-10-10:
Are you sure you pulled a fresh docker image of  *ghcr.io/inti-cmnb/kicad_auto_full:dev_k8*?
Try running this in the docker image:
```
python3 -c "import shapely; print(shapely.__version__)"
```
You should get 2.0.6, if you get 1.8.5 then this is not a fresh image.

Uvis-Acurable, 2024-10-10:
It reports V1.8.5 However the image I pulled is: ghcr.io/inti-cmnb/kicad_auto_full:dev_k8_1.8.2-b077362_k8.0.4_d_sid_b3.5.1

set-soft, 2024-10-10:
Ok, try it again, should be *dev_1.8.2-ed9644c_k8.0.4_d_sid_b3.5.1* or newer

Uvis-Acurable, 2024-10-10:
Still Getting the same:

```
% docker run --rm -it -v$(pwd):/workdir -entrypoint=/bin/bash --platform linux/amd64 ghcr.io/inti-cmnb/kicad_auto_full:dev_k6_1.8.2-ed9644c_k6.0.11_d12.1_b3.5.1
root@0ebcb4546285:/# python3 -c "import shapely; print(shapely.__version__)"
1.8.5
root@0ebcb4546285:/#
```

set-soft, 2024-10-10:
You are pulling a KiCad 6 docker image, KiKit doesn't support KiCad 6.
The KiCad 7 and 8 got Shapely 2.0.6.

```
$ docker pull ghcr.io/inti-cmnb/kicad8_auto_full:dev
dev: Pulling from inti-cmnb/kicad8_auto_full
648e0aadf75a: Already exists 
ec5923c9c2d5: Already exists 
289f6374749a: Already exists 
46d2a18d0c55: Already exists 
7b4d33e9c286: Already exists 
89c65cb2b67a: Already exists 
17fb47227644: Already exists 
e8f5fd8b671f: Already exists 
bd22edcf394a: Already exists 
afc4534271ec: Already exists 
5bb8019470fe: Already exists 
031b0fdd9ca2: Already exists 
0c8a3f87afd7: Already exists 
b79cc9037cfa: Already exists 
cb92b349c0ef: Already exists 
c7f77b2472db: Already exists 
c2b4aa001c4b: Already exists 
2d5e02407158: Pull complete 
abb685cc4521: Pull complete 
Digest: sha256:f17a96c2078b689bd3032ead26e311a365da4979a48a2af6fdd455a5c7829771
Status: Downloaded newer image for ghcr.io/inti-cmnb/kicad8_auto_full:dev
ghcr.io/inti-cmnb/kicad8_auto_full:dev
$ docker run --rm -it ghcr.io/inti-cmnb/kicad8_auto_full:dev
root@7b09499011fd:/# python3 -c "import shapely; print(shapely.__version__)"
2.0.6
root@7b09499011fd:/# 
```