INTI-CMNB/KiBot #685: [BUG] Simple and Full Reports are lacking physical units for Thickness, Epsilor_r and Loss tangent

IvanVnucec, 2024-10-11:
**Describe the bug**
Simple and Full Reports are lacking physical units for Thickness, Epsilor_r and Loss tangent under the Stackup header. Without them, the numbers are meaningful.

**To Reproduce**
`kibot --quick-start --out-dir generated`

**Expected behavior**
Thickness, Epsilor_r and Loss tangent columns have physical units.

**Screenshots**
At my project, `report_simple.txt` generated the following:
```txt
Stackup:

Impedance controlled: YES

| Name                 | Type                 | Color            | Thickness | Material        | Epsilon_r | Loss tangent |
|----------------------|----------------------|------------------|-----------|-----------------|-----------|--------------|
| F.SilkS              | Top Silk Screen      | Green            |           |                 |           |              |
| F.Paste              | Top Solder Paste     |                  |           |                 |           |              |
| F.Mask               | Top Solder Mask      |                  |        10 |                 |           |              |
| F.Cu                 | copper               |                  |        35 |                 |           |              |
| dielectric 1         | prepreg              |                  |        99 | FR4             |       4.5 |        0.020 |
| In1.Cu               | copper               |                  |        15 |                 |           |              |
| dielectric 2         | core                 |                  |      1265 | FR4             |       4.5 |        0.020 |
| In2.Cu               | copper               |                  |        15 |                 |           |              |
| dielectric 3         | prepreg              |                  |        99 | FR4             |       4.5 |        0.020 |
| B.Cu                 | copper               |                  |        35 |                 |           |              |
| B.Mask               | Bottom Solder Mask   |                  |        10 |                 |           |              |
| B.Paste              | Bottom Solder Paste  |                  |           |                 |           |              |
| B.SilkS              | Bottom Silk Screen   | Green            |           |                 |           |              |
```

**Environment (please complete the following information):**
Where are you running KiBot:
I'm running KiBot in GitHub Actions CI/CD with `ghcr.io/inti-cmnb/kicad8_auto_full:latest` container.

set-soft, 2024-10-11:
Hi @IvanVnucec !
I fail to see where is the bug, I guess this is feature request, please confirm it.
The units are the same used by KiCad, εᵣ is a relative value, so it doesn't have physical units.
In the case of tan δ we are in a similar case, this is the tangent of an angle, just a number, no physical units.
Perhaps you want to see the physical units for the thickness, as I said before they are the same used by KiCad, this is µm.
Also note that you can provide a template for the report adding anything you want.

Resuming: Are you asking to add " [µm]" after the "Thickness" word?

IvanVnucec, 2024-10-11:
> Hi @IvanVnucec ! I fail to see where is the bug, I guess this is feature request, please confirm it. The units are the same used by KiCad, εᵣ is a relative value, so it doesn't have physical units. [...]

Yes I would like to add [µm] after the "Thickness" word.
My apologies for εᵣ and tan δ as they have no physical units.

set-soft, 2024-10-14:
The above patch adds the units. Try using *dev* images.

commit 5c33f8f8 (2024-10-14): [Report][Added] Thickness units to the templates

Closes #685
closed 2024-10-14 by set-soft