INTI-CMNB/KiBot #688: [BUG] `Unable to expand KICAD8_3DMODEL_DIR` error for KiCad standard library part

IvanVnucec, 2024-10-14:
**Describe the bug**
KiBot GitHub Actions fails with the following log:
```shell
 ERROR:Unable to expand `KICAD8_3DMODEL_DIR` in `${KICAD8_3DMODEL_DIR}/Capacitor_SMD.3dshapes/C_0603_1608Metric.wrl` (kibot - log.py:179)
WARNING:(W098)  5 3D models downloaded or cached (kibot - out_base_3d.py:527)
[...]
ERROR:Unable to generate `/__w/project_name/project_name/generated/ExportedProject/3d_models/Capacitor_SMD.3dshapes/C_0603_1608Metric.step` from 'Project with all needed files' (_ExportProject) [copy_files] (kibot.gs - gs.py:845)
```

**To Reproduce**
My GitHub Actions `.github/workflows/hardware.yml` is as follows:
```yml
name: "hardware"

on:
  push:
  repository_dispatch:
    types: [run_qs]

jobs:
  kibot:
    runs-on: ubuntu-latest
    container: ghcr.io/inti-cmnb/kicad8_auto_full:latest
    steps:
    - uses: actions/checkout@v4
      with:
        # So we can run a diff between last 2 changes
        fetch-depth: '0'

    - name: Generate PCB outputs
      run: |
        kibot --out-dir generated

    - name: Retrieve results
      uses: actions/upload-artifact@v4
      with:
        name: Automatic_outputs
        path: project_name/generated
```

**Environment (please complete the following information):**
I'm running KiBot on GitHub Actions with `ghcr.io/inti-cmnb/kicad8_auto_full:latest` container.

**Additional context**
I've been using KiBot quick start feature by running `kibot --quick-start --out-dir generated` in Actions and the problem with 3D parts path didn't occur. Then I've committed `kibot-generated.kibot.yaml` file and ran KiBot as I've described and the problem started.

set-soft, 2024-10-14:
You forgot to post the configuration that triggers it, so I can only guess.

Try using *dev* images, looks related to an already fixed issue.

Also note that when using *--quick-start* KiBot runs like when using *--dont-stop*, and this might be the difference.

IvanVnucec, 2024-10-14:
Yes, `--dont-stop` is the cause of the difference.
I will try with the KiBot dev images.

My KiBot configuration file is as follows:

```yml
# This is a working example.
# For a more complete reference use `--example`
kibot:
  version: 1

global:
  filters:
  - number: 1007
  - number: 1015
  - number: 58
  - number: 107

import:
- file: Elecrow
  definitions:
    _KIBOT_MANF_DIR: Manufacturers/Elecrow
- file: ExportProject
- file: FusionPCB
  definitions:
    _KIBOT_MANF_DIR: Manufacturers/FusionPCB
- file: JLCPCB
  definitions:
    _KIBOT_MANF_DIR: Manufacturers/JLCPCB
    _KIBOT_POS_PRE_TRANSFORM: '[''_kicost_rename'', ''_rot_footprint'']'
    _KIBOT_BOM_ENABLED: 'false'
- file: MacroFab_XYRS
  definitions:
    _KIBOT_MPN_FIELD: ''
- file: PCBWay
  definitions:
    _KIBOT_MANF_DIR: Manufacturers/PCBWay
- file: PanelDemo_4x4

preflight:
  drc:
    dont_stop: true
    format: HTML,RPT,JSON,CSV
  erc:
    dont_stop: true
    format: HTML,RPT,JSON,CSV

outputs:
- name: basic_boardview
  comment: Board View export
  type: boardview
  dir: Assembly
- name: generic_bom_html
  comment: Generic Bill of Materials in HTML format
  type: bom
  dir: BoM/Generic
  options:
    format: HTML
    count_smd_tht: true
- name: generic_bom_csv
  comment: Generic Bill of Materials in CSV format
  type: bom
  dir: BoM/Generic
  options:
    format: CSV
    count_smd_tht: true
- name: generic_bom_hrtxt
  comment: Generic Bill of Materials in HRTXT format
  type: bom
  dir: BoM/Generic
[... 1168 lines cut ...]
  comment: Schematic in SVG format
  type: svg_sch_print
  dir: Schematic
- name: basic_vrml
  comment: PCB in VRML format
  type: vrml
  dir: 3D

...
definitions:
  _KIBOT_MANF_DIR_COMP: Manufacturers
  _KIBOT_GERBER_LAYERS: "- layer: 'F.Cu'\n        suffix: 'F_Cu'\n        description:\
    \ 'Front copper'\n      - layer: 'In1.Cu'\n        suffix: 'In1_Cu'\n        description:\
    \ 'Inner layer 1'\n      - layer: 'In2.Cu'\n        suffix: 'In2_Cu'\n       \
    \ description: 'Inner layer 2'\n      - layer: 'B.Cu'\n        suffix: 'B_Cu'\n\
    \        description: 'Bottom copper'\n      - layer: 'B.Paste'\n        suffix:\
    \ 'B_Paste'\n        description: 'Bottom solder paste'\n      - layer: 'B.Silkscreen'\n\
    \        suffix: 'B_Silkscreen'\n        description: 'Bottom silkscreen (artwork)'\n\
    \      - layer: 'F.Silkscreen'\n        suffix: 'F_Silkscreen'\n        description:\
    \ 'Front silkscreen (artwork)'\n      - layer: 'B.Mask'\n        suffix: 'B_Mask'\n\
    \        description: 'Bottom soldermask (negative)'\n      - layer: 'F.Mask'\n\
    \        suffix: 'F_Mask'\n        description: 'Front soldermask (negative)'\n\
    \      - layer: 'Edge.Cuts'\n        suffix: 'Edge_Cuts'\n        description:\
    \ 'Board shape'"
```

IvanVnucec, 2024-10-14:
@set-soft I've changed the KiBot container to the dev one `ghcr.io/inti-cmnb/kicad8_auto_full:dev` but the problem still persist:
```shell
ERROR:Unable to expand `KICAD8_3DMODEL_DIR` in `${KICAD8_3DMODEL_DIR}/Capacitor_SMD.3dshapes/C_0603_1608Metric.wrl` (kibot - log.py:179)
```
However, the run finishes successfully.

set-soft, 2024-10-14:
This is fine, the docker image doesn't have the 3D models, you didn't define *KICAD8_3DMODEL_DIR* (KiCad neither), and hence KiBot can't even guess its value (no 3D models). 

So KiBot is downloading them and using the downloaded model. The fatal problem was "ERROR:Unable to generate ..." which is already fixed.

closed 2024-10-14 by set-soft