INTI-CMNB/KiBot #689: [BUG]PcbDraw crashes when using 0R resistor

0x007E, 2024-10-14:
When running KiBot with GHA and a layout with THT resistors with a value of 0 ohm kibot fails when pcbdraw is in the pipeline.

[image]

``` yaml
- name: board_top_filled
  comment: "Top layer view with components"
  type: pcbdraw
  dir: './image'
  options:
    format: jpg
    dpi: 300
    show_components: all
```

```
- 'Top layer view with components' (board_top_filled) [pcbdraw]
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.8.1', 'console_scripts', 'kibot')())
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          ....
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/PcbDraw/plot.py", line 890, in _create_component
    self._apply_resistor_code(component_element, id_prefix, ref, value)
  File "/usr/lib/python3/dist-packages/kibot/PcbDraw/plot.py", line 920, in _apply_resistor_code
    power = math.floor(res.log10()) - 1
            ^^^^^^^^^^^^^^^^^^^^^^^
OverflowError: cannot convert Infinity to integer
```
In my opinion, if the value of the resistor is empty, zero or NAN, pcbdraw should ignore the value and leave the resistor blank.

**Environment:**
- Github Actions
- KiBot INTI-CMNB/KiBot@v2_k7

closed 2024-10-14 by set-soft

0x007E, 2024-10-15:
Now there is no error, but also there are no parts on the board after running kibot pcbdraw. Yes rendering is a better option but sometimes pcbdraw seems to be enough. So in the loop (Line 940 `plot.py`)

``` pyton
for res_i, res_c in enumerate(resistor_colors):
```
all other elements are ignored! @set-soft can you please check that again? Sorry about the circumstances!

set-soft, 2024-10-15:
Please create a small example where you see the problem.

I tried (in the repo):

```
src/kibot -c tests/yaml_samples/pcbdraw.kibot.yaml -b tests/board_samples/kicad_8/4Rs.kicad_pcb -d pp
```

And got:

[image]

0x007E, 2024-10-22:
I´ve made a demo project [here](https://github.com/0x007E/kicad). The generated images can be downloaded from [here](https://github.com/0x007E/kicad/releases/download/v1.0/kicad.zip). In the `image` folder you can see the generated images with pcbdraw. There are no components. Maybe i´ve missed something in the [config](https://github.com/0x007E/kicad/blob/main/.kibot/build.kibot.yaml)?

``` yaml
- name: board_top_filled
    comment: "Top layer view with components"
    type: pcbdraw
    dir: './image'
    options:
      format: jpg
      dpi: 300
      show_components: all
```

During the build pipleine i got these messages:

```
Options: -c '.kibot/build.kibot.yaml' -d './output' -b './pcb/demo.kicad_pcb' -e './pcb/demo.kicad_sch'
- Running the ERC
- Running the DRC
- 'Spreadsheet for the Bill of Materials' (bom_xlsx) [bom]
- 'Schematic PDF' (print_sch) [pdf_sch_print]
- 'PCB PDF TOP' (print_pdf_top) [pcb_print]
- 'Gerber Files' (gerbers) [gerber]
- 'Drill files' (drill) [excellon]
- 'Top layer view with components' (board_top_filled) [pcbdraw]
- 'Bottom layer view with components' (board_bottom_filled) [pcbdraw]
- 'STEP model' (mechanical) [step]
ERROR:Unable to expand `KICAD6_3DMODEL_DIR` in `${KICAD6_3DMODEL_DIR}/Capacitor_THT.3dshapes/CP_Radial_D4.0mm_P2.00mm.wrl` (kibot - config.py:114)
WARNING:(W122) Missing tolerance for R3, using 20% (kibot - out_base_3d.py:381)
WARNING:(W122) Missing tolerance for R1, using 20% (kibot - out_base_3d.py:381)
WARNING:(W122) Missing tolerance for R4, using 20% (kibot - out_base_3d.py:381)
WARNING:(W122) Missing tolerance for R2, using 20% (kibot - out_base_3d.py:381)
WARNING:(W098)  5 3D models downloaded or cached (kibot - out_base_3d.py:527)
Found 5 unique warning/s (26 total, 21 filtered)
```

set-soft, 2024-10-23:
In order to understand the problem better you should:

1. Generate just the pcbdraw output (i.e. `-s all board_top_filled`)
2. Log debug information (i.e. `-L logfile`)

set-soft, 2024-10-23:
Ok, try now, the dev images failed to include the PcbDraw footprints (they are in a submodule).
4347501ac7d9d9485ba9ce525cd54a3a195631f1 fixes it, here is a test https://github.com/set-soft/test_689/actions/runs/11481146054

0x007E, 2024-10-23:
Yes, now it seems to work. The parts are now on the board:

[image]

It think it can be merged into the main branch!

set-soft, 2024-10-23:
> It think it can be merged into the main branch! Thanks for the quick response and the testing!

The problem isn't in the code, just in the way that *dev* images were created.

0x007E, 2024-10-23:
Sorry for my misunderstanding. I see you adapted the dockerfiles. So using
``` yaml
uses: INTI-CMNB/KiBot@v2_dk7
```
should now automatically resolve the problems after a restart of the `GHA` pipeline!

commit 91eadbfe (2024-10-14): [PcbDraw][Fixed] Problems with 0 ohms THT resistors

When creating the resitor bands

Fixes #689