INTI-CMNB/KiBot #691: [FEATURE] Set default font for schematic PDF

nguyen-v, 2024-10-15:
Fill as much as you can, remove the rest

**Is your feature request related to a problem? Please describe.**
I have a custom template which uses the default KiCad font for most of the components (useful when working on it, as the default font is much faster to render than other fonts). When printing the schematic to PDF, I like to change the default KiCad font (in the Preferences) to 'Times New Roman' so that the output looks more like Altium. 

**Describe the solution you'd like**
A way to set the default KiCad font to a user-specified one, similar to the color_theme option in the pdf_schematic output. Currently I have my fonts in the kibot_resources/fonts folder in TTF format.

**Describe alternatives you've considered**
An alternative solution is to manually change all of the text fields in my schematic where the default font is used to 'Times New Roman'. This is not very efficient and I use multiple fonts across the schematic (Arial and Times New Roman), so the 'Edit Text and Graphic Properties' tool wouldn't work well. 

Another way to do it manually would be to write a script to add '(face "fontname")' directly in the .kicad_sch files

**Additional context**
Current output:
[image]

Desired output:
[image]

set-soft, 2024-10-15:
Lamentably this isn't currently possible because the `kicad-cli` tool just ignores the default font.

https://gitlab.com/kicad/code/kicad/-/issues/17715

I could use the GUI, like in old times, but maintaining a GUI export just for this looks a major overkill. 
To anybody wanting this feature I'll suggest putting pressure to KiCad maintainers by commenting on the above issue.

nguyen-v, 2024-10-15:
This is a shame, but thanks for having a look at it!

set-soft, 2025-02-21:
Now we have default font support on KiCad 9.
The above patch needs KiAuto from the repo (2.3.4)
I'm closing the issue to clean-up the list

set-soft, 2025-05-15:
Hi @edlongman !

Error 1:
The docs says *Name for the default font. Only for KiCad 9 and newer*. It won't work for KiCad 8. Also: you should upgrade KiAuto. And this is what the 2025-02-21 message above says

> Now we have default font support on KiCad 9.
> The above patch needs KiAuto from the repo (2.3.4)

Error 2:
The docs says this must be a string, so getting *Option `default_font` must be a string not `boolean`* is correct if you use a boolean.

edlongman, 2025-05-16:
Hello,
To clarify I think the problem is that when the `default_font` option is missing, the first error occurs. Or is this version of kibot not meant to work at all with KiCAD 8.0

set-soft, 2025-05-16:
> Hello, To clarify I think the problem is that when the `default_font` option is missing, the first error occurs. 

This is because you have an old KiAuto, which environment are you using to run KiBot?
The error is from `eeschema_do`.
Which KiBot version are you using?
Current KiBot enforces KiAuto 2.3.4 or newer, which supports `--default_font`, and uses it only for KiCad 9 (ignored for olders)

> Or is this version of kibot not meant to work at all with KiCAD 8.0

KiCad 5 is the minimum

edlongman, 2025-05-16:
I'm running on Ubuntu 22 (WSL)

```
Package: kiauto
Version: 2.3.5-1
Priority: optional
```

Working configuration yaml
```
kibot:
  version: 1
global:
  restore_project: true
outputs:
  - name: "Schematic"
    type: pdf_sch_print
    options:
      output: "%f-%i%I%v.%x"
      default_font: ''
    dir: &output_dir "out"
```

Before upgrade of kibot configuration yaml - now doesn't work due to unrecognised arguments error as above.
```
kibot:
  version: 1
global:
  restore_project: true
outputs:
  - name: "Schematic"
    type: pdf_sch_print
    options:
      output: "%f-%i%I%v.%x"
    dir: &output_dir "out"
```

set-soft, 2025-05-16:
Looks like some mess with versions, 2.3.5 supports **--default_font**, so you end running an older version.

You can try two things:
1. Run `kibot-check -p` to make sure it finds 2.3.5
2. Use `-L logfile.txt` and take a look at the log file looking for KiAuto version

edlongman, 2025-05-20:
Thanks. 
Some mix up in versions seems to be happening.
Running `kibot-check -p` gives 
```
KiBot installation checker

Core:
Linux: 4.4.0.19041 (Linux L9020561 4.4.0-19041-Microsoft #4355-Microsoft Thu Apr 12 17:37:00 PST 2024 x86_64 x86_64 x86_64 GNU/Linux)
  /usr/bin/uname
Python: 3.10.12 (main, Feb  4 2025, 14:57:36) [GCC 11.4.0]
  /usr/lib/python3.10/os.py
  /usr/bin/python3
KiCad: 8.0.9-8.0.9-0~ubuntu22.04.1
  /usr/lib/python3/dist-packages/pcbnew.py
  /usr/bin/kicad
Kibot: 1.8.4
  /usr/lib/python3/dist-packages/kibot/__main__.py
  /usr/local/bin/kibot

Modules:
...

Tools:
Bash: 5.1.16 (GNU bash, version 5.1.16(1)-release (x86_64-pc-linux-gnu))
  /usr/bin/bash
...
KiCad Automation tools (kiauto): 2.3.5 (pcbnew_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License: Apache)
  /home/edward/.local/bin/pcbnew_do
...
```

However, from the `kibot -L logfile.txt` I see the following lines for the `out_pdf_sch_print` suggesting eeschema_do version 2.3.4?
```
DEBUG:- Importing out_pdf_sch_print (kibot - kiplot.py:91)
DEBUG:- Processing dependencies for `out_pdf_sch_print` (kibot - dep_downloader.py:1047)
DEBUG: - Data: {'Dependencies': [{'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '2.3.4'}]} (kibot - dep_downloader.py:1048)
DEBUG: - Registering dep {'from': 'KiAuto', 'role': 'mandatory', 'command': 'eeschema_do', 'version': '2.3.4'} (kibot - dep_downloader.py:1059)
DEBUG: - Dep after applying from KiAuto: {'name': 'KiCad Automation tools', 'github': 'INTI-CMNB/KiAuto', 'command': 'eeschema_do', 'pypi': 'kiauto', 'downloader': 'pytool', 'id': 'KiAuto', 'from': 'KiAuto', 'role': 'mandatory', 'version': '2.3.4'} (kibot - dep_downloader.py:987)
DEBUG:- Importing out_pdfunite (kibot - kiplot.py:91)
DEBUG:- Importing out_populate (kibot - kiplot.py:91)
DEBUG:- Importing out_position (kibot - kiplot.py:91)
DEBUG:- Importing out_ps (kibot - kiplot.py:91)
DEBUG:- Importing out_ps_sch_print (kibot - kiplot.py:91)
DEBUG:- Processing dependencies for `out_ps_sch_print` (kibot - dep_downloader.py:1047)
```

Running `whereis eeschema_do` I get:
```
eeschema_do: /usr/bin/eeschema_do /usr/local/bin/eeschema_do /home/edward/.local/bin/eeschema_do
```
And evaluating the version for each of those:
```
$ eeschema_do --version
eeschema_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License: Apache 2.0
$ /usr/bin/eeschema_do --version
eeschema_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License: Apache 2.0
$ /usr/local/bin/eeschema_do --version
eeschema_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License: Apache 2.0
$ /home/edward/.local/bin/eeschema_do --version
eeschema_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License: Apache 2.0
```

set-soft, 2025-05-20:
> However, from the kibot -L logfile.txt I see the following lines for the out_pdf_sch_print suggesting eeschema_do version 2.3.4?

You should look for:

```
DEBUG:Starting tool check for KiCad Automation tools (kibot - dep_downloader.py:841)
DEBUG:- Checking binary tool KiCad Automation tools (kibot - dep_downloader.py:747)
DEBUG:- Looking for tool `eeschema_do` at system level (kibot - dep_downloader.py:676)
DEBUG:- Checking version for `/usr/bin/eeschema_do` (kibot - dep_downloader.py:643)
DEBUG:- Needed version (2, 3, 4) (kibot - dep_downloader.py:659)
DEBUG:- Running ['/usr/bin/eeschema_do', '--version'] (kibot - dep_downloader.py:609)
DEBUG:- Looking for version in `eeschema_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License:
Apache 2.0` (kibot - dep_downloader.py:635)
DEBUG:- Found version (2, 3, 5, 0) (kibot - dep_downloader.py:670)
DEBUG:- Returning `/usr/bin/eeschema_do` (kibot - dep_downloader.py:848)
```

> Running whereis eeschema_do I get:

Having 3 copies isn't good. Keep just one.
The version reported might be from one of the three copies, the others might be different. This is python broken packaging behavior.

edlongman, 2025-05-20:
This is what I got:
```
DEBUG:Starting tool check for KiCad Automation tools (kibot - dep_downloader.py:841)
DEBUG:- Checking binary tool KiCad Automation tools (kibot - dep_downloader.py:747)
DEBUG:- Looking for tool `eeschema_do` at system level (kibot - dep_downloader.py:676)
DEBUG:- Checking version for `/home/edward/bin/eeschema_do` (kibot - dep_downloader.py:643)
DEBUG:- Needed version (2, 3, 4) (kibot - dep_downloader.py:659)
DEBUG:- Running ['/home/edward/bin/eeschema_do', '--version'] (kibot - dep_downloader.py:609)
DEBUG:- Looking for version in `eeschema_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License:
Apache 2.0` (kibot - dep_downloader.py:635)
DEBUG:- Found version (2, 3, 5, 0) (kibot - dep_downloader.py:670)
DEBUG:- Returning `/home/edward/bin/eeschema_do` (kibot - dep_downloader.py:848)
```

I was surprised by the `/home/edward/bin/eeschema_do`, which also had a load of python stuff in. So I removed kicad, python, kiauto and kibot and reinstalled everything. 

I checked that `whereis` returned nothing for eeschema_do, kibot and kicad before reinstalling. 

Still same problem. End of log file says: 
```
DEBUG:- Checking binary tool KiCad Automation tools (kibot - dep_downloader.py:747)
DEBUG:- Looking for tool `eeschema_do` at system level (kibot - dep_downloader.py:676)
DEBUG:- Checking version for `/usr/local/bin/eeschema_do` (kibot - dep_downloader.py:643)
DEBUG:- Needed version (2, 3, 4) (kibot - dep_downloader.py:659)
DEBUG:- Running ['/usr/local/bin/eeschema_do', '--version'] (kibot - dep_downloader.py:609)
DEBUG:- Looking for version in `eeschema_do 2.3.5 - Copyright 2018-2024, INTI/Productize SPRL - License:
Apache 2.0` (kibot - dep_downloader.py:635)
DEBUG:- Found version (2, 3, 5, 0) (kibot - dep_downloader.py:670)
DEBUG:- Returning `/usr/local/bin/eeschema_do` (kibot - dep_downloader.py:848)
DEBUG:Replacing images in schematic (kibot - out_base.py:1288)
DEBUG:Executing: /usr/local/bin/eeschema_do export --file_format pdf -o '/mnt/c/Users/edward/KiCad/schematic/out/schematic.pdf' --all_pages --default_font 'KiCad Font' /mnt/c/Users/edward/KiCad/schematic/Wiring-Diagram.kicad_sch /mnt/c/Users/edward/KiCad/schematic/out (kibot - kiplot.py:183)
DEBUG:Command line: ['/usr/local/bin/eeschema_do', 'export', '--file_format', 'pdf', '-o', '/mnt/c/Users/edward/KiCad/schematic/out/schematic.pdf', '--all_pages', '--default_font', 'KiCad Font', '/mnt/c/Users/edward/KiCad/schematic/Wiring-Diagram.kicad_sch', '/mnt/c/Users/edward/KiCad/schematic/out'] (kibot - kiplot.py:185)
DEBUG:Failed with error 2, retrying ... (kibot - kiplot.py:193)
DEBUG:Output from command:
> usage: eeschema_do [-h] [--disable_interposer] [--info] [--interposer_sniff]
>                    [--record] [--rec_width REC_WIDTH]
>                    [--rec_height REC_HEIGHT] [--separate_info]
>                    [--start_x11vnc] [--use_wm] [--verbose] [--version]
>                    [--wait_key] [--wait_start WAIT_START]
>                    [--time_out_scale TIME_OUT_SCALE]
>                    {export,run_erc,netlist,bom_xml} ... schematic output_dir
> eeschema_do: error: unrecognized arguments: --default_font KiCad Font
>  (kibot - kiplot.py:197)
ERROR:/usr/local/bin/eeschema_do returned 2 (kibot.gs - gs.py:899)
DEBUG:Removing temporal files (kibot - out_base.py:1108)
```

set-soft, 2025-05-20:
How did you install KiAuto?

edlongman, 2025-05-20:
Installed with ubuntu `apt install kiauto`

edlongman, 2025-05-22:
Thanks, running `ls -l /usr/local/bin/eeschema_do` shows it was installed on May 18 , 2023

I've removed it and reinstalled with `apt install kibot` - Finally fixed it!

Thank you very much for your help.

commit eac60d23 (2025-02-21): [*SCH Print][KiCad 9][Added] Default font

- Now available from the CLI
- Pushed by our report:
  https://gitlab.com/kicad/code/kicad/-/issues/17715

Closes #691