INTI-CMNB/KiBot #692: [BUG] Description field not overriden by user-defined field in generated BOM

nguyen-v, 2024-10-15:
**Describe the bug**
When using user-defined Description fields, the fields used in the generated BOM use the default symbol Description field instead of the user-defined one.

Generated from KiBot:

https://github.com/EPFLXplore/XRE_LeggedRobot_HW/blob/f52db22f6a9d3dfabf1af66d7178b2a86dc32618/amulet_controller/Generated/Manufacturing/Assembly/amulet_controller-bom.csv

https://github.com/EPFLXplore/XRE_LeggedRobot_HW/blob/f52db22f6a9d3dfabf1af66d7178b2a86dc32618/amulet_controller/Generated/Manufacturing/Assembly/amulet_controller-bom.xlsx

Generated with included KiCAD 8 BOM tools:

https://github.com/EPFLXplore/XRE_LeggedRobot_HW/blob/f52db22f6a9d3dfabf1af66d7178b2a86dc32618/amulet_controller/Manufacturing/Assembly/amulet_controller-bom-k8.csv

With the following settings

[image]

**To Reproduce**
`git clone git@github.com:EPFLXplore/XRE_LeggedRobot_HW.git`
`git checkout 40c042b9b4e9c8ce27140e0d6f83ee261c21e3bc`
`cd ./XRE_LeggedRobot_HW/amulet_controller`
`kibot -c kibot.kibot.yaml`

**Expected behavior**
The generated BOM Description column should contain user-defined fields, according to https://kibot.readthedocs.io/en/master/configuration/outputs.html#bom-columns

**Environment (please complete the following information):**
WSL2 on Windows, running Ubuntu 24.04 LTS. Using the dev branch.

set-soft, 2024-10-15:
The KiBot generated list is right and you get warnings explaining why.

```
WARNING:(W135) Description field mismatch for `L3` (SCH: `Power Inductors - SMD 2.2uH Shld 20% 1.2A 100 mOhms` PCB: `Inductor`) (kibot - kiplot.py:348)
```

closed 2024-10-15 by set-soft

nguyen-v, 2024-10-15:
Sorry about the wrong ticket, updating the PCB from schematic solved the issue