INTI-CMNB/KiBot #695: [BUG] PCB Print: Unknown font attribute error when using WKS template with colors or external font

nguyen-v, 2024-10-16:
Fill as much as you can, remove the rest

**Describe the bug**
Using PCB Print to generate a PDF file with a custom WKS template containing external fonts or colors raises an error:
`ERROR:In section 'pdf_assembly_top' (pcb_print): Error reading `/home/vincent/XRE_LeggedRobot_HW/amulet_controller/Templates/XRE_Legged_Template_Fab.kicad_wks` (Unknown font attribute `[Symbol('face'), 'Times New Roman']`) (kibot.gs - gs.py:846)`

**To Reproduce**
`git clone git@github.com:EPFLXplore/XRE_LeggedRobot_HW.git`
`cd ./XRE_LeggedRobot_HW`
`git checkout 85aaa8934ffd9bde8666cc6a0f3caa740e331dee`
`cd ./amulet_controller`
`kibot -c kibot.kibot.yaml`

**Expected behavior**
The output PDF should use the worksheet fonts when printing

**Environment (please complete the following information):**
WSL2 on Windows, running Ubuntu 24.04 LTS. Using the dev branch.

**Additional context**
Changing the code slightly (`WksFont.parse`) to include color and font parsing similar to the `FontEffects.parse_font` function in `v6_sch.py` removes the error, but not sure how to actually get the font (optionally color) to be used in the final print

set-soft, 2024-10-16:
Hi @nguyen-v !
I changed the title because this is a bug in KiCad docs, not in KiBot.
The KiCad docs are [here](https://dev-docs.kicad.org/en/file-formats/sexpr-worksheet/) and they currently say the fonts entry is:

```
(font [(size WIDTH HEIGHT)] [bold] [italic]) 
```

So I need examples of files with undocumented features so I can add them. I'll look into it.

set-soft, 2024-10-16:
After wasting a lot of time I came to the conclusion that the best we can so is to just parse the values. 
Currently looks impossible to use them, various KiCad "limitations" (IMHO bugs) prevents it:

1. The plot API fails to support colors, despite the fact that the plot controller has a SetColorMode member. 
2. The EDA_TEXT class supports setting a font SetFont, but the API is incomplete because you must pass KIFONT::FONT object ... which isn't defined
3. So ok, we have the *kicad-cli* option, no? well it doesn't support colors.
4. KiBot can also ask KiCad to "print" the frame ... when I tried it in my system using a font the resulting SVG damaged another of the fonts in the worksheet

So I'll add a patch to avoid errors when using it, but colors and fonts will be just ignored.

nguyen-v, 2024-10-16:
Thanks for the fast support! Is it possible to plot with kicad_cli to get the output with fonts (without colors)?

set-soft, 2024-10-16:
At least for SVG it looks like `kicad-cli` can render the correct fonts.
But I don't know if this is worth the effort (I mean implementing a "kicad-cli" worksheet strategy)

nguyen-v, 2024-10-16:
Using `frame_plot_mechanism: 'gui'` almost works but is weirdly cut
[image]
Using `frame_plot_mechanism: 'plot'` looks good, but doesn't keep track of page numbers, as documented
[image]

set-soft, 2024-10-16:
*gui* uses the KiCad print mechanism, which is OS dependent, so this is unreliable.
*plot* won't do all substitutions.
This is why *internal* is there. Is the best, but the new features for fonts aren't in the Python API.

set-soft, 2024-10-21:
Hi @nguyen-v !

After hitting two KiCad bugs (https://gitlab.com/kicad/code/kicad/-/issues/18928 and https://gitlab.com/kicad/code/kicad/-/issues/18959) I think I found a method to get:

- Custom fonts
- Colored text
- Images

Check if it works for you (is used by the default *internal*)

nguyen-v, 2024-10-21:
@set-soft 
It works great with all colors, fonts and images! Thank you!