INTI-CMNB/KiBot #696: [BUG] PCB Print: hole color in footprints ignored when plotting only copper layer(s)

nguyen-v, 2024-10-16:
**Describe the bug**
When printing only copper layers (for example only F.Cu, or F.Cu + In1.Cu), `colored_holes` and `holes_color` options are ignored for holes defined in footprints. Hole color is correct when printing copper layers with other non-copper layers (e.g. Margin, User.Comments, F.Courtyard, etc.).

Not a critical issue as I could print the copper layers with an empty non-copper layer, but behavior is certainly not expected.

Example of output:
https://github.com/EPFLXplore/XRE_LeggedRobot_HW/blob/8448cbcd72b614c6eafe8d6ddd7beaad7256eb10/amulet_controller/Generated/Manufacturing/Fabrication/amulet_controller-fabrication_layers.pdf

With the following settings:
https://github.com/EPFLXplore/XRE_LeggedRobot_HW/blob/8448cbcd72b614c6eafe8d6ddd7beaad7256eb10/amulet_controller/kibot.kibot.yaml#L306

**To Reproduce**
`git clone git@github.com:EPFLXplore/XRE_LeggedRobot_HW.git`
`cd ./XRE_LeggedRobot_HW`
`git checkout 8448cbcd72b614c6eafe8d6ddd7beaad7256eb10`
`cd ./amulet_controller`
`kibot -c kibot.kibot.yaml`

**Expected behavior**
Hole colors should follow the `colored_holes` and `holes_color` options, even when plotting only copper layers

**Screenshots**
Only F.Cu plotted:
[image]

F.Cu + In1.Cu:
[image]

F.Cu + Margin (hole color is correct)
[image]

**Environment (please complete the following information):**
WSL2 on Windows, running Ubuntu 24.04 LTS. Using the dev branch.

nguyen-v, 2024-10-16:
Correct behavior when plotting with `frame_plot_mechanism: 'gui'`. Frame is weirdly cut but that should probably be discussed in another issue
[image]

set-soft, 2024-10-17:
Please avoid mixing things.
About the first: is consequence of a new KiCad "feature" which draws holes in layers that never did (i.e. the Rescue layer)
This new behavior impacted to various plugins and outputs.

set-soft, 2024-10-17:
The above patch should solve the problem

closed 2024-10-17 by set-soft

set-soft, 2024-10-17:
> Correct behavior when plotting with `frame_plot_mechanism: 'gui'`. Frame is weirdly cut but that should probably be discussed in another issue

I'll suggest trying from the GUI: just print your design to an SVG file and you'll see why you get this.

commit 97309eaa (2024-10-17): [PCB Print][Fixed] Drill marks issues

- When colored_vias/pads are disabled
- And using KiCad 8.0.4+ which likes to print them in all layers

Fixes #696