INTI-CMNB/KiBot #703: [HOW] Keep PCB setup for panel

N0cynym, 2024-10-24:
**What do you want to achieve?**

I would like to have untended vias in my panel.
If I create gerbers from my panelized pcb, the `tent_vias` -> `viasonmask` option will be reseted in the panelize output file.

Is there any way of setting `tent_vias` for the panelize tool in the KiBot configuration?

This option is availible in [PDF output](https://kibot.readthedocs.io/en/master/configuration/outputs/PDFOptions.html).

set-soft, 2024-10-24:
The gerber output also has *tent_vias*: https://kibot.readthedocs.io/en/master/configuration/outputs/GerberOptions.html

N0cynym, 2024-10-24:
Thanks for this information @set-soft!

Do you think a feature request for the panelize tool makes sense to keep the pcb preferences from the input file for the output file?

set-soft, 2024-10-24:
Please elaborate, if you use the `gerber` output to generate the gerbers you just need to set the `tent_vias`. 
Are you talking about the GUI when you open the panel?

N0cynym, 2024-10-24:
Yes I mean the output panel PCB.

The settings in the pcb file are reset to default if I open the board file in the gui.

Would it be possible to copy the board settings from the origin?

N0cynym, 2024-10-25:
> The gerber output also has _tent_vias_: https://kibot.readthedocs.io/en/master/configuration/outputs/GerberOptions.html

I need to report that this option does not work for KiCad 8 projects with KiBot 1.8.1.

N0cynym, 2024-10-25:
I created bug report #74

set-soft, 2024-10-25:
The above patch adds an option to copy the setting after creating a panel.

The default value is *auto* and currently means *yes* for KiCad 8.

commit 94a4a9b8 (2024-10-25): [Panelize][Added] `copy_vias_on_mask` option

- To workaround KiCad's bug 18991
- Also added warning about how to workaround it in plotter outputs

Closes #703 related to #704
closed 2024-10-25 by set-soft