INTI-CMNB/KiBot #704: [BUG] Gerber export `tent_vias` option does not work

N0cynym, 2024-10-25:
**Describe the bug**

If I set in the KiBot configuration the `tent_vias: false` and the `viasonmask no` is set in the PCB file, the vias will still be covered by soldermask in the gerber export.

No matter if I skip preflights or run all default preflights.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a `GitLab` CI/CD workflow
- Using docker image `kicad8_auto:dev_1.8.2-ebe3520_k8.0.4_d_sid`

**Configuration**

Preflights:
```
preflight:
  run_erc: true
  update_xml: true
  run_drc: true
  check_zone_fills: false
  ignore_unconnected: false
  pcb_replace:
    date_command: date -d @`git log -1 --format='%at' -- "$KIBOT_PCB_NAME"` +%Y-%m-%d_%H:%M:%S
  sch_replace:
    date_command: date -d @`git log -1 --format='%at' -- "$KIBOT_PCB_NAME"` +%Y-%m-%d_%H:%M:%S
```

Gerber export:

```
  - name: 'gerbers'
    comment: "Gerber export for manufacturing"
    type: gerber
    dir: Gerbers
    options:
      # generic layer options
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: false
      use_aux_axis_as_origin: false
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: false

      # gerber options
      line_width: 0.1
      subtract_mask_from_silk: false
      use_protel_extensions: false
      gerber_precision: 4.6
      create_gerber_job_file: true
      use_gerber_x2_attributes: true
      use_gerber_net_attributes: true

    layers:
      - layer: F.Paste
        suffix: F_Paste
      - layer: F.SilkS
        suffix: F_SilkS
      - layer: F.Mask
        suffix: F_Mask
      - layer: F.Cu
        suffix: F_Cu
      - layer: B.Cu
        suffix: B_Cu
      - layer: B.Mask
        suffix: B_Mask
      - layer: B.SilkS
        suffix: B_SilkS
      - layer: B.Paste
        suffix: B_Paste
      - layer: Edge.Cuts
        suffix: Edge_Cuts
```

set-soft, 2024-10-25:
You should provide an example, a PCB, and illustrate what's wrong

set-soft, 2024-10-25:
Ok, I can confirm this is a bug in KiCad

N0cynym, 2024-10-25:
Thanks you very much for testing and reporting! 🙏

set-soft, 2024-10-25:
I added a warning about it instructing to set the option from the GUI