INTI-CMNB/KiBot #706: [FEATURE] Layer permutations for specific components

nguyen-v, 2024-10-28:
**Is your feature request related to a problem? Please describe.**
Not a problem, but a feature I am trying to achieve is to highlight testpoints with a different color on the PCB Print, such as:
[image]
[image]

**Describe the solution you'd like**
I am not sure how it could be implemented, but it would be nice to be able to swap layers of a specific component during PCB print (for example `F.Fab` layer with `User.Eco1` and `B.Fab` layer for `User.Eco2`, only for the testpoints). Maybe this could be implemented with a `permutation_filter` option in PCB Print, which could take a filter defined as:

```
  - name: permute_testpoints_layers
    comment: "Permute testpoint layers"
    type: permute
		- input_layers: ['F.Fab', 'B.Fab']
		- target_layers: ['User.Eco1', 'User.Eco2']
    include_only:
      - column: Reference
        regex: "TP"
```

**Describe alternatives you've considered**
Currently, I need to have special footprints for bottom and top layers, where the reference designator and filled circle are plotted on specific layers (in my case User.Eco1 and User.Eco2 respectively). This works but is not very practical.

**Additional context**
Add any other context or screenshots about the feature request here.

nguyen-v, 2024-10-28:
> Does your testpoints have a circle in the fab? I think I managed to achieve what you need, but assuming the above

Hi @set-soft , thanks for having a look at it
Currently with my solution, my testpoints do not have anything on the fab layers. I have two custom testpoints footprints (top, bottom), with the top one having the circle and reference designator on User.Eco1 and the bottom one on User.Eco2

set-soft, 2024-10-29:
Hi @nguyen-v !

Ok, I implemented it in a different way. Now you can apply exclude filters to individual layers in a page. The filter is an exclude filter that will exclude specified components. So you can remove anything that isn't a testpoint and print the fab layer. Here is a config example:

```yaml
kibot:
  version: 1

filters:
  - name: only_testpoints
    comment: "Select only testpoints"
    type: generic
    include_only:
      - column: Value
        regex: "^TestPoint"

outputs:
  - name: 'print_front'
    comment: "Print Top layer, testpoints in red"
    type: pcb_print
    dir: Layers
    options:
      format: 'PDF'
      keep_temporal_files: true
      colored_pads: false
      colored_vias: false
      pages:
        - sheet: Front
          sheet_reference_color: "#A02020"
          sketch_pads_on_fab_layers: true
          sketch_pad_line_width: 1
          layers:
            - layer: Edge.Cuts
              color: '#A0A0A0'
            - layer: F.Cu
              color: '#A0A0A0'
            - layer: F.SilkS
              color: '#A0A0A0'
            - layer: F.Fab
              plot_footprint_refs: false
              plot_footprint_values: false
              color: '#202020'
            - layer: F.Fab
              exclude_filter: 'only_testpoints'
              color: '#FF0000'
              plot_footprint_values: false
              sketch_pads_on_fab_layers: false
            - layer: Dwgs.User
```

Applied to the *tests/board_samples/kicad_8/test_points.kicad_pcb* example produces:

[image]

Here TP4 has a solid circle in the Fab layer, so you get it.

closed 2024-10-29 by set-soft

nguyen-v, 2024-10-29:
Working great, thanks!
In my case, it seems like some of the elements on the F.Fab are not excluded (circle manually added in pcbnew and mousebite footprints not present on the schematic)
[image]

set-soft, 2024-10-30:
> (circle manually added in pcbnew 

You should move it to a corresponding footprint or just draw it in another layer.

> and mousebite footprints also manually added in pcbnew not present on the schematic)

Are those real footprints? Do they have some reference value?

nguyen-v, 2024-10-30:
> You should move it to a corresponding footprint or just draw it in another layer.
Got it

> Are those real footprints? Do they have some reference value?
Those are real footprints, currently with a REF*** reference. Should I manually give them a reference (e.g. MB*) and exclude them with `exclude_refs`? They are also set as "Not in schematic"

set-soft, 2024-10-30:
> Those are real footprints (only present on the PCB), currently with a REF** reference. Should I manually give them a reference (e.g. MB*) and exclude them with `exclude_refs` (not sure if that would work, does KiBot only look at the schematic)? They are also set as "Not in schematic".

KiBot first looks at the schematic, but when we have a PCB it can add the components found in the PCB to the list of components to filter (a global option disables it when needed). So yes, annotate those components, they should be available for filtering.

commit 6c352691 (2024-10-29): [PCB Print][Added] a mechanism to filter components for a particular layer

Closes #706