INTI-CMNB/KiBot #708: [BUG] missing resource directory in latest release

allardMVP, 2024-10-29:
**Describe the bug**
I have a kibot script that triggers via a github workflow.  Everything worked last week just fine.  Now with the latest 1.8.2 I get a strange error `ERROR:Missing resource directory `tools` (kibot.gs - gs.py:850)` when trying to generate a drawing using `pcb_print`.  I can't seem to force kibot to use 1.8.1 anymore as it automatically jumps to 1.8.2 regardless of what version I specify in the github action.

Just to verify this is actually a kibot bug I went to another PCB repo that ran last week and reran the workflow.  Nothing has changed except the release of 1.8.2.  Last week the workflow ran flawlessly.  Today it breaks with the same error as above.  Am I missing something?

**To Reproduce**
Run a workflow using kibot 1.8.1 that ran fine last week.  As part of the workflow generate a drawing using pcb_print.  Error 

**Expected behavior**
Generate a drawing as it did last week.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
Where are you running KiBot:
- CI/CD on Github.  Both as an auto triggered and manually triggered gets the same result.
-  I have tried both 1.8.1 and 1.8.2 but 1.8.1 now auto updates to 1.8.2 and fails.  **Is there a way to force kibot to use 1.8.1 as that will work for me if I can use it**  
- For docker image I have tried the latest and forcing it to be 8.0.4_d12.1_b3.5.1

set-soft, 2024-10-29:
Hi @allardMVP !
A script was missing in the docker images and Debian package (not in PyPi). The script is used by the *pcb_print* output.
Now should be working.
Thanks for reporting!

closed 2024-10-29 by set-soft

allardMVP, 2024-10-29:
@set-soft Works like a charm!  Thanks for the fast turn around on this one!

Out of curiosity is there a way in the CI/CD for force kibot to not use the latest version?

Appreciate you maintaining this tool.

set-soft, 2024-10-29:
> Out of curiosity is there a way in the CI/CD for force kibot to not use the latest version?

If you pull the 1.8.1 image you'll get 1.8.1:

```
$ docker pull ghcr.io/inti-cmnb/kicad8_auto_full:1.8.1
1.8.1: Pulling from inti-cmnb/kicad8_auto_full
Digest: sha256:3647eb6b54e1c3b7a8dbe2500456a50271622ad29469f9595e4b2eca19413a26
Status: Image is up to date for ghcr.io/inti-cmnb/kicad8_auto_full:1.8.1
ghcr.io/inti-cmnb/kicad8_auto_full:1.8.1
$ docker run --rm -it ghcr.io/inti-cmnb/kicad8_auto_full:1.8.1
root@6bcef900613a:/# kibot --version
KiBot 1.8.1 - Copyright 2018-2024, Salvador E. Tropea/INTI/John Beard - License: AGPL v3
```

I don't know what do you mean with *1.8.1 now auto updates to 1.8.2*

set-soft, 2024-10-29:
For GitHub Actions the tags are different, they are documented here: https://kibot.readthedocs.io/en/master/usage_with_ci_cd.html#github-actions-tags

For KiCad 8 is **v2_k8_1_8_1**