INTI-CMNB/KiBot #709: [HOW]To generate a gerber paste layer with all pads

MarkusSBS, 2024-10-30:
**What do you want to achieve?**
I want to generate a paste layer for alle variants and with no filter applied 
but the paste layer is removed for all parts that are missing for  variants

```
  - name: 'gerbers'
    comment: "Gerbers for the board house"
    type: gerber
    dir: Gerbers
    options:
      # generic layer options
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: true
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: false
      force_plot_invisible_refs_vals: false
      tent_vias: true
      use_protel_extensions: true
      create_gerber_job_file: false
      output: "${SBS_SAP}_%p-%v_Rev%r-%i_${SBS_Doc_status}_%D-${git_hash}.%x"
      gerber_precision: 4.6
      use_gerber_x2_attributes: false
      use_gerber_net_attributes: false
      disable_aperture_macros: true
      line_width: 0.1
      uppercase_extensions: true
      subtract_mask_from_silk: true
      inner_extension_pattern: '.g%n'
      edge_cut_extension: '.gml'
    layers:
      - 'copper'
      - 'F.SilkS'
      - 'B.SilkS'
      - 'F.Mask'
      - 'B.Mask'
      - 'Edge.Cuts'
      - 'F.Paste'
      - 'B.Paste'
      - 'F.Mark'
```
is it possible to have a gerber export job that doesn't  apply any variant filter?

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`
- Using a docker image? ghcr.io/inti-cmnb/kicad8_auto_full:dev

set-soft, 2024-10-30:
Set the global `remove_solder_paste_for_dnp` variable to **false**.

You can control various details using global variables: https://kibot.readthedocs.io/en/master/configuration/global.html

closed 2024-10-30 by set-soft

MarkusSBS, 2024-10-30:
but with this it is not possible to generate 2  gerber files. in 1 output job right? I want to generate 1 gerber with all cutouts and 1 with only specific cutouts

MarkusSBS, 2024-10-30:
I have tried a all Variant 

```
- name: 'gerbers_ALLPaste'
    comment: "Gerbers for the board house"
    type: gerber
    dir: Gerbers/ALLPaste
    options:
      # generic layer options
      variant: all
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: true
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: false
      force_plot_invisible_refs_vals: false
      tent_vias: true
      use_protel_extensions: true
      create_gerber_job_file: false
      output: "${SBS_SAP}_%p-%v_Rev%r-%i_${SBS_Doc_status}_%D-${git_hash}.%x"
      gerber_precision: 4.6
      use_gerber_x2_attributes: false
      use_gerber_net_attributes: false
      disable_aperture_macros: true
      line_width: 0.1
      uppercase_extensions: true
      subtract_mask_from_silk: true
      inner_extension_pattern: '.g%n'
      edge_cut_extension: '.gml'
    layers:
      - 'F.Paste'
      - 'B.Paste'
```
but something is alwayes removed by some variant

MarkusSBS, 2024-10-30:
i think the global does not help at all because i can only deactivate it for the projekt not for a single run?

set-soft, 2024-10-30:
Just run KiBot twice.

You can use the `-g` command line option to set a global variable.

So you can first run KiBot for the common case, then just run it again changing the global variable and asking to generate only the needed stuff.