INTI-CMNB/KiBot #711: [FEATURE] Add images generated by output to PCB Print

nguyen-v, 2024-10-30:
**Is your feature request related to a problem? Please describe.**
In my PCB assembly documents, I like to have pictures of the PCB
[image]
[image]

It would be nice to be able to place an image generated by an output on a specific layer and plot it in PCB Print just like any other layer.

**Describe the solution you'd like**
Ideally, it would use the 'reference image' object from KiCAD 8 (usually used for reverse engineering), and the position/size could be defined by a group (similar to the include_table preflight). 

**Describe alternatives you've considered**
Alternatively, I could run a custom script (e.g. using something like PyPDF) that adds the image to the PDF document after a KiBot run.

**Additional context**
I had a try at a first implementation, but it seems like using KiCAD's reference image object is quite complex (due to incomplete API).
The [pcbnew.PCB_REFERENCE_IMAGE Class](https://docs.kicad.org/doxygen-python-8.0/classpcbnew_1_1PCB__REFERENCE__IMAGE.html) only exposes a few functions, and I haven't managed to write to or read from a PCB_REFERENCE_IMAGE object on the PCB (I tried using the `MutableImage()`, `GetImage()`, `ReadImageFile()` but there is no definition of `BITMAP_BASE *` in the Python API that is returned by the first two functions.

It may be possible to do it with a temporary sheet specific WKS (?) but it seems a bit overkill for this

set-soft, 2024-10-31:
Hi @nguyen-v !
Hmmm ... I have a lot of doubts about it.
Looking at "reference image" I see they:

1. Are explicitly ignored by KiCad for the plot operations
2. They look "washed" in the screen, I understand this is to allow using them as guidance to copy the design, but you can't adjust the amount  of "washing".

So they have some limited use. They could be used just to define a position and size. I'll take a look at what they offer.

set-soft, 2024-10-31:
A problem with KiCad 8.0.0 to 8.0.6 is that it fails to load images when using Python.
Should be fixed in 8.0.7: https://gitlab.com/kicad/code/kicad/-/issues/18959
But currently you can't even load the board from Python.

set-soft, 2024-10-31:
dcaa5643d224ad134c7df57ee8579846f97258e7 is a WIP possible solution.

Uses the group approach, when KiCad's above mentioned bug is fixed could be complemented by a preflight to actually put images there, so you can see them.

nguyen-v, 2024-10-31:
It's working great, thanks! Getting closer to the perfect, fully automated assembly document :)

set-soft, 2024-11-01:
I removed the underscore from *_kibot_image_OUTPUT*, to make it coherent with the other group names