INTI-CMNB/KiBot #713: [BUG/FEATURE?] Incorrect (?) work layer used for KiCad 7+

nguyen-v, 2024-11-08:
**Describe the bug**
The `GS.work_layer` defined in `__main.py__` is defined as `User.9` for KiCad versions 6+. 

**Expected behavior**
Ideally, work layer should be set as 'Rescue' for KiCad 7+. This is useful because otherwise 'User.9' cannot be used for other purposes.

**Environment (please complete the following information):**
Where are you running KiBot:
WSL2 on Windows, running Ubuntu 24.04 LTS. Using the dev branch.

set-soft, 2024-11-11:
I don't even remember what error did it generate.
I'm enabling the use of "Rescue" for 7+

closed 2024-11-11 by set-soft

set-soft, 2024-11-11:
Well, now is clear, KiCad 6+ doesn't support loading a file with objects in the Rescue layer.
When you save the file it saves the Rescue layer, but when you load it you get an error.

nguyen-v, 2024-11-11:
How about temporarily using the 'Rescue' layer and deleting objects before saving? In my case I use the 'User.9' layer for graphical purposes (list of testpoints). As the 'Include Image' feature moves the group to the 'User.9' layer, it results in artifacts in PCB Print (should I open a separate issue for this?)

set-soft, 2024-11-12:
> How about temporarily using the 'Rescue' layer and deleting objects before saving? 

And then we can't undo the variant.

> In my case I use the 'User.9' layer for graphical purposes (list of testpoints).

I tested using the "Margin" layer and it passed the tests. So i'll add an option to choose the layer, defaulting to Margin.
According to what I found Margin is reserved, but not yet used by KiCad itself, so for 6, 7 and 8 it looks like a good candidate.

>  As the 'Include Image' feature moves the group to the 'User.9' layer, it results in artifacts in PCB Print when plotting User.9 (should I open a separate issue for this?)

Please elaborate, what is moved? I don't think this is related to the 'Include Image' mechanism.

set-soft, 2024-11-12:
Ok, the above patch changes it to "Margin" and you can now choose any layer you want.

commit a3e99c17 (2024-11-11): [Changed] Try to use "Rescue" layer again

- It failed for some 6.x version
- Seems to be working for 7/8

Closes #713