INTI-CMNB/KiBot #714: [FEATURE] Include image from output in schematic

nguyen-v, 2024-11-09:
**Is your feature request related to a problem? Please describe.**
In the schematic, I like to add front and back images of the PCB:
[image]

It would be nice to be able to include images generated from an output in the schematic.

**Describe the solution you'd like**
A way to add PNG images generated from an output at a specific location in the schematic. A possible approach could be to create a text box of the correct size with as an hyperlink (to avoid garbage text from appearing in the final PDF) something like `#kibot_image_*`. A PNG image could then be placed at the correct location/scale.

**Describe alternatives you've considered**
Using a custom python script after a kibot run to place the PNG image at the correct location on the correct page.

set-soft, 2024-11-12:
Ok, the above patch adds the mechanism.
Is documented in the global `sch_image_prefix` variable.
You add a text box like you proposed and the text is the prefix + _ + output name.
This affects some outputs, currently:

- Schematic print (PDF, SVG, PS, HPGL, DXF)
- Schematic variant

This isn't a preflight so it can be used in one pass, and you can also run a simple SCH variant target and replace the images (never do it without a revision system!).
It has some barely tested logic to allow already having the images, in this case they get replaced.
Note that only the box width is relevant **BUT** if you put default images then the text box height must also match (an error of 0.1 mm is accepted).
Also note that after pasting the images the text box becomes invisible. For this its color has 0 alpha, but also has the size is adjusted to the size of the image and the text is made of size 0. This is because the invisible box becomes visible in the PDF (I guess this is because transparency issues in the PDF)

Please let me know if it works for you.

nguyen-v, 2024-11-12:
Also, I think the bitmap y position should be set as:
`bmp.pos_y = box.pos_y + img_h/2*scale` so that it is top aligned in the textbox (currently if the image has a tall aspect ratio, it goes out of frame on the top)

set-soft, 2024-11-13:
> Hi @set-soft, thank you for taking the time to implement the feature. It's working, but I get an error `ERROR:Restore (kibot - out_any_sch_print.py:115)`. Not sure if it should throw an error on this particular line

Just a debug line I forgot to remove, thanks for reporting

nguyen-v, 2024-11-14:
Working nicely, thanks!

closed 2024-11-14 by nguyen-v