INTI-CMNB/KiBot #716: [BUG] Multipart component not selected, in schematic, interactive BOM, ... - -not in CSV BOM.

mdeweerd, 2024-11-11:
# Summary

A multipart component is shown as fitted on the schematic, PCB 3D model output, etc..
It is not shown as fitted the CSV BOM output when not selected by the variant.

In a KiBOT 1.6.6. dev version I had the desired output in April 2024 according to the build info.

# More

In fact, all these multipart components are shown as fitted regardless of the variant filter outcome.
Other components with the same variant field are not fitted (as expected).

These are the essential properties for J4, a component that should not be fitted..
[image]

This is the report int he log, I conclude that the filter result is correct (variant flw240v not matched);
```
DEBUG:J4#1 fitted: True included: True (kibot - var_kicost.py:87)
DEBUG:ref: J4#1 value: CONN_2 - p1/2 variants: flw240v -> False (kibot - var_kicost.py:94)
DEBUG:J4#2 fitted: True included: True (kibot - var_kicost.py:87)
DEBUG:ref: J4#2 value: CONN_2 - p2/2 variants: flw240v -> False (kibot - var_kicost.py:94)
```

The vesion used for Kibot is KiBot 1.8.3 with the docker image:
dev: Pulling from inti-cmnb/kicad8_auto_full
Digest: sha256:f43d77f4a9a4c7a736f7303d1ab30a7d50c35aa14c3f2014dc9a7b91a76a4182

For the interactive BOM, J4#{1,2} is in the blacklist, but the component is found in the BOM as J4.  I guess that this is why J4 is in the ibom (exluding J4#..., not excluding J4).
```
 '--blacklist', 'C10,C11,C12,C13,C14,C18,C2,C20,C21,C22,C23,C24,C25,C26,C29,C31,C34,C35,C38,C39,C4,C44,C45,C46,C47,C48,C49,C6,C7,C8,C9,D10,D2,D3,D4,D5,D7,D8,F1,J1#1,J1#2,J11,J12,J13,J14,J17,J18,J19,J4#1,J4#2,J5,J6,J7,J9,P40,Q2,Q3,Q4,Q5,R1,R10,R11,R12,R13,R15,R18,R19,R2,R20,R21,R23,R24,R25,R26,R27,R28,R29,R3,R31,R32,R33,R34,R37,R38,R39,R41,R42,R43,R44,R45,R46,R48,R49,R5,R50,R51,R53,R54,R56,R57,R6,R61,R62,R63,R66,R68,R7,R79,R80,R82,R83,R84,R85,R88,R89,R9,R90,R91,R92,R93,RV1,RV3,U10,U10,U10,U14,U15,U16,U2,U4,U5,U6,U6,U6,U7,U8,U8,U8,U9,U9,U9'] 
```

Variants are specified like this:
```
filters:
  - name: Tempo
    file_id: _Tempo
    variant: ^(mot1|noF1|board_type_tp|screen5v|vtx|64k|notvneg)$
    type: kicost
    pre_transform: [kicost_rename,subparts_splitter]
```

This was still fine with, on Apr. 26th, 2024 (a dev version):
```
KiCad: 8.0.1+dfsg-1
  /usr/lib/python3/dist-packages/pcbnew.py
  /usr/bin/kicad
Kibot: 1.6.6
  /usr/local/lib/python3.11/dist-packages/kibot/__main__.py
  /usr/local/bin/kibot
```

Using tagged KiBOT 1.6.5 version (ghcr.io/inti-cmnb/kicad8_auto_full:1.6.5) does not produce the desired result.
I am trying dev_1.6.6-f43fc40_k8.0.1_d_sid_b3.5.1 next.

Maybe this is enought to understand what the issue is, if not I'll set up a test ase.

set-soft, 2024-11-12:
Please provide a very simple reproducer example.

mdeweerd, 2024-11-20:
Here is a test case - I had to boil the actual design down to this - creating one from scratch did not reproduce the issue.

[variantTc.zip](https://github.com/user-attachments/files/17823911/variantTc.zip)

'variant1' should not have all the connectors selected but it does (mot1, flw240v are not selected by variant1).

```
docker compose run --rm dev_k8
# or
kibot -c variantTc.kibot.yaml -g variant=variant1 -b variantTc.kicad_pcb -e variantTc.kicad_sch info schematic_bw 
kibot -c variantTc.kibot.yaml -g variant=all -b variantTc.kicad_pcb -e variantTc.kicad_sch info schematic_bw 
```

I hope this reproduces it for you too.

set-soft, 2024-11-20:
Hi @mdeweerd !

The iBoM case is quite obvious, the black list mentions the subpart, and iBoM doesn't know about it. I doubt it ever worked. The solution here is to use `use_ref_sep_for_first: false`, so you have J3 and J3#1, and not J3#1 and J3#2.

The PDF case is much more subtle, J3 and J4 are split into 4 components, but none of them is the original! And when we export the SCH we get the originals.

set-soft, 2024-11-20:
Hi @mdeweerd !

I doubt it ever worked, I guess it worked before you converted these connectors to multiparts.

The iBoM issue is quite clear, J4#1 isn't known by iBoM so it doesn't help to black list it.

The SCH issue is much more subtle: neither J4#1 nor J4#2 are in the schematic, so marking them as "not fitted" doesn't make J4 a not fitted component.

I'm working on a solution.

set-soft, 2024-11-20:
Ok, the above patch passed your test example.
Maybe other outputs needs to take care of this detail, but now we have the infrastructure to solve it.

mdeweerd, 2024-11-20:
@set-soft 👍 Thanks for the patch!  Tested on the schematic output of the project and it works.

I've had this multipart since a long time and had good results in April and before, but I had to make some small updates and did not get the expected output recently.

Also, a similar test case with multiparts worked, created as a new project and I copy/pasted the part from the original design.

If I see issues in other outputs, I'll report them separately.

mdeweerd, 2024-11-20:
I've identified the difference.  I've added this since April:

```yaml
  - name: subparts_splitter
    type: subparts
    # We want to also split the `Description` field
    split_fields: ['Description','comment']
    split_fields_expand: true
    separators: ';'
```

From BOM generated in April:

16 | J1,J2 | 2 | 300V 15A Straight   Circular Pins 2mm 2P 7mm 3.5mm 5mm 1x2P Plugin,P=5mm Pin Header;Matching   Terminal block 2P | TL205P-02P-K06;TL205T-02P-K02S
-- | -- | -- | -- | --

So multi-parts were just a single part as far as kibot was concerned.

set-soft, 2024-11-21:
> So multi-parts were just a single part as far as kibot was concerned.

That makes sense.