INTI-CMNB/KiBot #721: [FEATURE] Exclude filter for ibom Fabrication layers

nguyen-v, 2024-11-13:
Fill as much as you can, remove the rest

**Is your feature request related to a problem? Please describe.**
Currently, when generating an HTML BOM, if `show_fabrication` option is true, all fabrication (graphics and reference designators) are shown. When there are many testpoints, the ibom can get really cluttered.
[image]

**Describe the solution you'd like**
It would be nice to be able to specify an exclude_filter (similar to the current implementation of PCB Print), which would allow to hide specific components on the PCB.

**Describe alternatives you've considered**
I've thought about using the `hide_excluded` option and setting all testpoints as DNF, but that would result in a red cross on all testpoints in the schematic.

set-soft, 2024-11-14:
Why not using a variant?

nguyen-v, 2024-11-18:
> Why not using a variant?

I'm not sure how to achieve that. Let's say that I have the variants `FULL`, `NO_XT60` and `NO_SNUBBER`, does that mean that I would need to create three new variants `FULL_TP`, `NO_SNUBBER_TP` and `NO_XT60_TP`, set the `Config` field of the testpoints to: `-FULL_TP,-NO_SNUBBER_TP,-NO_XT60_TP` and in the yaml file do something like:

```
outputs:
- name: html_bom
  type: ibom
  options:
    show_fields: 'Value,Footprint,Manufacturer Part Number'
    group_fields: 'Value,Manufacturer Part Number'
    show_fabrication: true
	variant: '%V_TP`
```
Is there a better way, or do I have to create an 'alternate' variant for each variant?

set-soft, 2024-11-19:
> set the `Config` field of the testpoints to: `-RELEASED_TP,-NO_SNUBBER_TP,-NO_XT60_TP` and in the yaml file do something like:

Nope, just add them to the exclude filter.

You can use the pre-processor to modify the already existing variants and use the same files, just in two separated runs.

But now I also added a local "exclude_filter" (really experimental), you can try adding a filter to the iBoM output. This filter should exclude the TPs. It will be applied *before* the variant.

nguyen-v, 2024-11-19:
It's working great, but needs the following patch which fixes an error raised when using the filter
`ERROR:Failed to create BoM, error 2 (kibot.gs - gs.py:881)`

```diff --git a/kibot/out_ibom.py b/kibot/out_ibom.py
index f2ac63ae..e00e3a1a 100644
--- a/kibot/out_ibom.py
+++ b/kibot/out_ibom.py
@@ -217,7 +217,7 @@ class IBoMOptions(VariantOptions):
         template = IBoMOptions()
         for k, v in template.get_attrs_gen():
             v = getattr(self, k)
-            if not v or k in ['output', 'variant', 'dnf_filter', 'pre_transform', 'hide_excluded', 'forced_name']:
+            if not v or k in ['output', 'variant', 'dnf_filter', 'exclude_filter', 'pre_transform', 'hide_excluded', 'forced_name']:
                 continue
             if k == 'offset_back_rotation' and version < (2, 5, 0, 2):
                 continue
```

set-soft, 2024-11-20:
Fixed, thanks!

closed 2024-11-21 by nguyen-v

set-soft, 2024-11-21:
Should be available now