INTI-CMNB/KiBot #723: [HOW] Import custom/external librarys

N0cynym, 2024-11-14:
**What do you want to achieve?**

I would like to use all ERC and DRC tests which KiCad offers.

That's working local pretty well but if I use a remote server for my KiBot execution I end up with a different result.

If I create or use some custom/external symbol or footprint librarys in my project I end up with ERC and DRC fails while executing KiBot as docker image:

* `lib_symbol_issues`
  * `The current configuration does not include the library 'xyz'`
  * `Symbol '123' not found in symbol library 'xyz'`
* `lib_footprint_issues`
  * `The current configuration does not include the library 'xyz'`
  * `Footprint '123' not found in library 'xyz'`

Is there any idea or way of importing missing librarys or do I need to build my own KiBot docker?

set-soft, 2024-11-14:
You have various options:

- Add the libs to your repo, directly or using sub modules
- Download them in your workflow and make them available
- Just filter those errors on the workflow

N0cynym, 2024-11-15:
Thanks @set-soft for your time to have a look into my issue.

* About the first option:
  * Don't we had kind of a bug with KiBot and Git submodules? 🤔 
* About option two:
  * That sounds great!
  * Could you provide me the path I should download it into the docker?

Egoruch, 2025-02-05:
> About option two:
> * That sounds great!
> * Could you provide me the path I should download it into the docker?

I'm using only global library (no local) to use components across all project, so full library stored in git repository.
And to use it correcly with GitHub Actions virtual machine I tried following mechanism and it worked:

1) Copy kicad config files (because sym-lib-table contain paths to my custom libs)
2) Checkout repository with library files

`name: "Variants demo using --quick-start"

# Controls when the action will run. Triggers the workflow on push or pull request
# events but only for the master branch
on:
  push:
    paths:
      - '*.kicad_sch'
      - '*.kicad_pcb'
      - '.github/workflows/kibot_quick_start.yml'
  pull_request:
    paths:
      - '*.kicad_sch'
      - '*.kicad_pcb'
      - '.github/workflows/kibot_quick_start.yml'
  repository_dispatch:
    types: [run_qs]

# A workflow run is made up of one or more jobs that can run sequentially or in parallel
jobs:
  make:
    runs-on: ubuntu-latest
    container: ghcr.io/inti-cmnb/kicad7_auto_full:latest

    steps:
    - uses: actions/checkout@v3
      with:
        # So we can run a diff between last 2 changes
        fetch-depth: '0'

    - name: Move KiCad Config files
      run: |
        ls
        cp -r kicad_config/. /root/.config/kicad/7.0/
        cd /root/.config/kicad/7.0/
        ls

    - name: Check config file
      run: |
        cd /root/.config/kicad/7.0/
        cat sym-lib-table

    - name: Check out my other private repo
      uses: actions/checkout@v3
      with:
        token: ${{ secrets.KICAD_LIB_SSH }}
        repository: Egoruch/KICAD-LIB
        path: kilib
        ref: main # Branch name????
        persist-credentials: false

    - name: Check Private Repo (Library Files)
      run: |
        ls
        cd kilib
        ls
        cd symbols
        ls
        cd custom
        ls

    - name: Quick Start
      run: |
        kibot -c KiBot/KICAD-REL-4L.kibot.yaml

    - name: Retrieve results
      uses: actions/upload-artifact@v2
      with:
        name: Automatic_outputs
        path: REL

    - name: Check REL Files
      run: |
        ls
        cd REL
        ls

    - name: Commit and Push Modifications (After Generating)
      uses: actions-js/push@master
      with:
        github_token: ${{ secrets.GITHUB_TOKEN }}

  # This step is to upload the results to another repo (web pages)
  deploy:
    runs-on: ubuntu-latest
    needs: make

    steps:
    - uses: actions/checkout@v3

    - name: Retrieve results
      uses: actions/download-artifact@v2
      with:
        name: Automatic_outputs
        path: REL`