INTI-CMNB/KiBot #725: [BUG] Component misalignment in PCB Print

nguyen-v, 2024-11-19:
Fill as much as you can, remove the rest

**Describe the bug**
Some PCB components and DNP crosses are sometimes misaligned, not sure what the cause is but seems to happen more often with THT components:

**To Reproduce**
```
git clone git@github.com:nguyen-v/kibot_tests.git
git checkout 4baff4313ff1c27bc6ed19d11dfb3790b7325f39
kibot -c kibot_tests.yaml
```

Result is in PCB folder

**Expected behavior**
Components should be correctly aligned

**Screenshots**
Pad not aligned:
[image]

DNP cross not aligned:
[image]

**Environment (please complete the following information):**
Where are you running KiBot:
WSL2 on Windows 10, dev branch

set-soft, 2024-11-21:
Ok, the problem here was about the "center" of rotation.
Most SMD components has its center at the geometric center. But this isn't true for THT components, in many cases with center at pin 1.
Now the rotated cross should be using the correct center.
The pad shift was because rotating a footprint might change its center ... so you must memorize the original center.

closed 2024-11-21 by set-soft

nguyen-v, 2024-11-21:
Thanks for the quick fix!

commit 22d4a5d5 (2024-11-21): [Do Not Fit footprint crosses][Fixed] Not centered

- Problems with footprints where the center isn't the geometric center

Fixes #725