INTI-CMNB/KiBot #726: [BUG/HOW?] Filter not behaving as expected for BOM with variants

nguyen-v, 2024-11-19:
**Describe the bug**
When generating a CSV list of testpoints with a variant selected, the CSV files generated are empty

**To Reproduce**
```
git clone git@github.com:nguyen-v/kibot_tests.git
git checkout 4baff4313ff1c27bc6ed19d11dfb3790b7325f39
kibot -c kibot_tests.yaml -g variant=RELEASED
```
last command can also be replaced with:
```
kibot -c kibot_tests.yaml -g variant=NO_LED
```

Results are in the Testpoints folder
**Expected behavior**
The CSV list of testpoints should not be empty when a variant is selected. Is there some sort of default variant applied, even when `exclude_filter` is given?

**Environment (please complete the following information):**
Where are you running KiBot:
WSL2 under Windows 10, on the dev branch

set-soft, 2024-11-20:
Hi @nguyen-v !
Please read the docs about KiBoM variants: https://kibot.readthedocs.io/en/latest/configuration/variants/kibom.html

Pay attention to the default `exclude_filter`

KiBoM variants behaves ... well like KiBoM, the defaults excludes mechanical stuff (including TPs)

closed 2024-11-20 by set-soft

set-soft, 2024-11-20:
> Is this also the case when a filter is given?

Please be more specific. 

>  In my case I give it a 'only_testpoints_top' and 'only_testpoints_bottom' filters (see https://github.com/nguyen-v/kibot_tests/blob/4baff4313ff1c27bc6ed19d11dfb3790b7325f39/kibot_tests.yaml#L49).

From what I see you pass it as a separated filter. The short answer is "no", but if you apply a KiBoM variant then "yes".

>  If this is the case, how can I create my own filters that do not exclude the testpoints during a variant?

In the variant you can specify an `exclude_filter`, don't use the default KiBoM exclude filter.
Also: KiCost variants are usually more flexible.