INTI-CMNB/KiBot #729: [BUG] Outputs concerned by multipart issue #716

mdeweerd, 2024-11-21:
In relation to #716, I found that the following outputs also suffer from the multipart selection issue:

- STEP output;
- 3D output (legacy) - I have not switched to blender yet;
- Fab layers (when outputting to PDF, I guess the gerbers also suffer from this).

As a workaround for me, I'll disable the subparts splitter for now.  I will enable it again in case of updates to kibot to verify the behavior.

If you want, I can update the testcase makefile to generate all the outputs.

mdeweerd, 2024-11-21:
As I've disabled the subpart splitter function, I now also notice that this is also causing a vert important performance hit.

With the subparts_filter active, generating all the outputs for one variant (a single kibot execution) took just over 50 minutes.
Without the option active, I am back down to under 5 minutes.

So that's a 10 fold performance impact.

set-soft, 2024-11-21:
> So that's a 10 fold performance impact.

If you are concerned about speed: Run KiBot in separated runs, in one run generate the variants, then run KiBot without variants but using the generated PCBs/Schematics in previous run.

set-soft, 2024-11-21:
> If you want, I can update the testcase makefile to generate all the outputs.

I'm currently looking at other stuff, please do it

set-soft, 2024-11-21:
To be honest I never liked the "multiparts" feature of KiCost, IMHO is dirty.
I include the "extra" components in the schematic and then I mark them as "not for the PCB".
I'll try to support some of the usage, but this will be marked as deprecated.

mdeweerd, 2024-11-21:
For fuse holders when there may be 3 parts:
- The fuse holder;
- The fuse;
- The cap to close the fuse holder.

And all three are mounted by the PCBA contractor.

set-soft, 2024-11-21:
So what?
Why don't you just draw the fuse and the cap holder in your schematic? (marking them as not for PCB)

set-soft, 2024-11-21:
BTW: I think you should try to apply the subparts filter only to the BoM, in the local transform filter.