INTI-CMNB/KiBot #730: [FEATURE] Standalone CSV PCB assembly statistics

nguyen-v, 2024-11-21:
**Is your feature request related to a problem? Please describe.**
In my assembly documents, I like to add a table with more details on the assembly:

**Describe the solution you'd like**
[image]

Currently, there is no way to easily get this information, there is the `hide_stats_info` in BOM but doesn't have all of the information. Ideally the columns would be:
- Type
- Front side
- Back side
- Unique components (similar to Component groups)
- Fitted
- Total

And the rows:
- THT
- SMD
- Total

Would be ideal if the column names could be parametrized too.

Then, the CSV file could be included with the `Include Table` preflight

**Describe alternatives you've considered**
Currently I fill in the table manually

set-soft, 2024-11-22:
Hi @nguyen-v !
The above patch adds the necessary stuff to use `report` output to achieve it.
The method is working [here](https://github.com/INTI-CMNB/KiBot/tree/dev/docs/samples/Component_Count_Table)

closed 2024-11-22 by set-soft

nguyen-v, 2024-11-22:
Hi @set-soft, it raises the error:
` Missing report template` when trying to generate the output, I think the report file is missing

set-soft, 2024-11-22:
True, the above commit contains the template

nguyen-v, 2024-11-22:
> Thanks, however the results given by the report and KiCost don't seem to coincide. It seems like it is not filtering the same components (in my case, the fiducial markers and net ties seem to be counted (they are set as exclude from BoM), but THT component like the XT60 are not counted)

Sorry, it seems my THT component was set as exclude from position files, which meant that its attribute was 0x0005 (= 0x001 (THT) & 0x0004 (exclude from position files)

nguyen-v, 2024-11-27:
Any thought on this? Maybe some code from other outputs could be reused for the component groups

set-soft, 2024-11-27:
> It seems like it is not filtering the same components

I added discriminated counters, so you can compute: ${top_total-top_total_dnp}
Note that for this:

1. Some variant or filter must be applied
2. The DNP/DNC/Excluded is the KiBot concept, loosely related to KiCad flags

set-soft, 2024-11-27:
> Any thought on this? Maybe some code from other outputs could be reused for the component groups

Looks like an overkill. Perhaps some simplified version

nguyen-v, 2024-11-27:
> I added discriminated counters, so you can compute: ${top_total-top_total_dnp} Note that for this:
> 
> 1. Some variant or filter must be applied
> 2. The DNP/DNC/Excluded is the KiBot concept, loosely related to KiCad flags

Thanks, however I had to change the report template (also substracting excluded components) to get consistent results with kibot, and using the `exclude_filter: '_mechanical'`. This is because not fitted components in the schematic are set with do not populate, but things like net ties are only set as 'exclude from bom' in the PCB.

```
Type,Front Side,Back Side,Total
THT,${top_tht-top_tht_dnp-top_tht_exc},${bot_tht-bot_tht_dnp-bot_tht_exc},${total_tht-total_tht_dnp-total_tht_exc}
SMD,${top_smd-top_smd_dnp-top_smd_exc},${bot_smd-bot_smd_dnp-bot_smd_exc},${total_smd-total_smd_dnp-total_smd_exc}
Total,${top_total-top_total_dnp-top_total_exc},${bot_total-bot_total_dnp-bot_total_exc},${total_all-total_all_dnp-total_all_exc}
```

set-soft, 2024-11-28:
Yes, this look right for a default, but don't forget templates are just examples.

commit d5761421 (2024-11-22): [Report][Added] Components count extras

- `top_total`, `bot_total`, `total_smd`, `total_tht` and `total_all`
  component counts
- `total_components` template, intended to generate CSV files with
  component counts
- A demo about it

Closes #730