INTI-CMNB/KiBot #734: [FEATURE] 2D Engineering drawing of PCB

nguyen-v, 2024-11-22:
**Is your feature request related to a problem? Please describe.**
Just an idea I am throwing out there. In altium, it is possible to make 2d engineering drawings of the PCB
[image]

**Describe the solution you'd like**
I think it would be nice to be able to make similar types of documents (even without the reference designators), with some of the most important dimensions (width, length, pcb/component height on both sides). This might be possible with the FreeCAD API

**Describe alternatives you've considered**
Importing the STEP into FreeCAD or Fusion360 or any CAD and doing the 2D drawings there.

set-soft, 2024-11-22:
I don't know much about using FreeCAD, much less about its Python API.

nopeppermint, 2025-04-01:
@nguyen-v someone else had a similar idea a year ago:

https://forum.freecad.org/viewtopic.php?style=1&t=84427

https://forum.freecad.org/viewtopic.php?t=41983

https://wiki.freecad.org/Manual:Generating_2D_drawings

what you actually want is to automate the use of https://wiki.freecad.org/TechDraw_Workbench

this seem to be the related Python API: https://wiki.freecad.org/TechDraw_API