INTI-CMNB/KiBot #735: [HOW] Can't panelize (bad config?)

Unichord, 2024-11-26:
Hello, I'm having problems making a panel, I'm probably making some mistakes in the yaml/configuration (attached), this is the error I have, but I'm not sure as the error is not really clear.

```
DEBUG:Writing panel config to /tmp/kibot_panel_cfgg53ga3p_.json (kibot - gs.py:987)
DEBUG:- Executing: /usr/bin/kikit panelize --preset /tmp/kibot_panel_cfgg53ga3p_.json /mnt/c/Users/Andrea/Documents/Progetti/CLE/GFP1/GFP1.kicad_pcb /mnt/c/Users/Andrea/Documents/Progetti/CLE/GFP1/Production/Panel/GFP1-panel.kicad_pcb (kibot - kiplot.py:158)
ERROR:Trace stack: (kibot.gs - gs.py:836)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 168, in run_command
    res = _run_command(command, change_to)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 154, in _run_command
    return run(command, check=True, stdout=PIPE, stderr=STDOUT, cwd=change_to)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
ERROR:Running ['/usr/bin/kikit', 'panelize', '--preset', '/tmp/kibot_panel_cfgg53ga3p_.json', '/mnt/c/Users/Andrea/Documents/Progetti/CLE/GFP1/GFP1.kicad_pcb', '/mnt/c/Users/Andrea/Documents/Progetti/CLE/GFP1/Production/Panel/GFP1-panel.kicad_pcb'] returned -6 (kibot.gs - gs.py:853)
DEBUG:Copying viasonmask option (kibot - out_panelize.py:782)
DEBUG:- Reading the old PCB (kibot - out_panelize.py:784)
DEBUG:- Reading the new PCB (kibot - out_panelize.py:793)
DEBUG:- No need to change (both no) (kibot - out_panelize.py:802)
WARNING:(W139) Keeping temporal files: ['/tmp/kibot_panel_cfgg53ga3p_.json'] (kibot - out_panelize.py:863)
```

I'm using Windows and docker with kicad8_auto_full:latest image. Can you clear things up please? Thank you in advance.

[kibot_dbg.zip](https://github.com/user-attachments/files/17921705/kibot_dbg.zip)

Unichord, 2024-11-27:
By doing more tests, I solved it.
At first it was:
```
text:
          type: simple
```
Changed it to:

```
text1:
          type: simple
```
The rest of the configuration was ok.

Also, I noticed some of the examples in `panelize_all_examples.kibot.yaml` are in the form of the first one (so `text`), maybe that's an old file?

Unichord, 2024-11-27:
Update:
`text1:` is ok error-wise, but is not really working, as the text is not written on the panel.
However if I comment it out/remove it, the previous error shows up again. It's surely not critical, but it's kind of annoying.

At this point I'm not sure if it's a kibot or kikit problem. If more info is needed, I will provide it.

set-soft, 2024-11-27:
1. Please try to create a clean config file where the problem appears. Your setup is too complex, it depends on a lot of particular details, the way imports are done is really fragile. I'll suggest using symlinks in the file system (Windows have some equivalent) to make things much more simpler.
2. You must enable KiKit debug, by default KiKit just exits on errors, is something that I don't understand, but is the way the author implemented it. By default it just says that something went wrong, but not what.

set-soft, 2024-11-27:
Ok, I removed some of the extra includes that doesn't have anything to do with this issue and was able to reproduce your issue. KiKit is just aborting when you set *tl* anchor in the first text. Using other anchors works.

Note: if you used *text1* you sure got some warning.

Unichord, 2024-11-27:
Thank you. Yes that's the problem, even moving it around on text2 or text3 makes kikit crash, no matter the project or the text. I will report a bug in kikit.

By the way, yes text1 gave a warning, I just didn't notice.

set-soft, 2024-11-27:
I fixed the problem, the fix is in the KiKit 1.6.0-4 Debian package. 
Will try to generate docker images with it

set-soft, 2024-11-27:
Ok, I updated the KiCad 8 images and the *dev* images with a patched KiKit.
I also created a PR in the KiKit project: https://github.com/yaqwsx/KiKit/pull/768

closed 2024-11-27 by set-soft