INTI-CMNB/KiBot #736: [BUG] Filter does not apply on "re-used" properties

mdeweerd, 2024-11-27:
I guess this is somewhere between a bug and a feature.

I use the following filter for certain BOMs:
```yaml
filters:
  - name: only_lcsc_parts
    comment: Only parts with LCSC code
    type: generic
    include_only:
      - column: LCSC#
        regex: ^C\d+ 
```

However, this filter works on the components without using the reuseable information.

I have R11 and R15 and others that have 10k values.
However, R11 did not have the 'lcsc#' property field and was excluded from the BOM.

The typical use case is that when certain fields are empty, that they inherit their value from other components where this value is defined.

I am not sure what the best method is to cope with this as the field names are not all predetermined.

Prevention wise, I think a FieldCheck is appropriate to make sure all these fields are filled - either with a value number or with a "NULL" value.

set-soft, 2024-11-27:
Relying on inherited values from other components is a bad idea, you can use it, yes, but if it fails just don´t do this.

Yes, "Check Fields" can be used to verify issues.

The "Field Modify", "Spec to Field", "Field Rename" and "Expand Text Variables" can be used along with the KiCad Dialog that allows massive changes to fill your gaps.

Also: I usually create local libs containing the components I'll use in a project and use *these* components. So I don't use a generic R for a 10k resistor, I use a 10k resistor that contains all the information, including the manufacturer and P/N. KiCad can update the components from the lib, and in fact report if they got out of sync.

set-soft, 2024-11-28:
BTW: If you are using this feature, why did you choose a component that could be excluded to be the reference for the others? Just move the data to a component that you'll never exclude