INTI-CMNB/KiBot #737: [HOW/FEATURE] Generate outputs for all variants

Unichord, 2024-11-27:
Hello, what I'm trying to do is a reusable script to generate all variants for every needed output (for my use case BOM, board PDF, pos file are needed) of a single project.

The variants are defined in a folder of the project, like this:
```
variants:
- name: default
  type: kibom
- name: Q
  type: kibom
  file_id: Q
  variant: Q
```

Then I defined a field in Kicad like this:
[image]

But I can't find a way to have all variants generated in the same run, unless doing something convoluted like using `--list-variants` and iterating through them, changing variables as I go.

Is there something I'm missing or doing wrong? Could it be easily done by Kibot itself?

Thanks in advance.

Unichord, 2024-11-29:
I thought so, thanks for confirming.
Is this something on the roadmap that I can expect anytime soon? Would it be easy to implement in Kibot?

set-soft, 2024-11-29:
The problem is about side effects.

The above patch implements a mechanism, you can use *--variant ALL* to specify you want to iterate all available variants. And you can even use *--variant ALL --variant NONE* to get all variants plus the case where no variant is in use.

It passes a very simple regression test that used repeated outputs and was adapted to use this mechanism. Can you test it?

closed 2024-11-29 by set-soft

Unichord, 2024-12-02:
Surely I can test it, thank you! Do I find it on the docker image? I don't see any update on it, I did a pull on docker desktop. Is it correct?

set-soft, 2024-12-02:
Is in the docker images tagged as *dev* i.e. 

```
docker pull ghcr.io/inti-cmnb/kicad8_auto_full:dev
```

Unichord, 2024-12-02:
Ok, done, thank you. A few comments and questions:

- I tried with only BOM at the moment, it generates different files but the content is identical. It seems that the `config` field is ignored.
- The naming scheme is not suitable for our needs. We have variants like this: ABC1, ABC1D, ABC1E, and the first variants' outputs are named ABC1default.csv for example (this can be easily bypassed by a script later, it's no problem). I tried to set `file_id` and `variant` as "" but it doesn't work, and when setting `name` as "" or '' it says a valid name should exist.
- In my opinion, `name` should be a generic name for the board/variant, and `file_id` should be the actual information used for file naming and so on. Not a problem either way, though.
- Does the variant `type` influence the outputs? I used just `kibom` as type, but all kinds of outputs are generated (ibom, pdf, etc). This is exactly what I need, I'm asking just out of general curiosity.
- Kibot runs are significantly slower now, probably due to the dev branch usage. Just letting you know.

Thank you for support as usual.

set-soft, 2024-12-02:
> Ok, done, thank you. A few comments and questions:
> 
> * I tried with only BOM at the moment, it generates different files but the content is identical. It seems that the `config` field is ignored.

Can you create a small example with your test case.
What I tried is: [737.zip](https://github.com/user-attachments/files/17978823/737.zip)
As you can see the configuration is:

```yaml
kibot:
  version: 1

variants:
  - name: 't1_v1'
    comment: 'Test 1 Variant V1'
    type: kibom
    file_id: '_(V1)'
    variant: V1

  - name: 't1_v2'
    comment: 'Test 1 Variant V2'
    type: kibom
    file_id: '_(V2)'
    variant: V2

  - name: 't1_v3'
    comment: 'Test 1 Variant V3'
    type: kibom
    file_id: '_V3'
    variant: V3

  - name: 'bla bla'
    comment: 'Test 1 Variant V1+V3'
    type: kibom
    file_id: '_bla_bla'
    variant: ['V1', 'V3']

preflight:
  drc: true

outputs:
  - name: 'bom_internal'
    comment: "Bill of Materials in CSV format"
    type: bom
    dir: BoM

  - name: 'report'
    comment: "Report"
    type: report
```

I invoke KiBot like this:

```
kibot --variant NONE --variant ALL -d Gen
```

This is all variants plus the case with no variants.
And you get different reports and different BoMs. In fact this is equivalent to:

```yaml
kibot:
  version: 1

variants:
  - name: 't1_v1'
    comment: 'Test 1 Variant V1'
    type: kibom
    file_id: '_(V1)'
    variant: V1

  - name: 't1_v2'
    comment: 'Test 1 Variant V2'
    type: kibom
    file_id: '_(V2)'
    variant: V2

  - name: 't1_v3'
    comment: 'Test 1 Variant V3'
    type: kibom
    file_id: '_V3'
    variant: V3

  - name: 'bla bla'
    comment: 'Test 1 Variant V1+V3'
    type: kibom
    file_id: '_bla_bla'
    variant: ['V1', 'V3']

outputs:
  - name: 'bom_internal'
    comment: "Bill of Materials in CSV format"
    type: bom
    dir: BoM

  - name: 'bom_internal_v1'
    comment: "Bill of Materials in CSV format for variant t1_v1"
    type: bom
    dir: BoM
    options:
      variant: t1_v1

  - name: 'bom_internal_v2'
    comment: "Bill of Materials in CSV format for variant t1_v2"
    type: bom
    dir: BoM
    options:
      variant: t1_v2

  - name: 'bom_internal_v3'
    comment: "Bill of Materials in CSV format for variant t1_v3"
    type: bom
    dir: BoM
    options:
      variant: t1_v3

  - name: 'bom_internal_bla_bla'
    comment: "Bill of Materials in CSV format for variant `bla bla`"
    type: bom
    dir: BoM
    options:
      variant: 'bla bla'

  - name: 'report'
    comment: "Report"
    type: report

  - name: 'report_v1'
    comment: "Report for variant t1_v1"
    type: report
    options:
      variant: t1_v1

  - name: 'report_v2'
    comment: "Report for variant t1_v2"
    type: report
    options:
      variant: t1_v2
```

> * The naming scheme is not suitable for our needs.

Which name scheme? It should be the same.

> We have variants like this: ABC1, ABC1D, ABC1E, and the first variants' outputs are named ABC1default.csv for example (this can be easily bypassed by a script later, it's no problem). I tried to set [...]

I suspect there are some issue with the global output pattern. What are you using?

> * In my opinion, `name` should be a generic name for the board/variant, and `file_id` should be the actual information used for file naming and so on. Not a problem either way, though.

Can you elaborate?

> * Does the variant `type` influence the outputs? I used just `kibom` as type, but all kinds of outputs are generated (ibom, pdf, etc). This is exactly what I need, I'm asking just out of general curiosity.

We have 3 variant mechanisms, they can be applied to most outputs. You can use `kibom` or `kicost` or `ibom` variants for any output that supports variants. But each variant works in a different way. Please read the docs.

> * Kibot runs are significantly slower now, probably due to the dev branch usage. Just letting you know.

Is that for any operation or for some operation in particular?

Unichord, 2024-12-03:
> Can you create a small example with your test case. What I tried is: [737.zip](https://github.com/user-attachments/files/17978823/737.zip) As you can see the configuration is:

The configuration I use is basically the same as bug #735, I just changed the way I invoke Kibot:
`kibot -c ${script_dir}/bom.kibot.yaml -e "${proj_dir}/${project}.kicad_sch" -b "${proj_dir}/${project}.kicad_pcb" --variant ALL`

> I invoke KiBot like this:
> 
> ```
> kibot --variant NONE --variant ALL -d Gen
> ```
> 
> This is all variants plus the case with no variants. And you get different reports and different BoMs. In fact this is equivalent to:

Maybe we think differently the concept of variants, I will try to explain more clearly how I think it should work, please let me know if I'm wrong somewhere:
- A variant is a variation within the same scheme/board, i.e. multiple components which are not mounted, a different value for the same Ref, something like this.
- So, ABC1 (which is the project, and we associate it to the gerber files) could be ABC1 (which has R1=10K and C1 mounted) or ABC1D (which has R1=22K and C1 mounted) or ABC1E (which has R1=100K and C1=DNP)
- The gerber files are the same for all variants, while the BOM, the mount files (interactive html BOM for example), and the 3D image are different
Given this, I don't understand what you mean with "all variants plus the case with no variants". _no variants_ should be already included in the _all variants_ case, as it is a variant itself. Am I wrong here thinking this?

> Which name scheme? It should be the same.
The one I mentioned earlier, so ABC1, ABC1D, ABC1E...

> I suspect there are some issue with the global output pattern. What are you using?
Please refer to bug #735, the project I'm using is the same.

> > * In my opinion, `name` should be a generic name for the board/variant, and `file_id` should be the actual information used for file naming and so on. Not a problem either way, though.
> 
> Can you elaborate?
This is the variants file:
```
variants:
- name: 'default'
  file_id: 'T'
  variant: 'T'
  type: kibom
- name: Q
  type: kibom
  file_id: Q
  variant: Q
```
And this is the output folder:
[image]
The same applies if I set `file_id: ''` and `variant: ''`, the suffix _default_ stays.

> > * Kibot runs are significantly slower now, probably due to the dev branch usage. Just letting you know.
> 
> Is that for any operation or for some operation in particular?
Using the _latest_ branch for BOM generation requires about 7 seconds, while _dev_ branch requires about 13 seconds. No variants are involved of course.

set-soft, 2024-12-03:
> The configuration I use is basically the same as bug #735, I just changed the way I invoke Kibot: `kibot -c ${script_dir}/bom.kibot.yaml -e "${proj_dir}/${project}.kicad_sch" -b [...]

I understand you want to use templates that can be applied to any project, and this is the way to go.
But in order to find problems you must remove all the extra complexity, which could introduce side effects and complicates the tests.

If you don't remove it I must do it, which is time consuming and will discourage me.

I took the example from #735 and adapted *bom.kibot.yaml* to be used standalone (no extra complexity), I reduced it to:

```yaml
kibot:
  version: 1

variants:
- name: default
  comment: GFP1
  type: kibom
- name: A
  comment: GFP1A
  type: kibom
  file_id: A
  variant: A

outputs:
- name: netlist
  comment: Generating netlist for iBOM
  type: netlist
  dir: BoM

# CSV default BOM
- name: generic_bom_csv
  comment: Generic Bill of Materials in CSV format
  type: bom
  dir: BoM
  options:
    # use_alt: true
    output: '%f%v.%x'
    columns:
      - Row
      - Part
      - References
      - Value
      - Footprint
      - Quantity Per PCB
      - Status
      - Datasheet
    format: CSV

# Interactive HTML BOM
- name: basic_ibom
  comment: Interactive HTML BoM
  type: ibom
  dir: BoM
  options:
    output: '%f%v.%x'
    layer_view: 'F'
    checkboxes: 'Placed'
    show_fields: 'Value,Footprint'
    extra_data_file: BoM/%p-netlist.net
    extra_fields: 'Config'
    dnp_field: 'Config'
```

Note that your example also uses a missing *kicad_wks* file.

> Maybe we think differently the concept of variants, I will try to explain more clearly how I think it should work, please let me know if I'm wrong somewhere:
>
> * A variant is a variation within the [...]

Not wrong just omitting the fact that variants are completely optional. You can generate all the outputs without applying any variant. In most cases this is what you call *default* variant. If you don't use this then `--variant ALL` is enough. My example is more generic, that's all.
[... 97 lines cut ...]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:359)
WARNING:(W020) Malformed value: `ECH-U1C223GX5` (no match in C9) (kibot - units.py:239)
WARNING:(W020) Malformed value: `Coil` (no match in L3) (kibot - units.py:239)
- 'Interactive HTML BoM' (basic_ibom) [ibom]
Found 3 unique warning/s (3 total)

real	0m3,018s
user	0m2,673s
sys	0m0,439s
Using SCH file: GFP1.kicad_sch
Using command line value `Q` for global option `variant`
- 'Generating netlist for iBOM' (netlist) [netlist]
- 'Generic Bill of Materials in CSV format' (generic_bom_csv) [bom]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:359)
WARNING:(W020) Malformed value: `ECH-U1C223GX5` (no match in C9) (kibot - units.py:239)
WARNING:(W020) Malformed value: `Coil` (no match in L3) (kibot - units.py:239)
- 'Interactive HTML BoM' (basic_ibom) [ibom]
Found 3 unique warning/s (3 total)

real	0m3,001s
user	0m2,725s
sys	0m0,391s
```

So perhaps has something to do with docker and/or the OS

Unichord, 2024-12-03:
> I understand you want to use templates that can be applied to any project, and this is the way to go. But in order to find problems you must remove all the extra complexity, which could introduce [...]

You're absolutely right, I'm sorry. I will do it next time.

> Note that your example also uses a missing _kicad_wks_ file.

Yes, that's present in my PC. That's another thing I wanted to ask you, if it was possible to include all the Kicad paths (of a local installation) into the script so they're seen by the Docker image. Now I do it manually and it's fine.
 
> So I can just guess your _KCT1default.csv_ file comes from some mistake in the config, like using %V instead of %v
> 
> And this is why I insist on creating simple examples that reproduce the problem, not just saying "use this, but changing that and mixng with this"

It probably all boils down to this. I checked and there are no %V instead of %v, anywhere. Tomorrow I will try and reproduce the problem on a simpler example and check if it's still present.

> How do you measure it, is this part of the docker initialization?
```
start=`date +%s`
...kibot runs etc...
end=`date +%s`
echo runtime=$((end-start))
```
It's not meant to be precise, just to have a general idea of Kibot runtime, since I'm still setting up the script suite. This is run after I launch Docker, so that should not be involved. I run Docker with this script:

```
#!/bin/sh
export USER_ID=$(id -u)
export GROUP_ID=$(id -g)
export CURRENT_DIR=$(pwd)
docker run --rm -it \
    --user $USER_ID:$GROUP_ID \
    --env NO_AT_BRIDGE=1 \
    --env DISPLAY=$DISPLAY \
    --workdir=$CURRENT_DIR \
    --volume=/tmp/.X11-unix:/tmp/.X11-unix \
    --volume="/etc/group:/etc/group:ro" \
    --volume="/home/$USER:/home/$USER:rw" \
    --volume="/etc/passwd:/etc/passwd:ro" \
    --volume="/etc/shadow:/etc/shadow:ro" \
    --volume="/home/$USER:/home/$USER:rw" \
    --volume="/mnt/c/:/mnt/c/:rw" \
    ghcr.io/inti-cmnb/kicad8_auto_full:latest /bin/bash /mnt/c/Users/Andrea/Documenti/Script/kibot/run_kibot.sh
```

> On my system I generate netlist, bom and ibom for both variants in 4 s:
> So perhaps has something to do with docker and/or the OS

It could be, honestly, I'm running Docker Desktop on Windows and WSL2 but I don't know if I made everything correctly. I will check, thank you.

Unichord, 2024-12-05:
Hello, I tried to simplify the structure, just for BOM generation:
[bug737.zip](https://github.com/user-attachments/files/18022501/bug737.zip)

This is the output folder:
[image]

set-soft, 2024-12-05:
The ZIP file doesn't contain the PCB/SCH, and the configs mention ./Doc/Settings/variants.kibot.yaml which is also missing.

Unichord, 2024-12-06:
Here is the full project.
[KCT1.zip](https://github.com/user-attachments/files/18034132/KCT1.zip)

I'm doing some more tests now, I'll keep you updated.

set-soft, 2024-12-06:
Looking at this example you are using a *kibom* output, and the docs says:

[image]

So here the KiBot variants doesn't apply, this output is just for people that had a KiBoM setup working before using KiBot and doesn't have the time to migrate.

Unichord, 2024-12-09:
Oh, I'm using _kibom_ because it's the only one working, though.
With _bom_ in both `bom.kibot.yaml` and `variants.kibot.yaml` I get this error, with _dev_ branch:

```
ERROR:Trace stack: (kibot.gs - gs.py:870)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 1048, in load_config
    outputs = cr.read(cf_file)
              ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 710, in read
    self._parse_import(v1, fstream.name, collected_definitions)
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 579, in _parse_import
    imported = self._parse_import(data['import'], fn, collected_definitions, apply=False, depth=depth)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 593, in _parse_import
    all_collected.variants.update(self._parse_import_variants(vars, explicit_vars, fn_rel, data, imported,
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 423, in _parse_import_variants
    imported.variants.update(self._parse_variants(data['variants'], is_internal))
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 248, in _parse_variants
    o_var = self._parse_variant_or_filter(o, 'variant', RegVariant, is_internal)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/config_reader.py", line 232, in _parse_variant_or_filter
    raise KiPlotConfigurationError("Unknown {} type: `{}`".format(kind, otype))
ERROR:Unknown variant type: `bom` (kibot.gs - gs.py:884)
```

If I use _bom_ in `variants.kibot.yaml` and _kibom_ in `bom.kibot.yaml` the error persists.

If I use _bom_ in `bom.kibot.yaml` and _kibom_ in `variants.kibot.yaml` the error disappears, but the output BOMs of the variants are identical.

With _latest_ branch, not using variants, the error of course doesn't occur. What am I doing wrong? :(

I used the project I sent on last comment.

set-soft, 2024-12-09:
> Oh, I'm using _kibom_ because it's the only one working, though. With _bom_ in both `bom.kibot.yaml` and `variants.kibot.yaml` I get this error, with _dev_ branch:
>
> ```
> ERROR:Trace stack: (kibot.gs [...]

You are messing the concepts. One thing is a *kibom* variant and other is *kibom* output.

The *kibom* variant is the mechanism used by *kibom*  for variants, and shouldn't be applied to a *kibom* output. This is because KiBoM implements it internally. 

The above error clearly states that there is no variant named *bom*. This is the name of an output.

> If I use _bom_ in `bom.kibot.yaml` and _kibom_ in `variants.kibot.yaml` the error disappears, but the output BOMs of the variants are identical.

Which documented. The filters and variants doesn't apply to KiBoM output. This is why you should us *bom* output.

> With _latest_ branch, not using variants, the error of course doesn't occur. What am I doing wrong? :(

Mixing the concepts. KiBoM output is just for KiBoM users that didn't have the time to migrate to the internal BoM output.

Unichord, 2024-12-14:
> You are messing the concepts. One thing is a _kibom_ variant and other is _kibom_ output.
> 
> The _kibom_ variant is the mechanism used by _kibom_ for variants, and shouldn't be applied to a _kibom_ output. This is because KiBoM implements it internally.

I see, thank you for clarifying. This was not clear at all to me, sorry for mixing different concepts, and this was probably the root cause of my problem.

Thank you for your continued support, I probably solved the problem altogether. Sorry I'm a bit weak on yaml and configurations, the docs are well written but are huge and some concepts probably slipped or I misread.

To sum up, I tested some configurations I have and the switch `--variant ALL` works fine, as far as I know. Thank you again.