INTI-CMNB/KiBot #744: [HOW/BUG] Crosses for dnp flagged parts on Fab layers never drawn

burbschat, 2024-12-17:
**What do you want to achieve?**
Crosses on dnp flagged components on the Fab layers. All the documentation indicates that this should work by default.
Not sure if I am doing something wrong or if this is a bug.

What I expect to happen:
Set dnp flag in kicad -> cross on component on the Fab layer in pdf output.

What happens: 
No cross drawn at all.

What I found (likely that this isn't the root cause of the issue though, but perhaps related):
Check in `out_base.py`, `load_list_components()` causes the function to immediately return, which results in components never being loaded from the schematic. For reference the function is:
```py
    def load_list_components(self):
        """ Makes the list of components available """
        self._files_to_remove = []
        # The following check always seems to cause the function to return (all checked values are None):
        if not self.dnf_filter and not self.variant and not self.pre_transform:
            return
        load_sch()
        # Get the components list from the schematic
        comps = GS.sch.get_components()
        get_board_comps_data(comps)
        # Apply the filter
        reset_filters(comps)
        comps = apply_pre_transform(comps, self.pre_transform)
        apply_fitted_filter(comps, self.dnf_filter)
        # Apply the variant
        if self.variant:
            # Apply the variant
            comps = self.variant.filter(comps)
            self._sub_pcb = self.variant._sub_pcb
        self._comps = comps
```

**Do you have some PCB/Schematic to use as example?**
[kibot_minimal_test.zip](https://github.com/user-attachments/files/18163502/kibot_minimal_test.zip)

**Do you have some configuration file (.kibot.yaml) that you are using?**
Only configuration used is the one included with the example

**Environment (please complete the following information):**
Where are you running KiBot:
Run in fresh virtual environment created with `python -m venv --system-site-packages kibot_venv`.
Then kibot is installed with `pip install --no-compile kibot`.
I also had to downgrade lark to `1.1.9` as the newest version apparently breaks something (probably unrelated to this question though).
Host OS is Arch Linux. Kicad is the newest version (8.0.6-1).
`kibot-check` does not indicate any critical missing dependencies.

set-soft, 2024-12-17:
> Check in `out_base.py`, `load_list_components()` causes the function to immediately return, which results in components never being loaded from the schematic. For reference the function is:

Yes, this is the intended behavior. As you can see the code looks for a filter and/or variant. As mentioned in the docs the filters and variants aren't always enabled, the cross is the result of a filter, so you need to trigger the variants and filters mechanism.

In you example just adding:

```diff
--- config.kibot.yaml.ori	2024-12-17 06:48:33.000000000 -0300
+++ config.kibot.yaml	2024-12-17 07:32:33.935776959 -0300
@@ -11,6 +11,7 @@
   type: pcb_print
   dir: pdf/pcb
   options:
+    pre_transform: _none
     format: PDF
     pages:
     - layers:
```

Will do the work. If you want to enable the mechanism for all the outputs you can just use:

```diff
--- config.kibot.yaml.ori	2024-12-17 06:48:33.000000000 -0300
+++ config.kibot.yaml	2024-12-17 07:38:35.537738580 -0300
@@ -5,6 +5,13 @@
 #   cross_footprints_for_dnp: true
 #   cross_using_kicad: true
 
+globals:
+  variant: default
+
+variants:
+- name: default
+  type: ibom
+
 outputs:
 - name: basic_pcb_print_pdf
   comment: PCB Layers PDF
```

Both will produce:

[image]

closed 2024-12-17 by set-soft

burbschat, 2024-12-17:
Alright that makes sense. I actually tried defining a variant but missed that I have to set a global variable to select it.
Thank you for the quick response 😊