INTI-CMNB/KiBot #750: [HOW] Fix oversized image in BOM

Kampi, 2024-12-24:
**What do you want to achieve?**

My BOM (generated by type `bom`) contains an oversized image. How can I remove / fix it?

[image]

**Do you have some configuration file (.kibot.yaml) that you are using?**

```yaml
  - name: "BOM for PCBWay"
    comment: "BOM for PCBWay"
    type: bom
    dir: docs
    category: "purchasing"
    options:
      format: XLSX
      count_smd_tht: true
      output: "BOM_PCBWay.%x"
      ref_separator: ','
      columns:
        - field: Row
          name: Item
        - field: References
          name: Ref Des
        - field: Build Quantity
          name: Quantity
        - field: Mfr.
          name: Manufacturer
        - field: Mfr. No.
          name: Mfg Part#
        - field: Value
          name: Description / Value
        - field: Footprint
          name: Package
        - field: Footprint Type
          name: Type
        - field: Footprint Side
          name: Side
      xlsx:
        title: "%f BOM"
        hide_stats_info: false
        quote_all: true
```

**Environment (please complete the following information):**

- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`)

**Additional context**
Add any other context about your setup/project.

set-soft, 2024-12-26:
Hi @Kampi !
This is an error in the scale, I changed the logo but didn't notice the new PNG has a different PPI scale.
You can just use `scale_logo: 0.5` in the *xlsx* section or provide your own logo using the `logo` option (also in the *xlsx* section).

set-soft, 2024-12-26:
The above patch should fix it.

closed 2024-12-26 by set-soft

Kampi, 2024-12-28:
Thank you! Will you update the Docker container too?

set-soft, 2024-12-30:
> Thank you! Will you update the Docker container too?

*dev* images are updated. The main images are updated only for releases.

commit 9d8dd726 (2024-12-26): [BoM][XLSX][Fixed] Logo size

To match the old PPIs

Fixes #750