INTI-CMNB/KiBot #754: [HOW] Fix missing KiCost entries

Kampi, 2024-12-30:
**What do you want to achieve?**

KiCost output in BOM files is not complete but part numbers are complete.

[image]

**Do you have some PCB/Schematic to use as example?**

Please use the `Add_KiCost` branch of this repo: 
Reference: https://github.com/Kampi/ZSWatch-HW/actions/runs/12552235169

**Do you have some configuration file (.kibot.yaml) that you are using?**

All in the repo

**Environment (please complete the following information):**

- GitHub actions (CI/CD)

set-soft, 2025-01-02:
If KiCost can't find a part just add the distributor P/N.

Kampi, 2025-01-02:
I don´t understand the issue. Take `IC307` (RV-8263C8) for example. It has the Mouser part number `428-204291-MC01` but KiBot says this

```
Progress:  38%|███▊      | 17/45 [00:06<00:11,  2.46part/s]
Progress:  40%|████      | 18/45 [00:06<00:10,  2.47part/s]
                                                           
DEBUG:P/N: 428-204291-MC01 (kibot.kicost.dist - api_mouser.py:318)

Progress:  40%|████      | 18/45 [00:06<00:10,  2.47part/s]
                                                           
DEBUG:Looking in cache /github/home/.cache/kicost/Mouser/mou_428-204291-MC01.dat (kibot.kicost.dist - distributor.py:92)

Progress:  40%|████      | 18/45 [00:06<00:10,  2.47part/s]
                                                           
WARNING:(WC007) No information found at Mouser for part/s '['IC307']' (kibot.kicost.dist - api_mouser.py:329)
```

And it´s also not listed in the BOM:

[image]

set-soft, 2025-01-02:
This is different to what the first image shows, where IC307 has a price.
You should take a look at the full log, here I don't see the result. 
Note that this depends on internet connection and Mouser servers, so you could get fails for many reasons. 
I think you are not implementing a cache ... are you sure you didn't hit the maximum number of queries?

closed 2025-01-02 by Kampi