INTI-CMNB/KiBot #759: [BUG] Zones not refilled before applying KiCAD 8 DRC

richardbarlow, 2025-01-02:
**Describe the bug**

I am using `set_text_variables` to set some text in the copper of the board. With the previous `run_drc` preflight this worked OK. I am in the process of updating my CI pipeline to KiCAD 8 and have changed the preflight to the new `drc`. However, this no longer seems to work, as the copper zone is not being refilled after setting the text, but before running the DRC.

**To Reproduce**
See minimum working example here:

[text_fill_mwe.zip](https://github.com/user-attachments/files/18292855/text_fill_mwe.zip)

```
> kibot -b text_fill_mwe/text_fill_mwe.kicad_pcb -c text_fill_mwe/kibot_generated.kibot.yaml
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:359)
WARNING:(W010) Unable to find KiCad libraries (kibot - config.py:434)
WARNING:(W010) Unable to find KiCad templates (kibot - config.py:434)
WARNING:(W010) Unable to find KiCad footprints (kibot - config.py:434)
WARNING:(W010) Unable to find KiCad 3D models (kibot - config.py:434)
- Running the DRC
ERROR:1 DRC errors detected (kibot.pre_drc - pre_drc.py:147)
ERROR:(clearance) Clearance violation (zone clearance 0.5000 mm; actual 0.0000 mm)
    @(86.5 mm, 73.2 mm): PCB Text '${test}' on F.Cu
    @(74.9 mm, 56.1 mm): Zone [<no net>] on F.Cu
    Check: Violations (kibot.pre_drc - pre_drc.py:162)
ERROR:DRC errors: 1 (kibot.gs - gs.py:845)
```

I tried setting `check_zone_fills: true` too and it has no effect.

**Expected behavior**
I expect the zones to be refilled after setting text and before running the DRC.

**Environment (please complete the following information):**

In CI: ghcr.io/inti-cmnb/kicad8_auto_full:1.8.1 in GitHub
Locally: Fedora 41 with KiBot 1.8.1 and KiCAD 8.0.7

set-soft, 2025-01-03:
Hi @richardbarlow !

Now `check_zone_fills` will do the work. You can also use `fill_zones` to get a persistent result.
This is available in the *dev* branch and docker images.

closed 2025-01-03 by set-soft

richardbarlow, 2025-01-03:
Thanks for the fast fix @set-soft. It appears to be working perfectly :smile:

commit 754b9bb4 (2025-01-03): [Check Zone Fills][Fill Zones][Fixed] DRC issues

- Check Zone Fills: wasn't applied to the new DRC
- Fill Zones: was applied before set_text_variables, which could need a refill

Fixes #759