INTI-CMNB/KiBot #767: Fill technicall information table when exporting using PCB_PRINT

houkhouk, 2025-01-09:
Hi guys, 
having some trouble specifying footer sheet information when exporting my layers to pdf using pcb_print.  

**What do you want to achieve?**
I'm trying to properly fill  the technical sheet information footer (title, revision, date, etc) when using 'pcb_print' export. However, I cannot find a .yaml configuration that updates theses info. What am I missing ?  

**Do you have some PCB/Schematic to use as example?**
Unfortunately I am not at liberty to disclose this information

**Do you have some configuration file (.kibot.yaml) that you are using?**
Here is my current config
  - name: 'PCB artwork'
    comment: 'Drawing of PCB layers'
    type: 'pcb_print'
    dir: 'Project_Outputs/Documentation'
    options:
      dpi: 800
      frame_plot_mechanism: 'internal'
      pages:
          - layers:
              - layer: 'F.Cu'
              - layer: 'Edge.Cuts'
              - layer: 'F.Silkscreen'
                color: '#FFFF00'
            title: 'Top copper layer and silkscreen'
            scaling: 1.3
            page_id: '%02d'
          - layers:
              - layer: 'F.Mask'
              - layer: 'Edge.Cuts'
            title: 'Top solder mask'
            scaling: 1.3
          - layers:
              - layer: 'F.Fab'
              - layer: 'Edge.Cuts'
            title: 'Top fab layer'
          - layers:
              - layer: 'In1.Cu'
              - layer: 'Edge.Cuts'
            scaling: 1.3
            title: 'In1 copper layer '
          - layers:
              - layer: 'In2.Cu'
              - layer: 'Edge.Cuts'
            scaling: 1.3
            title: 'In2 copper layer '
          - layers:
              - layer: 'In3.Cu'
              - layer: 'Edge.Cuts'
            scaling: 1.3
            title: 'In3 copper layer '
          - layers:
              - layer: 'In4.Cu'
              - layer: 'Edge.Cuts'
            scaling: 1.3
            title: 'In4 copper layer '        
          - layers:
              - layer: 'B.Cu'
              - layer: 'Edge.Cuts'
              - layer: 'B.Silkscreen'
                color: '#FFFF00'
            scaling: 1.3
            mirror: true
            title: 'Bottom copper layer and silkscreen'
          - layers:
              - layer: 'B.Mask'
              - layer: 'Edge.Cuts'
            scaling: 1.3
            mirror: true
            title: 'Bottom solder mask'
          - layers:
              - layer: 'B.Fab'
              - layer: 'Edge.Cuts'
            mirror: true
            title: 'Bottom fab layer'

**Environment (please complete the following information):**
I am running kibot on a desktop machine:
Application: KiCad x86_64 on x86_64
Version: 8.0.7-8.0.7-0~ubuntu22.04.1, release build
Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.18
Platform: Ubuntu 22.04.5 LTS, 64 bit, Little endian, wxGTK, X11, budgie-desktop, x11
OpenGL: Intel, Mesa Intel(R) Xe Graphics (TGL GT2), 4.6 (Compatibility Profile) Mesa 24.0.0-devel (git-3ca1f35cbf)
Build Info:
	Date: Dec  3 2024 15:56:08
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.81.0
	ngspice: 43
	Compiler: GCC 11.4.0 with C++ ABI 1016

**Additional context**
I don't recall ever managing to make this work, even on previous versions of kicad/kibot

set-soft, 2025-01-09:
Usually you use variables in your *worksheet*, i.e. you assign `${REVISION}` to the box for the revision. 

Then you use `set_text_variables` preflight to update the variable, i.e. assign the current commit hash to the *REVISION* variable.

In this way you can change any detail of the *worksheet*, and also text in the PCB layers.

Is that what you need?

houkhouk, 2025-01-09:
Thank you for your fast response. This is what I tried, however no matter how I set it, the fields remain blank:
[image]

set-soft, 2025-01-09:
As your project can't be disclosed I suggest creating a very simple test case, so you can detect what's going wrong.
This mechanism is used in various projects and examples.
Then you can post it here.

houkhouk, 2025-01-09:
I have created a very simple board where I reproduced the problem. This "test project" inherits all kicad configuration used in the real one.
Revision is set as a text variable in the kicad project, titles for each page of the pdf created by pcb_print is set in the .yaml file.

The generated file I have problem with is located in Test/Project_Outputs/Documentation/test-assembly.pdf

[Test.zip](https://github.com/user-attachments/files/18360734/Test.zip)

set-soft, 2025-01-09:
Ok, a bug here. You can workaround it by providing your own worksheet, i.e. 
[default_wks.zip](https://github.com/user-attachments/files/18361482/default_wks.zip)

houkhouk, 2025-01-09:
Great thank you very much, I can work with this !

Keep up this amazing work :)

set-soft, 2025-01-09:
And now the *dev* images should contain the fix.

commit 4180a9af (2025-01-09): [PCB Print][Fixed] Problems when using the default worksheet

- And the internal frame plotter
- Variables not expanded

Fixes #767
closed 2025-01-09 by set-soft