INTI-CMNB/KiBot #769: [HOW/FEATURE] Specify order of preflights

nguyen-v, 2025-01-19:
**What do you want to achieve?**
In my workflow, I use the xml file (generated by the `update_xml` preflight to get information about the schematic layout (pages and so on). At the same time, I use the `set_text_variables` preflight to dynamically update a table of contents in the schematic, using a custom script in the `command` field. The issue is that I need the XML file to be generated before the `set_text_variable` preflight is called. Is there a way to specify the execution order of preflights?

**Environment (please complete the following information):**
Where are you running KiBot:
- WSL2, dev branch

set-soft, 2025-01-21:
Nope, there is no mechanism. And the use for the XML file isn't the objective of *update_xml*.
I think this is a candidate for a multipass use.
*set_text_variables* can change the XML.
BTW: Perhaps we need to make available the information you are extracting from the XML

nguyen-v, 2025-01-21:
In my case I use the fields
```
<sheet number="num" name="sheet_path" tstamps="tstamp">
```

I have a python script that extracts the sheetpath for a given num, and use this to change a text variable called ${SHEET_NAME_num}. I guess it could be useful to make available this information, but I feel that my usecase is very uncommon though

nguyen-v, 2025-01-21:
> I think this is a candidate for a multipass use.
For multi-pass, is there a way to run the `update_xml` with no specified output(s)? So only the preflight

set-soft, 2025-02-10:
> > I think this is a candidate for a multipass use.
> 
> For multi-pass, is there a way to run the `update_xml` with no specified output(s)? So only the preflight

Yes! you can use `-i` with no outputs specified (the default is all, so the inversion is none)

P.S. sorry for the delay

closed 2025-02-18 by nguyen-v