INTI-CMNB/KiBot #772: [BUG] Populate: some components are always shown, even if not explicitly mentionned

dhaillant, 2025-01-30:
Fill as much as you can, remove the rest

**Describe the bug**
When using Populate, some components are always shown, regardless of the instructions of the .md populate file.

This bug may have something to do with attributes: I had one component marked with both attributes "Exclude from position files" AND "Exclude from bill of materials" that was showing all the time. By removing these attributes, I've been able to make it disappear from 3D view (and make it appears later in the populate doc, per explicit instructions).

I've also been able to reproduce the bug and make other components appear regardless of the instructions of populate.md by checking the attributes mentioned above.

However, I have other components that do not follow this simple workaround, and so far, I haven't been able to remove them from the 3D views of the populate doc. So maybe there are other interactions happening.

**To Reproduce**
Open pcbnew, choose a component, open its footprint properties and check both attributes "Exclude from position files" and "Exclude from bill of materials". Save the pcb file.

Then start populate:
From Makefile:
`kibot -d $(DIR) -c $(KIBOT_CFG) -b $(PCB) -g variant=sub_pcb[main] populate`

From YAML:
```
  - name: populate
    comment: "Populate front PCB"
    type: populate
    dir: Populate/%f-%r%v-%D/top
    options:
      renderer: top_3d
      input: kibot/populate.md

  - name: top_3d
    comment: Top view render
    type: render_3d
    dir: PCB%v/png
    options:
      output: '%p-%r-%i%I%v-%D.%x'
      width: 1280
      height: 960
      orthographic: true
```

From populate.md:
```
# Assembly procedure
- [[front | ]] This is the front side of the board we are populating.
- [[back | ]] This is the back side of the board we are populating.
(...)
```

From the very first steps, even if no components are explicitly mentioned, the 3D models of components with both attributes checked are shown in the 3D views.

**Expected behavior**
The components that aren't explicitly listed in populate.md shouldn't be shown.

**Environment (please complete the following information):**
- KiBot 1.8.2 - Copyright 2018-2024, Salvador E. Tropea/INTI/John Beard - License: AGPL v3
- KiCad PCB Editor x86_64 on x86_64 Version: 8.0.8-8.0.8-0~ubuntu22.04.1, release build

set-soft, 2025-01-30:
Please provide a very small example PCB where the problem is reproduced.
Also explain which component is wrong.

dhaillant, 2025-01-30:
[populate_components_always_showing.zip](https://github.com/user-attachments/files/18605288/populate_components_always_showing.zip)

`make populate_iso`

set-soft, 2025-02-04:
Hi @dhaillant !

I don't see a simple example in the ZIP file and I don't see the explanation of what is wrong.
I don't have time these days so please help if you want a solution.

dhaillant, 2025-02-05:
Hi @set-soft
Sorry for the impractical demo I provided last time. (Explanations where written in User.Comments layer.)

Here is a fresh, very simple demo: 3 resistors, and only one is demonstrating the bug.

The bug only happens when kibot is called with `-g variant=` option:

```kibot -d $(DIR) -c $(KIBOT_CFG) -b $(PCB) -g variant=sub_pcb[main] populate_iso```

and with options
 - `Exclude from position files`
 - `Exclude from bill of materials`
_both checked_ in "Footprint Properties".

R1 has only `Exclude from position files` checked and is ok.
R2 has only `Exclude from bill of materials` checked and is ok.

However, R3 has **both options checked** and is showing right at the first image generated by `populate`, even if _not mentionned_ in `populate.md`

The results can be seen in dir `demo/kibot/Populate/img`
Image `populating_1.png` shows R3 already present.

You can reproduce the bug by calling `make` from `demo` folder.

Edit `Makefile` and comment/uncomment lines 16 and 17 to see the differences.

---

Here are the outputs from the commands. First without `-g variant=`:
```
$ make
kibot -d kibot -c kibot/generic.kibot.yaml -b demo.kicad_pcb populate_iso
- 'Step by step Populate PCB' (populate_iso) [populate]
WARNING:(W143) This output depends on KiCad version, use `blender_export` instead (kibot - out_render_3d.py:307)
WARNING:(W135) Sim.Params field mismatch for `R1` (SCH: `r=${VALUE}` PCB: `r=R`) (kibot - kiplot.py:348)
WARNING:(W135) Sim.Params field mismatch for `R2` (SCH: `r=${VALUE}` PCB: `r=R`) (kibot - kiplot.py:348)
WARNING:(W135) Sim.Params field mismatch for `R3` (SCH: `r=${VALUE}` PCB: `r=R`) (kibot - kiplot.py:348)
WARNING:(W047) Missing 3D model for R1: `/home/david/tests/populate_components_always_showing2/demo/kicad-packages3D/Resistor_THT.3dshapes/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal.wrl` (kibot - out_base_3d.py:514)
WARNING:(W047) Missing 3D model for R2: `/home/david/tests/populate_components_always_showing2/demo/kicad-packages3D/Resistor_THT.3dshapes/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal.wrl` (kibot - out_base_3d.py:514)
Found 6 unique warning/s (16 total)
```

And this is the output with `-g variant=`:
```
$ make
kibot -d kibot -c kibot/generic.kibot.yaml -b demo.kicad_pcb -g variant=sub_pcb[main] populate_iso
Using command line value `sub_pcb[main]` for global option `variant`
- 'Step by step Populate PCB' (populate_iso) [populate]
WARNING:(W135) Sim.Params field mismatch for `R1` (SCH: `r=${VALUE}` PCB: `r=R`) (kibot - kiplot.py:348)
WARNING:(W135) Sim.Params field mismatch for `R2` (SCH: `r=${VALUE}` PCB: `r=R`) (kibot - kiplot.py:348)
WARNING:(W135) Sim.Params field mismatch for `R3` (SCH: `r=${VALUE}` PCB: `r=R`) (kibot - kiplot.py:348)
WARNING:(W143) This output depends on KiCad version, use `blender_export` instead (kibot - out_render_3d.py:307)
WARNING:(W047) Missing 3D model for R3: `/home/david/tests/populate_components_always_showing2/demo/kicad-packages3D/Resistor_THT.3dshapes/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal.wrl` (kibot - out_base_3d.py:514)
WARNING:(W047) Missing 3D model for R1: `/home/david/tests/populate_components_always_showing2/demo/kicad-packages3D/Resistor_THT.3dshapes/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal.wrl` (kibot - out_base_3d.py:514)
WARNING:(W047) Missing 3D model for R2: `/home/david/tests/populate_components_always_showing2/demo/kicad-packages3D/Resistor_THT.3dshapes/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal.wrl` (kibot - out_base_3d.py:514)
Found 7 unique warning/s (22 total)
```

---

[populate_components_always_showing2.zip](https://github.com/user-attachments/files/18669009/populate_components_always_showing2.zip)

(Also, images are cropped weirdly when using `-g variant=` ??)

Thank you so much for your time and your awesome project!
David

set-soft, 2025-02-05:
Ok, I found the problem. 
This is some legacy issue. KiCad 6 removed the concept of virtual components and marked them as not for BoM and Pos.
These components are marked as excluded by the *_mechanical* filter.
According to the docs components marked as *excluded* are excluded from the BoM processing.
Variants were originally created for the BoM, but with time they also applied to things like 3D models.
So the code excluded components marked as *excluded* from various tasks, like the 3D models removal.
R3 was detected as a "virtual" component and hence excluded from processing.
This doesn't have much sense, so I changed the code. The docs doesn't mention excluding them from other tasks.
I hope it doesn't have nasty side effects.

closed 2025-02-10 by set-soft