INTI-CMNB/KiBot #774: [Feature] Detect extra fields

set-soft, 2025-02-04:
### Discussed in https://github.com/INTI-CMNB/KiBot/discussions/773

<div type='discussions-op-text'>

<sup>Originally posted by **Strooom** February  4, 2025</sup>
I am using KiBot in a CI/CD github workflow, and it's really good so I am applying it to all my designs.
One thing is still missing however : I'd like some automated check and/or cleanup of the custom fields used in the project, its symbols and its footprints.

Often a needed symbol and/or footprint is not present in the standard KiCAD libraries but it is made available by its vendor or by sites such as SnapEDA, SamacSys, etc. These symbols and footprints however contain custom fields of metadata which are polluting the design and they may show up in BOM exports etc.

So in general I will clean an imported symbol / footprint before using it in a design, aligning all metadata field names to my standard and removing excess fields. But of course errors can and do occur, and so I'd like to automate this check with KiBOT

**output field_check**
The user provides a list of 'allowed' field names in the YAML file, and KiBOT writes a report file with all symbols and footprints that have fields outside this list., eg. `symbol 'CP2102N' has field 'height'`
User could also provide a list of 'mandatory' field names and KiBOT writes in the report for which symbols / footprints these fields are missing or empty, eg. `symbol 'CP2102N' has empty field 'Datasheet'`
Ideally, the lists of allowed and mandatory fields could be different for symbols, footprints, the schematic and the layout

**This will help a lot with the cleanup of your design's metadata.**
If this would be possible with the current features of KiBOT, I would be pleased to learn how to do that.</div>

set-soft, 2025-02-04:
Hi @Strooom !

The above patch implements a mechanism to report extra fields, i.e.:

```yaml
kibot:
  version: 1

preflight:
  check_fields:
    - field: MFR
      regex: '.*'
      severity: warning
      severity_missing: continue
    - field: MPN
      regex: '.*'
      severity: warning
      severity_missing: continue
    - field: "LCSC#"
      regex: '.*'
      severity: warning
      severity_missing: continue
    - field: "*"
      severity_missing: warning
```

Can report things like:

```
WARNING:(W162) D5 extra fields: bogus (kibot - pre_check_fields.py:109)
```

closed 2025-02-04 by set-soft

Strooom, 2025-02-04:
Ok, that's really fast!

So in the example, the last field with "*" wildcard name catches all other not already listed fields ?
And there will be a warning when something is present ? In that case severity_missing is maybe confusing...

set-soft, 2025-02-04:
> I did a few tests and this seems to work. However, the matching of field names is not case sensitive. It would be better to be case sensitive as this results in a cleaner set of metadata.

All field name comparisons in KiBot are not case sensitive. And is the same for other tools like KiCost.
There is no point in doing case sensitive detection here when the BoM (and all other outputs) will be not sensitive.

>  I think in KiCAD the field names are also case sensitive.

This is true for modern versions.

I understand that is better to report the real name, so I added a translation.

Strooom, 2025-02-05:
Ok, I understand.

As modern versions of KiCAD treat the field names case sensitive, I think at some point KiBOT will have to follow.
More and more PCB manufacturers and PCBA providers are accepting the kicad_pcb file as input, and are extracting BOM data directly from this file. So that's the reason we have to get these field names and values clean.
I am using [aisler.net](https://aisler.net/) for my prototyping and production.

Again, thanks for the quick fix, it's already a big step forward.

set-soft, 2025-02-05:
> So that's the reason we have to get these field names and values clean.

I agree, but I don't see why case sensitive fields is an issue. 
When more than one person works on the same project errors like `Voltage` vs `voltage` are quite frequent. Making KiBot to handle them as different isn't good.

You can use the `field_rename` filter to rename `voltage` to `Voltage`

Strooom, 2025-02-05:
Ok, but what happens when you have both a 'mpn' and a 'MPN' field, with **different values** and you rename them to your standard capitalization ? 

The solution I see is to only allow one field with (eg) camelCase naming, and have the field_check report all other / incorrect case variants. It is then up to the designer to correct the fields and make sure the correct content is retained.
For that to work, the check has to be case-sensitive.

set-soft, 2025-02-05:
> Ok, but what happens when you have both a 'mpn' and a 'MPN' field, with **different values** and you rename them to your standard capitalization ?

KiCad doesn't allow it.

[image]

Strooom, 2025-02-05:
If I add a field 'MPN' to one symbol, and 'mpn' to another, then this is how it looks when I open the Symbol Fields Table on the schematic...

[image]

There I can edit both fields for any symbol, so I may end up with this...

[image]

Going back to the symbol, both MPN and mpn are present...

[image]

This is not something you would do intentionally, but it's the kind of mess that you import when importing symbols from vendors and 3rd party sites such as SnapEDA, SamacSys etc.

set-soft, 2025-02-07:
> This is not something you would do intentionally, but it's the kind of mess that you import when importing symbols from vendors and 3rd party sites such as SnapEDA, SamacSys etc.

Looks like a bug in KiCad. It should avoid it as with the regular dialog.
I think this particular case should be reported by KiBot as a warning.

set-soft, 2025-02-14:
Note that KiBot already reports a warning when generating a BoM and things like `mnp` vs `MNP` has a different value.
I'm adding a message to the loading function with better information.

commit 75bef495 (2025-02-04): [Check Fields][Added] Mechanism to check for extra fields

Closes #774