INTI-CMNB/KiBot #776: [BUG] ERROR:Missing page layout file (kibot.gs - gs.py:850)

stefan894, 2025-02-05:
**Describe the bug**
I am using kibot to create an assembly plan inside a gitlab runner with:

```
kibot --log Generated/kibot.log -c V3/Assembly_DNP.kibot.yaml -b V3/Kicad/xx_PCB_PCB.03x.kicad_pcb
```

this works fine with the following docker containers:
- ghcr.io/inti-cmnb/kicad8_auto:1.8.1
- ghcr.io/inti-cmnb/kicad_auto:dev_k8

but fails with:
- ghcr.io/inti-cmnb/kicad8_auto:1.8.2 with error:

```
ERROR:Missing page layout file: /kicad_template/xxx_A3_pcb.kicad_wks (kibot.gs - gs.py:850)
```

pcbnew settings: 

[image]

(seems strange to use a A3 named template for A4 output, but works for me ;-) 

template is located in root directory of runner:

```
    - cd .. && rm -rf kicad_template && git clone git@gitlab.com:xxxx/kicad_template.git
    - cd ci_pcb
    - kibot --log Generated/kibot.log -c V3/Assembly_DNP.kibot.yaml -b V3/Kicad/xx_PCB_PCB.03x.kicad_pcb
```

**To Reproduce**
can not share design, only reporting bug in case others are struggling with the same, seems to be fixed in kibot dev.

**Expected behavior**
a new release of kibot, in kibot 1.8.2 something seems to be broken with relative paths

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`
- Using a docker image? 
  #name: ghcr.io/inti-cmnb/kicad8_auto:1.8.1 # ok
  #name: ghcr.io/inti-cmnb/kicad8_auto:1.8.2 # fails
  #name: ghcr.io/inti-cmnb/kicad8_auto_full:latest # currently 1.8.2 which fails
  name: ghcr.io/inti-cmnb/kicad_auto:dev_k8 # ok

stefan894, 2025-02-05:
.kibot.ymal file:

```
kibot:
    version: 1
    
outputs:     
  - name: print_pdf
    comment: "PDF for the PCB"
    type: pcb_print
    dir: Generated
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      dnf_filter: _kibom_dnf_Config
      scaling: 1.0
      pages:
        - layers: [ F.Silkscreen, F.Fab ]
          sheet: 'Assembly Top'
        - layers: [ B.Silkscreen, B.Fab ]
          sheet: 'Assembly Bottom'
          mirror: true          
```

set-soft, 2025-02-05:
If this is fixed in *dev* images then I must close it. Of course it is good for people looking for information.
I guess this is related to #767 which is different, but forced some changes in the code that looks for the WKS in `pcb_print`

closed 2025-02-05 by set-soft