INTI-CMNB/KiBot #777: [HOW] update kibot and kicad update causes different assembly plan

stefan894, 2025-02-05:
**What do you want to achieve?**
I updated my kibot docker file from:

`ghcr.io/inti-cmnb/kicad8_auto:1.8.1`  to `ghcr.io/inti-cmnb/kicad_auto:dev_k8`

and I see the following changes when executing: 

``
kibot --log Generated/kibot.log -c V3/Assembly_DNP.kibot.yaml -b V3/Kicad/PCB.0003x.kicad_pcb
``

[image]

[image]

seems all holes / pins are now also printed, they were not printed with kibot 1.8.1

**Do you have some PCB/Schematic to use as example?**
sorry can not share design, I might try to make a minimal example

**Do you have some configuration file (.kibot.yaml) that you are using?**

.kibot.ymal file:

```
kibot:
    version: 1
    
outputs:     
  - name: print_pdf
    comment: "PDF for the PCB"
    type: pcb_print
    dir: Generated
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      dnf_filter: _kibom_dnf_Config
      scaling: 1.0
      pages:
        - layers: [ F.Silkscreen, F.Fab ]
          sheet: 'Assembly Top'
        - layers: [ B.Silkscreen, B.Fab ]
          sheet: 'Assembly Bottom'
          mirror: true          
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`
- Using a docker image? `ghcr.io/inti-cmnb/kicad8_auto:1.8.1` and  `ghcr.io/inti-cmnb/kicad_auto:dev_k8``Dockerfile`

stefan894, 2025-02-05:
I think I found it already, the option for visible "Pads" in Kicad might be default on now compared to before.

[image]

[image]

[image]

comparing on/off setting of this on the Silkscreen layer gives me exactly the mentioned difference, so I just  have to figure out how to set it in kibot ;-)

stefan894, 2025-02-05:
I tried "exclude_pads_from_silkscreen = true" doesn't work, as I am not using kicad 5.x  [(as written in documentation)](https://kibot.readthedocs.io/en/master/configuration/outputs/PagesOptions.html)

is there a similar option for kicad 8.0.x ?

stefan894, 2025-02-05:
reproducible with my copy of https://github.com/INTI-CMNB/kibot_variants_arduprog/commit/0931c6eb52cb798f53ae0dbc289b20b281a80a9d

which you can find [here](https://gitlab.com/stefan_bozz/kibot_silkscreen_problem)

pipeline results with kibot 1.8.1:

https://gitlab.com/stefan_bozz/kibot_silkscreen_problem/-/commit/26279de41c5126724aca7c99942c83c96b51eaed

[t1-assembly_kibot_1.8.1.pdf](https://github.com/user-attachments/files/18670984/t1-assembly_kibot_1.8.1.pdf)

[image]

here with kibot 1.8.3 (dev)

https://gitlab.com/stefan_bozz/kibot_silkscreen_problem/-/commit/76751239ab1f874c3eb812420e5dd7a91600f671

[t1-assembly_1.8.3_dev.pdf](https://github.com/user-attachments/files/18670981/t1-assembly_1.8.3_dev.pdf)

[image]

stefan894, 2025-02-05:
kibot 1.8.2 , seems to fail:

https://gitlab.com/stefan_bozz/kibot_silkscreen_problem/-/commit/ce8a85b82080f8e41b9926f394e95d112ee8814f

```
ERROR:Running ['/usr/bin/python3', '/usr/share/kibot/tools/frame_plotter', '/tmp/tmp-kibot-pcb_print_tmp_pcb_for_frame-iijy6dnc/t1.kicad_pcb', '0.2823529411764706', '0.0', '0.0'] returned 1 (kibot.gs - gs.py:853)
```

set-soft, 2025-02-05:
The problem was with the *Fab* layer, not the *SilkScreen*.
Now should work.

closed 2025-02-05 by set-soft

stefan894, 2025-02-05:
thanks @set-soft  for your fast fix, will test it

stefan894, 2025-02-05:
tested with latest ghcr.io/inti-cmnb/kicad_auto_full:dev_k8, works, thank

set-soft, 2025-02-05:
> kibot 1.8.2 , seems to fail:
>
> https://gitlab.com/stefan_bozz/kibot_silkscreen_problem/-/commit/ce8a85b82080f8e41b9926f394e95d112ee8814f
>
> ```
> ERROR:Running ['/usr/bin/python3', [...]

Most probably fixed on 460e79277010281b48843ecc697ba8379aa72164