INTI-CMNB/KiBot #781: [BUG] No module named 'pcb2blender_importer'

cooked, 2025-02-24:
running a blender_export output fails with the following error message:
 
[image]

even though the importer is show active from within Blender 

[image]

Running Kibot locally with the following environment:

- Ubuntu 24.04.1 LTS
- Kicad 8.0.9
- KiBot 2.16
- Blender 4.3.1

**Additional note**
The pcb3d file can be imported correctly in Blender via the add-on when it's done from the Blender GUI (and not automated with KiBot)

set-soft, 2025-02-24:
Here is the problem: https://github.com/30350n/pcb2blender/issues/79#issuecomment-2678867621

stefavi, 2025-02-24:
> > * KiBot 2.16
> 
> I guess this is the pcb2blender version and KiBot is 1.8.2. In this case pcb2blender 2.16 isn't compatible with KiBot 1.8.2. Try KiBot 1.8.3 (from the _dev_ branch) it supports various names used by the pcb2blender plug-in.

correct, I was running 1.8.2  

moving to the dev branch fixed the issue

cooked, 2025-02-24:
> > * KiBot 2.16
> 
> I guess this is the pcb2blender version and KiBot is 1.8.2. In this case pcb2blender 2.16 isn't compatible with KiBot 1.8.2. Try KiBot 1.8.3 (from the _dev_ branch) it supports various names used by the pcb2blender plug-in.

correct, I was running 1.8.2  

moving to the dev branch fixed the issue