INTI-CMNB/KiBot #785: [FEATURE] v2_k9 GitHub action

flaviut, 2025-03-02:
**Is your feature request related to a problem? Please describe.**

I would like to use KiBot with KiCad 9 in my GitHub actions workflow.

**Describe the solution you'd like**

Tag a v2_k9 release.

set-soft, 2025-03-05:
Hi @flaviut !

The KiCad 9 support is in the KiBot 1.8.3, which is currently *beta* (in the *dev* branch).

So we can't have a `v2_k9` tag yet. I tagged `v2_dk9`, the GitHub action that uses *dev* images for KiCad 9. Please try using this tag.

The current status on 1.8.3 is that things work after migrating from 8 to 9, but new features in 9 might not work. An example is the use of "DNP" flags on a whole hierarchy, it will be ignored.

I think this state is ok for a release with "preliminary KiCad 9" support, which could be done soon. But I need confirmation from "beta testers". The code is passing more than 500 tests on native KiCad 9 files, but all of them are just KiCad 8 projects migrated to KiCad 9.

G-Pereira, 2025-03-11:
Using a dev tagged version worked for me:
`container: ghcr.io/inti-cmnb/kicad9_auto:dev_1.8.3-a484698_k9.0.0_d_sid`

chille, 2025-03-17:
I'm playing around with KiBot 1.8.3 and KiCad 9.0.0 right now.

Is there a proper way to run KiBot 1.8.3 in a Docker container? I'm currently using `ghcr.io/inti-cmnb/kicad9_auto_full` in my CI pipeline. The problem is that it is downloading the latest version of KiBot from the Releases in the KiBot repository. The latest dev version is of course not available there.

I have solved the problem in a bit of a hackish way right now. Which, so far, seems to work properly. I have cloned the KiBot repository dev branch and just mounted some of the source files via a Docker volume.

```
-v /path/to/KiBot/src/kibot:/usr/bin/kibot:ro \
-v /path/to/KiBot/kibot:/usr/lib/python3/dist-packages/kibot:ro \
```

I have about 20 different steps in my `*.kibot*.yml`. All seems to work as expected, on a small project i recently created in KiCad 9.

However, the report generation gives me some errors.

I'm using this configuration:
```
  - name: 'Report'
    type: 'report'
    dir: 'report'
```

And get a bunch of this error:
```
/src/kicad/pcbnew/pcb_track.cpp(362): assert "false" failed in GetWidth(): Warning: PCB_VIA::GetWidth called without a layer argument
```

The report is generated and I can't see any obvious errors in it.

I'm not sure if this is due to my Frankenstein KiBot setup, or if it's an actual bug.

@set-soft Do you have any more specific suggestions on what we could test to verify KiBot 1.8.3 with KiCad 9.0.0? Or is it just like "all new functions"?

set-soft, 2025-03-18:
> Is there a proper way to run KiBot 1.8.3 in a Docker container? I'm currently using `ghcr.io/inti-cmnb/kicad9_auto_full` in my CI pipeline.

Just use a *dev* tag, `ghcr.io/inti-cmnb/kicad9_auto_full:dev`

> And get a bunch of this error:
>
> /src/kicad/pcbnew/pcb_track.cpp(362): assert "false" failed in GetWidth(): Warning: PCB_VIA::GetWidth called without a layer argument

This is because KiCad 9 implements separated sizes for vias on each layer. But the Python API allows calling GetWidth without specifying a layer. So this is a KiCad inconsistency. Don't worry.

> @set-soft Do you have any more specific suggestions on what we could test to verify KiBot 1.8.3 with KiCad 9.0.0? Or is it just like "all new functions"?

We know that some new stuff needs to be implemented, but KiBot can be used, just will ignore these new options. They will be added in another release.

Currently the main problem is about new stuff in the schematic format. They aren't documented and we must try using them and see what KiCad generates. We found various things like tables, bezier curves and  rule areas. If more remain the problem will be that schematics using them won't be usable.

set-soft, 2025-03-18:
Now we have 1.8.3 released and v2_k9 should work.

closed 2025-04-24 by set-soft