INTI-CMNB/KiBot #788: [KiCad 9 BUG] Left/right value fields in PDF schematic with custom font not aligned correctly

nguyen-v, 2025-03-06:
Fill as much as you can, remove the rest

**Describe the bug**
When using left or right alignment in the value field of components using a font other than the KiCad font, the alignment of that text doesn't seem to work correctly. The alignment also is not consistent for the same font, if it is set with the ```default_font``` option (here set as 'Times New Roman') or just in the text itself.

In KiCad schematic editor:

[image]

Output PDF:

[image]

**To Reproduce**
Download the following project and run
[kicad9_tests.zip](https://github.com/user-attachments/files/19116041/kicad9_tests.zip)

```kibot -c 'kibot_yaml/kibot_main.yaml' pdf_schematic```

**Expected behavior**
The output should be similar to what is seen on the schematic

**Environment (please complete the following information):**
Where are you running KiBot:
- tried in both CI/CD (v2_dk9) and locally (last dev branch).

**Additional context**
Might be an issue with KiCad and not KiBot, but still reporting here for posterity

set-soft, 2025-03-07:
Seems to be an issue with the CLI interface when using proportional fonts.

Eeschema:

[image]

Note that right after loading the file, and without setting any particular default font, the Times New Roman alignment was wrong.

Plot from GUI:

[image]

Plot from CLI:

[image]

Plot from CLI specifying the same default font as in the GUI:

[image]

set-soft, 2025-03-07:
I filled the following issue: https://gitlab.com/kicad/code/kicad/-/issues/20257

Note that the issue also affects the GUI! I get the wrong alignment from a fresh load of the file.