INTI-CMNB/KiBot #790: [FEATURE] Support for KiCad 9 rule areas in SCH

nguyen-v, 2025-03-10:
**Is your feature request related to a problem? Please describe.**
Currently, using rule areas in the schematic results in the following error when plotting the PDF schematic
```
kicad_sch attribute `[Symbol('rule_area'), [Symbol('polyline'), [Symbol('pts'), [Symbol('xy'), 160.655, 91.44], [Symbol('xy'), 160.655, 109.22], [Symbol('xy'), 164.465, 109.22], [Symbol('xy'), 164.465, 91.44]], [Symbol('stroke'), [Symbol('width'), 0], [Symbol('type'), Symbol('dash')]], [Symbol('fill'), [Symbol('type'), Symbol('none')]], [Symbol('uuid'), Symbol('05707f66-503e-4f50-8291-ad729e8a4fd1')]]]` (kibot.gs - gs.py:897)
```
Where the rule areas is used to set the netclass of multiple nets, for example in my case:

[image]

**Describe the solution you'd like**
Support for the rule areas

**Describe alternatives you've considered**
One netclass directive per net

set-soft, 2025-03-11:
Ok, should work now.
Also bezier curves.

closed 2025-03-11 by set-soft

nguyen-v, 2025-03-11:
Working great, thanks for the quick implementation!

commit 6c0b49bf (2025-03-10): [Schematic][KiCad 9][Added] Support for rule areas

Closes #790