INTI-CMNB/KiBot #791: [BUG] ERROR:Unknown symbol attribute `[Symbol('embedded_fonts'), Symbol('no')]` (kibot.gs - gs.py:848)

doragasu, 2025-03-10:
Fill as much as you can, remove the rest

**Describe the bug**
Kibot fails with error: ERROR:Unknown symbol attribute `[Symbol('embedded_fonts'), Symbol('no')]` (kibot.gs - gs.py:848)

**To Reproduce**
The following CI/CD run using latest KiCad auto image for version 9, shows the problem: https://gitlab.com/doragasu/sms-sl2map/-/jobs/9366666881

This can be reproduced with the files in the following repo, devel branch: https://gitlab.com/doragasu/sms-sl2map/-/tree/devel?ref_type=heads

**Expected behavior**
Kibot completes without error

**Environment (please complete the following information):**
Where are you running KiBot: Gitlab CI/CD

set-soft, 2025-03-11:
Only *dev* images currently support KiCad 9.
The *latest* are stable. This is why they aren't mentioned in the docs.
KiCad 9 support is beta, please try the *dev* images.

closed 2025-03-11 by set-soft

doragasu, 2025-03-11:
It throws an awful lot of warnings with the dev version, but it looks like generated files are all correct. Thanks!

set-soft, 2025-03-11:
Try to figure out what the warnings are saying, if they aren't important just filter them.