INTI-CMNB/KiBot #793: [HOW] Do I force BoM to use the schematic for the BoM output instead of my panel

harm363, 2025-03-13:
**What do you want to achieve?**
The BOM output defaults to building from my panelized PCB, and this gives me the wrong component counts (even the parts needed for 1 panel) and makes ordering parts more difficult. i do not have this problem when i only need a frame (panel containing 1 device), because then there is only one device.

the best option would be to have the BoM exported from the schematic. so how do I set that option or force that?

**Do you have some PCB/Schematic to use as example?**
i have an source file of a schematic,PCB, panel, all the fabrication outputs i make and e 

**Do you have some configuration file (.kibot.yaml) that you are using?**
You can attach it or paste the content in the following section:
```yaml
# Gerber and drill files for JLCPCB
kibot:
  version: 1

global:
  out_dir: fabrication
  drc_exclusions_workaround: true
  
preflight:
  set_text_variables: 
    name: "version"
    #use: ${version}
    #text: "VERSION GOES HERE"
    #command: 'jq .GitTag configs/kikit_post.json'
    command: 'grep "." "configs/.version" | tail -n 1'
  check_zone_fills: true
  fill_zones: true
  drc: 
    filters:
      - filter: "ignore silkscreen overlaps"
        error: "silk_overlap"
      - filter: "ignore silk over copper"
        error: "silk_over_copper"
      - filter: "library warning"
        error: "lib_footprint_issues"

# Gerber and drill files for JLCPCB
kibot:
  version: 1

preflight:
  set_text_variables: 
    name: "version"
    #use: ${version}
    #text: "VERSION GOES HERE"
    #command: 'jq .GitTag configs/kikit_post.json'
    command: 'grep "." "configs/.version" | tail -n 1'

filters:
  - name: 'only_tp'
    comment: 'Inductors removed'
    type: generic
    include_only:
      - column : 'references'
        regex: 'tp'
      - column : 'references'
        regex: 'FID'
  
  - name: "remove_mousebites"
    comment: "removes mousebites from files"
    type: generic
    exclude_any:
      - column: 'references'
        regex: 'KiKit'
      - column: 'references'
        regex: 'tp'
      - column: 'references'
        regex: 'pinhole'
      - column: 'references'
        regex: 'arrow'
        
groups:
  - name: 'boms'
    outputs:
      - PNPpositions
      - netlist_test
      - interactiveBOM
      - bom
      
  - name: "not_boms"
    outputs:
      - testpointlist
      - testpointlist_csv
      - JLCPCB_gerbers
      - JLCPCB_drill

outputs:

  - name: PNPpositions
    type: position
    comment: generate position files
    #dir: fabrication    
    options:
      bottom_negative_x: false
      format: CSV
      only_smd: false
      include_virtual: true
      dnf_filter: remove_mousebites

  - name: netlist_test
    type: netlist
    dir: '../source'
    options: 
      output: '%f.%x'
      format: classic

  - name: interactiveBOM
    type: ibom
    comment: interactive bom
    #dir: "fabrication/"
    options:
      highlight_pin1: true
      dark_mode: true
      blacklist: 'REF*,SYM*,TP*,FID*,KiKit*'
      extra_fields: 'Partnumber'
      show_fields: 'Value,Footprint,Partnumber'
      normalize_field_case: true
      extra_data_file: '/tmp/kibot_build/source/%f.net'
      dnf_filter: remove_mousebites

  - name: testpointlist
    type: ibom
    comment: interactive testpoint list
    #dir: "fabrication/"
    options:
      output: 'testpoints.html'
      highlight_pin1: false
      dark_mode: true
      #dnp_field: 'Partnumber'
      #blacklist: 'REF*,SYM*,FID*,J*,R*,U*,L*,C*,M*,D*,RN*,BT*,H*,Y*,F*,SW*,Q*,V*,G*,P*,TR*'
      dnf_filter: 'only_tp'
      blacklist: 'REF*'
      
  - name: testpointlist_csv
    type: position
    options:
      output: "testpoint-%i.csv"
      bottom_negative_x: false 
      format: CSV
      only_smd: false
      include_virtual: true
      dnf_filter: 'only_tp'

  - name: bom
    type: bom
    #dir: 'fabrication/'
    options:
      format: CSV
      output: 'bom.csv'
      csv:
        #hide_header: true
        hide_pcb_info: true
        hide_stats_info: true
      columns: 
        - field: 'Value'
        - field: 'References'
        - field: 'Footprint'
        - field: 'Partnumber' 
        - field: 'Quantity Per PCB'
        - field: 'Description'
      exclude_filter: "remove_mousebites"

  - name: JLCPCB_gerbers
    comment: Gerbers compatible with JLCPCB
    type: gerber
    #dir: fabrication/JLCPCB
    dir: JLCPCB
    options: &gerber_options
      exclude_edge_layer: true
      exclude_pads_from_silkscreen: true
      plot_sheet_reference: false
      plot_footprint_refs: true
      plot_footprint_values: true
      force_plot_invisible_refs_vals: false
      tent_vias: true
      use_protel_extensions: true
      create_gerber_job_file: false
      disable_aperture_macros: true
      gerber_precision: 4.6
      use_gerber_x2_attributes: false
      use_gerber_net_attributes: false
      line_width: 0.1
      subtract_mask_from_silk: true
      inner_extension_pattern: '.gp%n'
    layers:
      # Note: a more generic approach is to use 'copper' but then the filenames
      # are slightly different.
      - F.Cu
      - B.Cu
      - In1.Cu
      - In2.Cu
      - In3.Cu
      - In4.Cu
      - In5.Cu
      - In6.Cu
      - F.SilkS
      - B.SilkS
      - F.Mask
      - B.Mask
      - B.Paste
      - F.Paste
      - Edge.Cuts

  - name: JLCPCB_drill
    comment: Drill files compatible with JLCPCB
    type: excellon
    #dir: fabrication/JLCPCB
    dir: JLCPCB
    options:
      pth_and_npth_single_file: false
      pth_id: '-PTH'
      npth_id: '-NPTH'
      metric_units: true
      map: gerber
      route_mode_for_oval_holes: false
      output: "%f%i.%x"

```

**Environment (please complete the following information):**
Where are you running KiBot:
I run a docker build from ```docker FROM ghcr.io/inti-cmnb/kicad8_auto:latest```
i only add some python packages for extra PnP support for our pnp machines

**Additional context**
Add any other context about your setup/project.

set-soft, 2025-03-13:
When you work with a panel you run KiBot twice:

1. To get most outputs, including the BoM and the panel. In this run you use the original project
2. To get the PCB fabrication outputs that applies to the panel (gerbers, position, drill, etc.). In this run you use the panel generated in the first step as input

You can use one config file just selecting which outputs are generated on each run.

harm363, 2025-03-13:
i will do that than, thank you.

closed 2025-03-13 by set-soft