INTI-CMNB/KiBot #795: [BUG] non-standard rotated components are rendered on wrong in gerber output

harm363, 2025-03-13:
Fill as much as you can, remove the rest

**Describe the bug**
i have a project that dictates a circle of 16 u.FL connectors, this means that each connector has an rotation of 22.5 degrees (360/16). 
but when i generate gerber files via KiBot only the components rotated on 0,90,180 and 270 degrees are placed correctly. 
if I export gerbers via PCBnew (with the same settings) the gerbers are working correctly.
[image]
[image]
I will add the 2 gerbers in a care package for your viewing pleasure :)

I will also add my KiBot settings there and a screenshot of my manual export. It is very well possible that i did something wrong in KiBot. 

If needed i can make an example project, 

[care_package_rotation_gerber_question.zip](https://github.com/user-attachments/files/19230765/care_package_rotation_gerber_question.zip)

**Environment (please complete the following information):**
I run everything inside a docker container build from the kicad8_auto:latest
```docker FROM ghcr.io/inti-cmnb/kicad8_auto:latest```

harm363, 2025-03-13:
[rotation_issue.zip](https://github.com/user-attachments/files/19230965/rotation_issue.zip)

zip archive with an example project and my standard kibot.yaml file

set-soft, 2025-03-13:
Hi @harm363 !

Try using the current *dev* images. I think the issue is already solved.
Is some interaction between the `testpointlist_csv` and the gerbers with rotation filters.

harm363, 2025-05-20:
Hey @set-soft, 

My apologies for my absence, it has been busy.
Last week I updated everything to kicad9 with the latest docker images. 
i ran all of the affected projects with the new image and everything looks good now,

many thanks for fixing the problem and for your patience

closed 2025-05-20 by harm363