INTI-CMNB/KiBot #799: [BUG] `type: export_3d` not showing colors

wico-sarin, 2025-03-18:
**Describe the bug**
When exporting from KiCad 9 via PCB `editor->Export>STEP` I get a nice looking **colored** 3D file. However, when using type: `export_3d` in kibot with standard configuration the result is **without colors, all gray.**

**To Reproduce**
Run KiBot with the following configuration file:
```
  - name: step
    comment: 3D model using KiCad's built-in export
    type: export_3d
```

**Expected behavior**
A colored 3D model the same as when using PCB editors built-in export option.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
CI workflow using the GitHub action with the tag **`INTI-CMNB/KiBot@v2_dk9`.

set-soft, 2025-03-18:
Might be a KiCad CLI limitation.
I need a very simple test example and a mechanism to verify it. 
I tried using FreeCAD and I always get gray.

wico-sarin, 2025-03-19:
When exporting using `kicad-cli pcb export step Test.kicad_pcb` I get a STEP file including colors. Only if using KiBot `export_3d` the problem occurs.

> I tried using FreeCAD and I always get gray.

Yes I also use FreeCad to view the output. Same observation as you.

set-soft, 2025-03-19:
> When exporting using `kicad-cli pcb export step Test.kicad_pcb` I get a STEP file including colors. Only if using KiBot `export_3d` the problem occurs.

Then please provide a test case, the `export_3d` output **uses** kicad-cli.

> 
> > I tried using FreeCAD and I always get gray.
> 
> Yes I also use FreeCad to view the output. Same observation as you.

So then I need some mechanism to verify colors are there.

wico-sarin, 2025-03-21:
I can confirm that using the latest docker image `ghcr.io/inti-cmnb/kicad9_auto_full:latest` I get a STEP with colors both, using KiCad and KiBot on my local machine.

The problem seems to be related to the environment using the Github Action `INTI-CMNB/KiBot@v2_dk9`.

set-soft, 2025-03-25:
> What does it mean to be "self-contained"?

That all the needed stuff is in your project, in particular all 3D models.

When I use freecad I always get gray PCBs, so I need some tool to see when it happens, without it I can investigate the source of the problem.

set-soft, 2025-03-26:
Hi @wico-sarin !

Do you get colors using https://3dviewer.net/ ?

I tried exporting with the GUI and I got only one color, used for the core, the rest is gray.

[image]

What software are you using to see the step file generated locally?

wico-sarin, 2025-03-26:
Hi @set-soft,

I created a sample project for this problem, see attached. Just some random components on a board.
[3DExportTest.zip](https://github.com/user-attachments/files/19468041/3DExportTest.zip)

I use the following configuration file:
```
kibot:
  version: 1
  
outputs:
  # 3D Model (KiCad 9)
   - name: step
     type: export_3d
     options:
       output: "Export with KiBot"
       include_pads: true
       include_silkscreen: true
       include_soldermask: true
       include_tracks: true
```

Exporting using KiBot from the latest docker `ghcr.io/inti-cmnb/kicad9_auto_full:latest` release ends with a green board and grey components viewed from `FreeCad 1.0`.

[image]

The same step file viewed in `3dviewer.net` looks all grey.

[image]

Viewed with KiCad 9.0, I got colors.
[image]

set-soft, 2025-03-26:
The KiCad internal 3D viewer doesn't show STEP files, is an OpenGL render done by KiCad.
If it works for Windows, but not for Ubuntu this is most probably related to the `occt` libs. 
I have v7.6 on my system, the same used by the docker images, but the KiCad from Debian testing uses v7.8.
I know that `PLY` files can't be generated using v7.6, so I guess the Windows binaries has v7.7 or v7.8 included. 
So it looks like this problem is completely outside KiBot.

set-soft, 2025-04-11:
@oliv3r : KiBot can also export your project containing all the 3D models, the copy_files output can do it using `source_type: prj`

oliv3r, 2025-04-11:
No denying that!  I was just trying to help explain self-contained.

And I do use this for other reasons myself (and when using it via the cli this works too) I think its a little chicken and egg story when looking at CI workflows.

mdeweerd, 2025-05-22:
In my case, the step models are "self-contained", proven by the fact that they are in the STEP export.

I compared my windows export STEP content with the docker export:

Windows:
```log
  'Open CASCADE STEP processor 7.8','KiCad to STEP converter','Unknown'
  );
```

Docker image (kicad9_auto_full:1.8.4):
```log
  'Open CASCADE STEP processor 7.6','KiCad to STEP converter','Unknown'
```

And also processor 7.6 in the kicad9_auto_full:dev image .

So maybe it's solved with a version later than KiCad: 9.0.1+1 as windows is on KiCAd Version: 9.0.2 .

set-soft, 2025-05-23:
> So maybe it's solved with a version later than KiCad: 9.0.1+1 as windows is on KiCAd Version: 9.0.2 .

I don't think so. In Windows the library is part of KiCad distribution, KiCad has to include OCCT, wxWidgets, Python, etc. 
On Linux these elements aren't part of KiCad, they are on their own package. So Debian package for KiCad 9.0.2 won't change anything.
The v7.8 of the lib isn't in the stable distro so, unless Debian people backports it, we'll have to wait until trixie (testing) is released, presumably in the next few months.

set-soft, 2025-05-23:
BTW: What's is the real use of colored STEP files?
They are used to verify distances, not to generate renders, for renders the VRML files are much better.
On the other side: VRML files are useless for real size verification.
This is why KiCad has both models (WRL + STEP)