INTI-CMNB/KiBot #802: [BUG] kicad9 embedded datasheet fails

allardMVP, 2025-03-24:
When Running kibot on a kicad 9 project that contains a symbol whos datasheet is embedded into the symbol kibot fails 

ERROR:While loading `/github/workspace/myproj.kicad_sch` (kibot.gs - gs.py:897)
ERROR:Unknown symbol attribute `[Symbol('embedded_files'), [Symbol('file'), [Symbol('name'), 'si864x-datasheet.pdf'], [Symbol('type'), Symbol('datasheet')], [Symbol('data'), 

The error dumps the entire content of the symbol at this point I will not reproduce here.  This was run on the latest kibot during the design report phase of the run.

set-soft, 2025-03-25:
Hi @allardMVP !

The above patch solves the problem. Thanks for reporting!

P.S. I changed it to "feature" because this isn't documented at KiCad's site

closed 2025-03-25 by set-soft

allardMVP, 2025-03-26:
Hi @set-soft 

Thanks for working on this.  I tested against commit: 62b4cb78442594c4bf25c18612d5b375cfeb335c

but I still get an error:
```
ERROR:Unknown symbol attribute `[Symbol('embedded_files'), [Symbol('file'), [Symbol('name'), 'si864x-datasheet.pdf'], [Symbol('type'), Symbol('datasheet')], [Symbol('data'), Symbol('|KLUv/aA3exAAZFYOzOobJVBERi0xLjcKJeLjz9MNCjYgMCBvYmoKMTEzNDMKZW5kb2JqCjQ8PC9U'), Symbol('eXBlL1hPYmplY3QKL1N1YnR5cGUvSW1hZ2UKL1dpZHRoIDE4ODUKL0hlaWdodCA2MjUKL0ZpbHRl'), Symbol('ciAvRmxhdGVEZWNvZGUvQml0c1BlckNvbXBvbmVudCA4Ci9Db2xvclNwYWNlIC9EZXZpY2VSR0Ig'), Symbol('L1NNYXNrIDUgMCBSIC9MZW5ndGggNiAwIFIKPj4Kc3RyZWFtCnja7d1RbmQ5cgXQHdjrMODZzKzK'), Symbol('G/L6cj4EFBqqkuopH8kXEfcc5KfdXQoGGeQdddbrBQAAAEX9z//9/8dHKQAAAAAAEvyKhYXDISRp'), Symbol('PiXDwmEAAAAAgASSNH5tGAAAAAAgjWSQcM3vybBgPMkwQBq/NgwAAAAAkEYyEEg4kOaPNqtkWDCY'), Symbol('jV8bBiB8/KkGAAAA4U9jyTAAUVNPQQAAAPBMFg4DkDPp1AQAAAAkwwCEDDg1AQAAgG8ezh7RAEya'), Symbol('a2oCAAAAFx/RXtMADJhlygIAAAA/fU17VgPQdIQpCwAAANx5WXtfA9BobCx5YntrA1B/Wqx9a3t0'), Symbol('A1B5SKkMAAAA7Ht3e30DUGowHHiAe4YDUGEenHyJe5ID8OAYUhkAAAB46lXubQ7A4dGjMgAAAPD4'), Symbol('89w7HYBjE0dxAAAAoM473YMdgK2DRuab3csdgOXDRfjj3RMeg
```

Not sure why this

set-soft, 2025-03-27:
I tried this file: 

[test.zip](https://github.com/user-attachments/files/19483821/test.zip)

Without problems, I used the following config:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: 'sch'
    comment: "Schematic copied"
    type: sch_variant
    options:
      copy_project: true
```

What do you get using it?

allardMVP, 2025-03-27:
I checked your schematic.  I am running on windows so running kibot is difficult.  I have been running it as a github action as it is much simpler.  I did notice it looks like there is nothing embedded in your schematic symbol.  Here is a simple project with the symbol that is failing during the github action.  Also included is the symbol exported to a library so you can see the embedded datasheet.  Unfortunately Kicad 9 embedded datasheets you have to open the library symbol and launch the datasheet from there.  I believe the kicad team is working on it though.

[test.zip](https://github.com/user-attachments/files/19486468/test.zip)

set-soft, 2025-03-27:
> I checked your schematic. I am running on windows so running kibot is difficult. 

I recommend trying to install docker and using docker images.

> I have been running it as a github action as it is much simpler. I did notice it looks like there is nothing embedded in your schematic symbol. 

At line 976 you can see the data:

```lisp
                       (embedded_files
                                (file
                                        (name "bom-F_Paste.pdf")
                                        (type datasheet)
                                        (data |KLUv/WA9FrVQAGp7AB04IOptA6/8ZllZk2EUwpPXhEgQwzD3RK/VloQfSDMqW7LIwH85RCjXMXZ3
                                                d/fmysVn0PsxvGgFcALuAf0BiQF1JB4vKhyh0mf1t9bkrnvS65JpRxsrCKLnrKgYEDdtMFL8zvgF
                                                +LSIRhFAD0I/F6biZEIy4e8RoJs2ZJQEsSOjBFGfEovEjje4mCq8tbi9KUE1DMzYMaIlNSkD37lF
                                                3Aga3txgqBzdiQCHqmqNiQODXI445WEqfLMLM+kLspGgXY0bEsMUtDPItkKyict7Mfa5iWn2qOKx
```

> Here is a simple project with the symbol that is failing during the github action. 

How are you running the github action?

> Also included is the symbol exported to a library so you can see the embedded datasheet. Unfortunately Kicad 9 embedded datasheets you have to open the library symbol and launch the datasheet from there. I believe the kicad team is working on it though.

It works using the current KiBot code.  You must use *dev* images, i.e. `v2_dk9` GitHub Action or *ghcr.io/inti-cmnb/kicad9_auto_full:dev*

allardMVP, 2025-03-27:
Here are the snippets from my workflow yaml:
```yaml
    runs-on: ubuntu-latest
    container: ghcr.io/inti-cmnb/kicad9_auto_full:dev
-------
 - uses: INTI-CMNB/KiBot@dev
```

I was using the commit sha rather than the dev branch.

Running the above I still get the same error.  I don't know if I am grabbing the correct version of kibot though as I get a version of 1.8.3 which I believe it should be 1.8.4 as you updated the dev branch.  But I don't see how to correct it.

allardMVP, 2025-03-27:
Alright......  I got kibot mostly running with WSL.  The schematic, report, BOM, assembly drawing and 3d export seem to all be working.  Not sure why github actions is not grabbing the wrong version of kibot.....

But, the 3d step is not quite working.  The embedded stp file in one of the footprint is unable to expand.  Error: 
```
ERROR:Unable to expand `kicad-embed` in `${kicad-embed}///SOIC8P508_2108X1676X335L88X71.stp` (kibot - config.py:130)
```

I think the embedded 3d model will needed to extracted and unzipped then stored somewhere and linked into the end step file.

allardMVP, 2025-03-27:
Gotcha....  I thought I could run off a branch that way and didn't need to specify a tag.  Up and running though the 3d models that are embedded into footprints.  I'll take a look into how to extract the embedded data.

Based on the construction of the kicad_pcb file it looks like all embedded files are listed at the bottom and can be extracted then referenced when building the 3d file.

Thanks again for the help and for the Awesome Tool!

set-soft, 2025-03-27:
> ```
> ERROR:Unable to expand `kicad-embed` in `${kicad-embed}///SOIC8P508_2108X1676X335L88X71.stp` (kibot - config.py:130)
> ```

Where is `kicad-embed` defined? It isn't defined by KiCad 9.0.0

[image]

I also tried using the name provided by KiCad:

[image]

[image]

So it doesn't work in the KiCad I have. What are you doing/using?

set-soft, 2025-03-27:
> Based on the construction of the kicad_pcb file it looks like all embedded files are listed at the bottom and can be extracted then referenced when building the 3d file.

They should be extracted by KiCad! The STEP export is done running KiCad. Can you provide a small example file that can be loaded in the GUI with working embedded 3D models?

allardMVP, 2025-03-27:
Huh...  Kicad embedded files end up in: `C:\Users\USER\AppData\Local\KiCad\9.0\embed`

I'll put together a small kicad project with an embedded model and post it in a couple minutes.

allardMVP, 2025-03-27:
[test.zip](https://github.com/user-attachments/files/19490158/test.zip)

One of the symbols has an embedded datasheet and the other has a embedded 3d model in the footprint.  I tossed a basic pcb outline so it should work.

Also, both the symbols and footprints are exported to standalone libraries for both schematic and footprint.

set-soft, 2025-03-27:
I tried it and things aren't fully working on the GUI:

1. If I edit the footprint I can see the 3D model and the embedded file
2. If I ask for a 3D view I can't see the 3D model
3. If I add another embedded file and try to use it I get an error like I shown before

So, at least for KiCad 9.0.0, it isn't functional

After looking at the generated file I found one of the issues, when you try to copy and paste the "kicad-embeded" URI KiCad is adding an extra space to the name!!!! This makes the URI invalid.
After manually removing it I got the 3D model working.
So the remaining issue is that they don't work for the 3D Viewer.

In any case: these names are `kicad-embed://SOIC8P508_2108X1676X335L88X71.stp`, not `${kicad-embed}///SOIC8P508_2108X1676X335L88X71.stp` ... What 's the name you see?

set-soft, 2025-03-27:
I think I know where the issue is, I'll verify it tomorrow. But I think the problem is that the URI looks like a KiCad 6 Alias.

allardMVP, 2025-03-27:
Weird they don't work in the 3d viewer for you.  Yes the added space bug drives me nuts as well.  I have the kicad-embed://  (No $)

[image]

I have both 3d models showing fine in the 3d viewer: 

[image]

Even if I clear the embedded folder and force kicad to recreate just opening the pcb and launching the 3d viewer the models show up fine.

Alright.

set-soft, 2025-03-28:
Ok, most misteries solved:

1. The `${kicad-embed}` was clearly because the `kicad-embed://NAME` is quite similar to `ALIAS_LIB:NAME` from KiCad 6. Now embedded models aren't processed by the variables expansion
2. My missing 3D model was just because I had the SMD models disabled on KiCad 9
3. I found that the mechanism, at least on 9.0.0, has another bug: if you add a model to a footprint you can then see this is available in the "Board Setup|Board Data|Embedded Files", but you can't reference it on other footprint. The names are global, you can't add a file that is already there. If you add it locally to another footprint this is allowed ... which is a mess because then you have two with the same name and only one listed, which is IMHO really wrong, if they are the same a waste of resources, if they are different a problem because you can't export both. If you add the URI first and then the file ... you must close the window to make it work ... full of bugs, looks very poorly designed and tested.

Anyways, now should work.

BTW: I left the THT resistor coloring disabled for embedded models. This is because the API is incomplete and the implementation problematic, it uses a CRC that I couldn't verify

commit 6f6b78c6 (2025-04-03): [Schematic][Added] Support for files embedded on symbols

Closes #802