INTI-CMNB/KiBot #804: [BUG] THT resistors casuses python KeyError for blender/pcb2blender for second run.

oliv3r, 2025-04-11:
**Describe the bug**
When running the blender output for a second time, THT parts trigger the following traceback:
```
ERROR:Error from Blender run:
Traceback (most recent call last):
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 226, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 565, in import_pcb3d
    component = self.component_cache[url]
                ~~~~~~~~~~~~~~~~~~~~^^^^^
KeyError: 'components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl'
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/kibot/blender_scripts/blender_export.py", line 464, in <module>
    main()
  File "/usr/local/lib/python3.11/dist-packages/kibot/blender_scripts/blender_export.py", line 438, in main
    bpy.ops.pcb2blender.import_pcb3d(**ops)
  File "/usr/bin/4.2/scripts/modules/bpy/ops.py", line 109, in __call__
    ret = _op_call(self.idname_py(), kw)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Error: Python: Traceback (most recent call last):
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 226, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 565, in import_pcb3d
    component = self.component_cache[url]
                ~~~~~~~~~~~~~~~~~~~~^^^^^
KeyError: 'components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl'
Location: /usr/bin/4.2/scripts/modules/bpy/ops.py:109

Importing PCB3D file ...
Add-on not loaded: "pcb2blender_importer", cause: No module named 'pcb2blender_importer'
Add-on not loaded: "pcb3d_importer", cause: No module named 'pcb3d_importer'
Add-on not loaded: "bl_ext.blender_org.pcb3d_importer", cause: No module named 'bl_ext.blender_org.pcb3d_importer'
Skipping wheel for other system (win_amd64 != linux_x86_64): skia_python-87.6-cp311-cp311-win_amd64.whl
Skipping wheel for other system (macosx_11_0_arm64 != linux_x86_64): skia_python-87.6-cp311-cp311-macosx_11_0_arm64.whl
Skipping wheel for other system (macosx_10_9_x86_64 != linux_x86_64): skia_python-87.6-cp311-cp311-macosx_10_9_x86_64.whl
Skipping wheel for other system (win_amd64 != linux_x86_64): pillow-11.0.0-cp311-cp311-win_amd64.whl
Skipping wheel for other system (macosx_11_0_arm64 != linux_x86_64): pillow-11.0.0-cp311-cp311-macosx_11_0_arm64.whl
Skipping wheel for other system (macosx_10_10_x86_64 != linux_x86_64): pillow-11.0.0-cp311-cp311-macosx_10_10_x86_64.whl
Error: Python: Traceback (most recent call last):
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 226, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 565, in import_pcb3d
    component = self.component_cache[url]
                ~~~~~~~~~~~~~~~~~~~~^^^^^
KeyError: 'components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl'
Location: /usr/bin/4.2/scripts/modules/bpy/ops.py:109

Blender quit (kibot.gs - gs.py:899)
```

**To Reproduce**
```
% cd /tmp
% git clone https://gitlab.com/ci-includes/kici.git
% cd kici
% git reset --hard fe7165e
% docker run \
    --env HOME='/tmp' \
    --interactive \
    --pull always \
    --rm \
    --tty \
    --user "$(id -u):$(id -g)" \
    --volume "$(pwd):/workdir" \
    --workdir "/workdir" \
    'ghcr.io/inti-cmnb/kicad9_auto_full:dev' \
    '/bin/bash'
$ cd hello
$ export KICI_BLENDER_SAMPLES=3
$ export CI_JOB_NAME='KiCi'
kibot --defs-from-env --stop-on-warnings --plot-config '.kibot/kici_blender.kibot.yaml'
```

Note, that when running the above with `3D_isometric_view` appended, it works fine, it only fails if the `3D_isometric_view` is run after `3D_top_view` was run before. When using the 'world' directory however, all 3 views are generated normally (as it does not contain any THT parts).

The exporting of the samples (e.g. setting options:samples) seems to be affecting things, but I'm not sure yet how.

**Expected behavior**
All 3 outputs get generated.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? GitLab
- In a GitHub action? Which tag are you using? dev and 1.8.4
- Using a docker image? kicad9_auto_full:dev(

**Additional context**
This was working fine with kibot 1.8.2 and KiCad 8.

set-soft, 2025-04-11:
I can't reproduce it:

```shell
$ ./run.sh 
dev: Pulling from inti-cmnb/kicad9_auto_full
Digest: sha256:688cdaf6c5b58abe1562b945f2bb520a47648681e6f0da222dd4aadc7122c5fc
Status: Image is up to date for ghcr.io/inti-cmnb/kicad9_auto_full:dev
kicad@94e33f40e014:/workdir$ export KICI_BLENDER_SAMPLES=3
kicad@94e33f40e014:/workdir$ export CI_JOB_NAME='KiCi'
kicad@94e33f40e014:/workdir$ cd hello/
kicad@94e33f40e014:/workdir/hello$ kibot --defs-from-env --stop-on-warnings --plot-config '.kibot/kici_blender.kibot.yaml'
Using SCH file: hello.kicad_sch
- '3D render from the top.' (3D_top_view) [blender_export]
- '3D render from an isometric viewpoint.' (3D_isometric_view) [blender_export]
- '3D render from the bottom.' (3D_bottom_view) [blender_export]
Found 0 unique warning/s (2 total, 2 filtered)
kicad@94e33f40e014:/workdir/hello$ exit
exit
$ cat run.sh 
docker run \
    --env HOME='/tmp/' \
    --interactive \
    --pull always \
    --rm \
    --tty \
    --user "$(id -u):$(id -g)" \
    --volume "$(pwd):/workdir" \
    --workdir "/workdir" \
    'ghcr.io/inti-cmnb/kicad9_auto_full:dev' \
    '/bin/bash'
```

Got:

[image]

[image]

[image]

Also note that the fail is inside the `pcb2blender` import plug-in (executed by Blender):

```
Error: Python: Traceback (most recent call last):
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 226, in execute
    if (pcb := self.import_pcb3d(context, filepath)) == {"CANCELLED"}:
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/4.2/extensions/system/pcb3d_importer/importer.py", line 565, in import_pcb3d
    component = self.component_cache[url]
                ~~~~~~~~~~~~~~~~~~~~^^^^^
KeyError: 'components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl'
```

Checking the content of the exported 3D model I get:

```
kici/hello/outputs/pcb3d$ unzip -t hello-blender_export.pcb3d 
Archive:  hello-blender_export.pcb3d
    testing: pcb.wrl                  OK
    testing: components/              OK
    testing: components/DIP-8_W7.62mm.wrl   OK
    testing: components/BatteryHolder_Keystone_103_1x20mm.wrl   OK
    testing: components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl   OK
    testing: components/LED_D20.0mm.wrl   OK
    testing: components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_1_10.wrl   OK
    testing: components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_5_6_3_10.wrl   OK
    testing: components/CP_Radial_D6.3mm_P2.50mm.wrl   OK
    testing: layers/F_Cu.svg          OK
    testing: layers/B_Cu.svg          OK
    testing: layers/F_Paste.svg       OK
    testing: layers/B_Paste.svg       OK
    testing: layers/F_Mask.svg        OK
    testing: layers/B_Mask.svg        OK
    testing: layers/F_SilkS.svg       OK
    testing: layers/B_SilkS.svg       OK
    testing: layers/bounds            OK
    testing: pads/Battery_Cell_BT1_0_0   OK
    testing: pads/Battery_Cell_BT1_0_1   OK
    testing: pads/1_kΩ_R1_1_0       OK
    testing: pads/1_kΩ_R1_1_1       OK
    testing: pads/1_µF_C1_3_0        OK
    testing: pads/1_µF_C1_3_1        OK
    testing: pads/56_kΩ_R2_4_0      OK
    testing: pads/56_kΩ_R2_4_1      OK
    testing: pads/100_Ω_R3_5_0      OK
    testing: pads/100_Ω_R3_5_1      OK
    testing: pads/LM555xN_U1_6_0      OK
    testing: pads/LM555xN_U1_6_1      OK
    testing: pads/LM555xN_U1_6_2      OK
    testing: pads/LM555xN_U1_6_3      OK
    testing: pads/LM555xN_U1_6_4      OK
    testing: pads/LM555xN_U1_6_5      OK
    testing: pads/LM555xN_U1_6_6      OK
    testing: pads/LM555xN_U1_6_7      OK
    testing: pads/LED_D1_9_0          OK
    testing: pads/LED_D1_9_1          OK
    testing: layers/stackup           OK
    testing: boards/                  OK
No errors detected in compressed data of hello-blender_export.pcb3d.
```

The name of the file is valid (`components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl`) and the file is inside the 3D model.

The problem you get is when the importer parses the VRML generated file looking for the 3D models of the components, here the VRML mentions `components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl` and the code looks for this file in the `pcb3d` file. 

It might be the result of some very bizarre race condition in your system, but in this case the error should be visible before this point.

According to what you say the first call to Blender always works, but the second fails, unless other process is changing the `hello-blender_export.pcb3d` I don't see how it can happen. 

Also: You can put the 3 points of view in 1 call to Blender to avoid loading Blender three times

oliv3r, 2025-04-18:
So I am trying it on my work laptop right now, and works great. On both my Desktop and Server it causes problems. The only difference is Intel (laptop) vs AMD. I see if I can try it on my other laptop (also intel) to see if the race condition somehow relates to that. Obviously I used the exact same commands as pasted above (so re-cloning in /tmp (I just did change it to do the cloning within the container, to be even more reproducible).

Meanwhile (and I should have linked it already) I do have the [pipeline run](https://gitlab.com/ci-includes/kici/-/pipelines/1764287576) of the [blender job](https://gitlab.com/ci-includes/kici/-/jobs/9696895611). But just logs, failed pipelines do not keep artifacts ... but re-doing it manually (as above) without the gitlab runner on my server, it does indeed crash as expected. I've attached the generated output in hopes we can see what's going on? 

[hello.zip](https://github.com/user-attachments/files/19814019/hello.zip)

> Also: You can put the 3 points of view in 1 call to Blender to avoid loading Blender three times

Really? How? :) Not that'll speed up things much, renderings will still take considerably long, but If it could simplify my yaml ...

set-soft, 2025-04-22:
> edit: just ran the gitlab-runner on my laptop; and it now also crashes here. The only difference I can see is single threaded vs multi-threaded runs, my gitlab-runners run in single-threaded mode ...

A lot of outputs uses KiAuto which, in general, can't be run in parallel. You can use separated VMs if you want to exploit parallelism.
 
> > Also: You can put the 3 points of view in 1 call to Blender to avoid loading Blender three times
> 
> Really? How? :) Not that'll speed up things much, renderings will still take considerably long, but If it could simplify my yaml 

The `point_of_view` option can be a *dict*, just one view, or a *list of dicts*, this is a list of views. So you can put all the views as YAML list and they will be rendered in the same Blender instance.

oliv3r, 2025-04-22:
> A lot of outputs uses KiAuto which, in general, can't be run in parallel. You can use separated VMs if you want to exploit parallelism.

Actually, I'm fine with it running in a single thread, I start multiple jobs (doing different isolated things, in isolated docker containers) to gain parallelisms, but that's not even what I'm doing here, the 'concurrency' is set to 1, e.g. 1 job, 1 container, 1 thread. So this can't be the problem.

So I'm still stuck in trying to resolve this blender/pcb2blender error, and other then the attached zip file above, I am lost on things to try? I can very much reproduce it, just have no idea where to look/debug :(

> The `point_of_view` option can be a _dict_, just one view, or a _list of dicts_, this is a list of views. So you can put all the views as YAML list and they will be rendered in the same Blender [...]

So you mean to say I can replace:
```yaml
outputs:
  - name: '3D_top_view'
    run_by_default: @RUN_BY_DEFAULT@
    comment: '3D render from the top.'
    type: 'blender_export'
    dir: 'outputs/pcb3d'
    options: &blender_options
      render_options:
        transparent_background: true
        no_denoiser: @KICI_BLENDER_NO_DENOISER@
        samples: @KICI_BLENDER_SAMPLES@
      outputs:
        - type: 'blender'
          dir: '../blender'
        - type: 'render'
          dir: '../png'
  - name: '3D_isometric_view'
    run_by_default: @RUN_BY_DEFAULT@
    comment: '3D render from an isometric viewpoint.'
    type: 'blender_export'
    dir: 'outputs/pcb3d'
    options:
      <<: *blender_options
      point_of_view:
        rotate_x: @KICI_BLENDER_ISO_X@
        rotate_y: @KICI_BLENDER_ISO_Y@
        rotate_z: @KICI_BLENDER_ISO_Z@
        view: 'front'
  - name: '3D_bottom_view'
    run_by_default: @RUN_BY_DEFAULT@
    comment: '3D render from the bottom.'
    type: 'blender_export'
    dir: 'outputs/pcb3d'
    options:
      <<: *blender_options
      point_of_view:
        view: 'bottom'
```

with just
```yaml
outputs:
  - name: '3D_view'
    run_by_default: @RUN_BY_DEFAULT@
    comment: '3D render from the top.'
    type: 'blender_export'
    dir: 'outputs/pcb3d'
    options: &blender_options
      render_options:
        transparent_background: true
        no_denoiser: @KICI_BLENDER_NO_DENOISER@
        samples: @KICI_BLENDER_SAMPLES@
      outputs:
        - type: 'blender'
          dir: '../blender'
        - type: 'render'
          dir: '../png'
      point_of_view:
        - rotate_x: @KICI_BLENDER_ISO_X@
          rotate_y: @KICI_BLENDER_ISO_Y@
          rotate_z: @KICI_BLENDER_ISO_Z@
        - view: 'front'
        - view: 'bottom'
```

set-soft, 2025-04-22:
> Actually, I'm fine with it running in a single thread, I start multiple jobs (doing different isolated things, in isolated docker containers) to gain parallelisms, but that's not even what I'm doing [...]

I executed various times your config and it never fails in my system.
Plus it works the first time and you say no other process changes the files between the first and second run.
So I don't see how it can happen.

I checked the file inside hello.zip and is correct.

Looking at your pipeline run it failed on the third render.  So I guess your problem is about resources, the first run seems to work, but the fail is in the 2nd/3rd because some resource is fully consumed (memory? execution time?)
 
> So you mean to say I can replace:

Yes.

> But the downside would be that you cannot run them individually, and the most time consuming will be the rendering anyway, so the 'net-win' would be a couple of seconds on a (in my case) 10 minute run?

But in your case it will mean that you can generate the 3 images instead of 1. Unless the resource is time.

oliv3r, 2025-04-22:
> I executed various times your config and it never fails in my system. Plus it works the first time and you say no other process changes the files between the first and second run. So I don't see how [...]

It depends, on my desktop I can get it to fail too, and I have 24 threads and 128G of ram available there. On my server, it's 'only' 6 cores (but running single threaded) and 'only' 32G of ram.

Also it 'crashed' after 48 seconds, the pipeline timeout is like 2hours (because I know blender takes forever on that thing). It also only started to happen with your latest kicad9_dev container. If you look at the [pipelines overview](https://gitlab.com/ci-includes/kici/-/pipelines) you'll see hundreds of correct runs. E.g. the last one](https://gitlab.com/ci-includes/kici/-/jobs/8644798817) with kicad8 and kibot 1.8.2 which ran for 16 minutes for all 6 views ... Also it being a python crashdump ... anyway, if you can give me pointers into how to debug this, what kind of prints to put where, I'll be happy to dig much deeper, but my python knowledge is rather limited, and most I have to defer to chatgpt :)

> Yes.
>
>
> But in your case it will mean that you can generate the 3 images instead of 1. Unless the resource is time.
Time is not really an issue, the pipeline just 'runs'. Having 3 entries just allows one to run `kibot label_b` to only run that single one, which is I guess where the advantage is?

set-soft, 2025-04-23:
Ok, looking at the full debug information now I have a better idea of what is going on.

It looks like even when KiCad exited and finished creating the WRL your system doesn't immediately have the files in your disk.

Here is the third run:

```
DEBUG:Generating `/workdir/kici/hello/outputs/pcb3d/hello-blender_export.pcb3d` archive (kibot - out_compress.py:255)
DEBUG:Adding /tmp/tmprdwigdcb/pcb.wrl as pcb.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmprdwigdcb/components as components (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmprdwigdcb/components/BatteryHolder_Keystone_103_1x20mm.wrl as components/BatteryHolder_Keystone_103_1x20mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmprdwigdcb/components/CP_Radial_D6.3mm_P2.50mm.wrl as components/CP_Radial_D6.3mm_P2.50mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmprdwigdcb/components/DIP-8_W7.62mm.wrl as components/DIP-8_W7.62mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmprdwigdcb/components/LED_D20.0mm.wrl as components/LED_D20.0mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmprdwigdcb/layers/F_Cu.svg as layers/F_Cu.svg (kibot - out_compress.py:113)
```

So here we don't have **R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl**, in contrast the other runs say:

``
`DEBUG:Generating `/workdir/kici/hello/outputs/pcb3d/hello-blender_export.pcb3d` archive (kibot - out_compress.py:255)
DEBUG:Adding /tmp/tmpp7n_cgt9/pcb.wrl as pcb.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components as components (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components/BatteryHolder_Keystone_103_1x20mm.wrl as components/BatteryHolder_Keystone_103_1x20mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl as components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components/CP_Radial_D6.3mm_P2.50mm.wrl as components/CP_Radial_D6.3mm_P2.50mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_5_6_3_10.wrl as components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_5_6_3_10.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_1_10.wrl as components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_1_10.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components/DIP-8_W7.62mm.wrl as components/DIP-8_W7.62mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/components/LED_D20.0mm.wrl as components/LED_D20.0mm.wrl (kibot - out_compress.py:113)
DEBUG:Adding /tmp/tmpp7n_cgt9/layers/F_Cu.svg as layers/F_Cu.svg (kibot - out_compress.py:113)
```

All components are there. As the list of components is created by KiCad we just scan the subdir, i.e.

```
DEBUG:- Adding subdir: /tmp/tmpozugnmca/components (kibot - out_compress.py:193)
```

I saw a similar bug in the docker for MacOS when using some particular filesystem to "accelerate" things. 
If this is the case this is a bug in the docker you use.
How to workaround?
Simple: create the PCB3D file using *PCBBlender_2_7* template, do other stuff and finally run the renders, always using the generated PCB3D file.

Here is an example of use of the template:

```
kibot:
  version: 1

import:
  - file: PCB2Blender_2_7

outputs:
  - name: '3d_export'
    comment: "Exports the PCB in blender format"
    type: blender_export
    options:
      pcb3d: _PCB2Blender_2_7
      render_options:
        transparent_background: true
        samples: 10
      point_of_view:
        rotate_x: 30
        rotate_z: -20
      outputs:
        - type: blender
        - type: render
```

So it looks like an aggressive optimization that is creating a race condition.

> Time is not really an issue, the pipeline just 'runs'. Having 3 entries just allows one to run kibot label_b to only run that single one, which is I guess where the advantage is?

You avoid creating the PCB3D over and over, and starting a huge tool over and over.

oliv3r, 2025-04-23:
> Ok, looking at the full debug information now I have a better idea of what is going on.
>
> It looks like even when KiCad exited and finished creating the WRL your system doesn't immediately have the [...]

So basically, it's a cache-invalidation bug :) or optimization or whatever. Hard to point it to anything specific though; filesystem, filesystem caching etc. But still odd for this to happen. Needs some defensive programming I suppose, but I've seen these kinds of bugs before too, where I had to add while sleep check|timeout loops to my code before because of some insane issues.

So what does your snippet do exactly, and how does it differ from the blender job I am currently using? I see nested outputs, blender & render outputs and a template. I'll read the template later and study the 'blender_export' job more precisely.

oliv3r, 2025-04-24:
This is truly perplexing!

I found that the code responsible for generating the bands is located here: https://github.com/INTI-CMNB/KiBot/blob/270c544eb96ae23b7d12a19996c0ce06919a29e6/kibot/out_base_3d.py#L327 (which I know is no surprise to you :)

After adding prints for the filenames:
```
WARNING:Creating file /tmp/.cache/kibot/3d/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_2_10.wrl (kibot - out_base_3d.py:328)
WARNING:Creating components by copying /workdir/hello/packages3D/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal.wrl (kibot - out_base_3d.py:331)
```

I see that the target is a 'cache' location, the source is my models location. So quite often, the cache is not being copied to the proper location (components). Very often as mentioned before. I'll dig further.

set-soft, 2025-04-25:
The PCB2Blender_2_7 templates only generates the PCB3D

oliv3r, 2025-04-28:
That's what's seems to be going wrong now, I always get a *_export.pcb3d file, but sometimes/often it does not include the generated models, they do not seem to get copied out of the ~/.cache dir. I've pin-pointed it to the vrml job, so I have something easily reproduceable/triggerable now and will dig deeper to see what I can find.

set-soft, 2025-04-30:
Hi @oliv3r !
I don't follow you, how this can explain the fact that it always work here and sometimes works in your setup?

set-soft, 2025-04-30:
I mean, if the generated models have errors, how can it be that sometimes it works and other times it doesn't?

oliv3r, 2025-05-01:
I agree and understand, amd once I have figured out this bug we'll know. I am leaning towards a racey isssue, as it is so random. And since it only happens for THT it seems, not a lot of people run into this I suppose.

I first have to figure out now if the generated models on working/non-working setups are identical though, to see if thats what makes kicad skip it. i have since built the kicad code so should be able to add debug prints.

oliv3r, 2025-05-08:
Sorry for not having been entirely clear/precise, but it's the 'kicad export-vrml not copying the files when told to do so, without raising an error'. Before kicad-cli i have the files in /tmp, after kicad-cli i don't have the files in components, because it didn't copy things.

Meanwhile, I have managed to compile kicad:master and can run kicad and export the vrml (without the copy) as mentioned earlier. I can add printfs too, so I will start debugging things now.

Once I know more, we'll know if it's a bug in the model (unlikely) or in kicad (more probable).

BTW, I've been doing all the latest tests locally, without gitlab involved at all. So that theory sadly doesn't hold :(

set-soft, 2025-05-08:
> Sorry for not having been entirely clear/precise, but it's the 'kicad export-vrml not copying the files when told to do so, without raising an error'. Before kicad-cli i have the files in /tmp, after kicad-cli i don't have the files in components, because it didn't copy things.

My doubt is about the model files, why is KiCad skipping some of them? Are they corrupted?
 
> Meanwhile, I have managed to compile kicad:master and can run kicad and export the vrml (without the copy) as mentioned earlier. I can add printfs too, so I will start debugging things now.
> 
> Once I know more, we'll know if it's a bug in the model (unlikely) or in kicad (more probable).

Ok
 
> BTW, I've been doing all the latest tests locally, without gitlab involved at all. So that theory sadly doesn't hold :(

Is KiBot downloading any model or you have all models available locally? My doubt is about the integrity of the downloaded files. This is why I added a SHA256 computation. Lamentably I can't reproduce the issue.

> So I found the 'problem', in that if the models are not in the '3d cache', they get ignored/not copied. I'll try to figure out why this is, before filing a bug report. https://gitlab.com/kicad/code/kicad/-/blob/master/pcbnew/exporters/exporter_vrml.cpp?ref_type=heads#L1052
> 
> we get 'nullptr' for mod3d, and thus the rest of the code is ignored. However it is only relevant in the 'writeVRML' OR non-inline case. Wonder if this is a recent kicad change ... and what's needed to get the generated model into the cache.

The line of code you point is quite old. You can see it using the "blame" option at the top.
The question is why it returns NULL. Is because it failed to load the file?

oliv3r, 2025-05-08:
Yeah, it was returning a `nullptr` indeed. The line above it was quite new (changed last week) and I think it has all to do with the new `kicad-embed://` feature.

I opened the bug report https://gitlab.com/kicad/code/kicad/-/issues/20877 (and a bugfix https://gitlab.com/kicad/code/kicad/-/merge_requests/2241).

So sad now is, I found out that I was using the wrong models locally here to test things. During my debugging sessions, I added a print to the generated models, it's just a comment, but the wrong one `# Header#VRML...` which caused at least the model to not be in the cache. Of course after fixing the model, it all worked locally again on one system, so now I'm trying it on my build-server to see if I can trigger it there, and continue debugging.

The bugfix does make sense imo, a nullptr on the cache shouldn't mean 'stop working'... Anyway, to be continued.

set-soft, 2025-05-08:
> Yeah, it was returning a `nullptr` indeed. The line above it was quite new (changed last week) and I think it has all to do with the new `kicad-embed://` feature.

The embedded stuff is new, but the use of cache is old.

> I opened the bug report https://gitlab.com/kicad/code/kicad/-/issues/20877 (and a bugfix https://gitlab.com/kicad/code/kicad/-/merge_requests/2241).

I can't reproduce any problem using your zip file and KiCad 9.0.1.
But when I tried 9.99.0 I consistently get a coredump at exit:

```
$ kicad-cli pcb export vrml -o 5/pcb.wrl -f --units m --models-relative --models-dir components --user-origin 0.0x0.0mm hello_kibot.kicad_pcb
Successfully exported VRML to 5/pcb.wrl
double free or corruption (!prev)
Aborted (core dumped)
```

So it might be a memory corruption bug, which might explain why is so random.

> 
> So sad now is, I found out that I was using the wrong models locally here to test things. During my debugging sessions, I added a print to the generated models, it's just a comment, but the wrong one `# Header#VRML...` which caused at least the model to not be in the cache. Of course after fixing the model, it all worked locally again on one system, so now I'm trying it on my build-server to see if I can trigger it there, and continue debugging. Sadly, manually running kicad-cli on the zip file I attached to the bug report doesn't cause the issue in the kibot container :(
> 
> The bugfix does make sense imo, a nullptr on the cache shouldn't mean 'stop working'... Anyway, to be continued.

Not sure, under normal use KiCad won't stop because a model is missing.

If you get a fail please capture the log files using at least `-vvv` this will provide more information.

set-soft, 2025-05-09:
Today I tried running a VRML test on my local machine using the docker images. 
I executed 50 runs, no error.
It looks like the hardware is an important detail to reproduce your issue.

oliv3r, 2025-05-09:
> Today I tried running a VRML test on my local machine using the docker images. I executed 50 runs, no error. It looks like the hardware is an important detail to reproduce your issue.

Yeah, I noticed that too. I think I complaint a bit earlier about that. At first I thought it was about SSE or not, Intel vs AMD, high-power vs low-power; lot of ram vs light on RAM, but I haven't found anything conclusive. Just that this one (virtual hosted) machine fails consistently all the time. I will split the jobs as I wanted to do before, just so that I can run the pipeline more often without having to wait for the whole blender render, as it's just the pcb2blender stuff.

it's just a bit annoying that nothing actually fails in this step, so a pipeline run would always be successful, and I really don't want to build some test step (unzip, double check if files are there) as my repo's purpose is not to find issues or bugs in kibot/kicad themselves of course.

oliv3r, 2025-05-10:
Just posting some results, though I'll also copy them to the kicad bug report. Didn't quite run the 10k :) but I think it's quite obvious that something is a miss. One last thing I'll try is to disable all but one P core, and see how that affects the results.

Anyway, I think this bug should remain open until kicad fixes this.

```
Using SCH file: hello.kicad_sch
- 'VRML model for PCB3D 2.7' (_PCB2Blender_vrml_2_7) [vrml]
WARNING:(W174) Missing component in generated VRML: `components/R_Axial_DIN0207_L6.3mm_D2.5mm_P10.16mm_Horizontal_1_0_1_10.wrl` (kibot - out_vrml.py:190)
ERROR:Warnings treated as errors (kibot - log.py:142)
failures=2743
Using SCH file: hello.kicad_sch
- 'VRML model for PCB3D 2.7' (_PCB2Blender_vrml_2_7) [vrml]
Found 0 unique warning/s (1 total, 1 filtered)
success=1642
```

set-soft, 2025-05-12:
Hi @oliv3r !

Yes, I added a warning when a model is missing.
But, as I mentioned in the KiCad bug report, you should focus on narrowing the issue. Then try things like a compiled KiCad.
For this:

1. Just use the command KiBot is using `kicad-cli pcb export ...` 
2. Remove some 3D models, first remove half of them, if it fails remove the other half, if not remove 1/4, etc.

Once you have a simple case we can try various things.

oliv3r, 2025-05-13:
> Hi [@oliv3r](https://github.com/oliv3r) !
>
> Yes, I added a warning when a model is missing. But, as I mentioned in the KiCad bug report, you should focus on narrowing the issue. Then try things like [...]

Yes, the problem with that was, it wasn't triggering anymore, so went one step back where it was triggering often. I'll 'zoom-in' again and try the same loop with just kicad-cli (after 1 run of kibot to ensure the generated models exist in `.cache`.

If it only happens on an invocation through kibot(-container) (with the exact kicad-cli arguments) then I'll have to come up with a different approach, like copying the custom compiled kicad binaries, but that's proving tougher then expected :) I'll have to look into your container generation scripts to see how you do kicad, but afaik you actually do just install it from a deb ...

set-soft, 2025-05-13:
Running a KiCad v9.99 compiled locally the VRML is generated correctly, by I consistently get a memory error at exit (when releasing the memory). So at least 9.99 is doing something wrong during the VRML export.

I enabled the address sanitizer with KICAD_SANITIZE_ADDRESS=ON and I'm getting:

```
$ kicad-cli pcb export vrml -o 5/pcb.wrl -f --units m --models-relative --models-dir components --user-origin 0.0x0.0mm hello_kibot.kicad_pcb
Successfully exported VRML to 5/pcb.wrl
=================================================================
==922==ERROR: AddressSanitizer: heap-use-after-free on address 0x60b0000e4c80 at pc 0x7f63e88b8681 bp 0x7fffad62c270 sp 0x7fffad62ba20
WRITE of size 104 at 0x60b0000e4c80 thread T0
    #0 0x7f63e88b8680 in __interceptor_memset ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:799
    #1 0x7f63e54371c1 in std::_Hashtable<char const*, std::pair<char const* const, int>, std::allocator<std::pair<char const* const, int> >, std::__detail::_Select1st, iequal_to, fnv_1a, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear() /usr/include/c++/12/bits/hashtable.h:2501
    #2 0x7f63e54371c1 in std::_Hashtable<char const*, std::pair<char const* const, int>, std::allocator<std::pair<char const* const, int> >, std::__detail::_Select1st, iequal_to, fnv_1a, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::~_Hashtable() /usr/include/c++/12/bits/hashtable.h:1584
    #3 0x7f63e54371c1 in std::unordered_map<char const*, int, fnv_1a, iequal_to, std::allocator<std::pair<char const* const, int> > >::~unordered_map() /usr/include/c++/12/bits/unordered_map.h:102
    #4 0x7f63e173aff6 in __cxa_finalize stdlib/cxa_finalize.c:83
    #5 0x7f63e4250e66  (/home/salvador/kicad/kicad-real/debian/build/common/libkicommon.so.9.99.0+0x650e66)

0x60b0000e4c80 is located 0 bytes inside of 104-byte region [0x60b0000e4c80,0x60b0000e4ce8)
freed by thread T0 here:
    #0 0x7f63e892b3c8 in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:164
    #1 0x7f63e173aff6 in __cxa_finalize stdlib/cxa_finalize.c:83

previously allocated by thread T0 here:
    #0 0x7f63e892a4c8 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:95
    #1 0x7f63c7ab0885  (/usr/lib/x86_64-linux-gnu/libgcc_s.so.1+0x2885)

SUMMARY: AddressSanitizer: heap-use-after-free ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:799 in __interceptor_memset
Shadow bytes around the buggy address:
  0x0c1680014940: fd fd fd fd fd fa fa fa fa fa fa fa fa fa fd fd
  0x0c1680014950: fd fd fd fd fd fd fd fd fd fd fd fa fa fa fa fa
  0x0c1680014960: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c1680014970: fd fa fa fa fa fa fa fa fa fa fd fd fd fd fd fd
  0x0c1680014980: fd fd fd fd fd fd fd fa fa fa fa fa fa fa fa fa
=>0x0c1680014990:[fd]fd fd fd fd fd fd fd fd fd fd fd fd fa fa fa
  0x0c16800149a0: fa fa fa fa fa fa fd fd fd fd fd fd fd fd fd fd
  0x0c16800149b0: fd fd fd fd fa fa fa fa fa fa fa fa fd fd fd fd
  0x0c16800149c0: fd fd fd fd fd fd fd fd fd fd fa fa fa fa fa fa
  0x0c16800149d0: fa fa fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c16800149e0: fa fa fa fa fa fa fa fa fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==922==ABORTING
```

oliv3r, 2025-05-14:
So you are suggesting it might be fixed in -dev. I'll try to compile things based on the 9.0.1 tag to see where I strand then ...

oliv3r, 2025-05-15:
So, I think I figured it out. And also can make sense of things.

I never got the issue when running kicad-cli, because I used the existing (e.g. previously kibot generated) files. In that case, the timestamps differ enough to not be an issue.

However, I had the issue with a kibot run, because we generate and run kicad-cli so fast in succession, we are within the same second. This wouldn't be a problem, except the debian container we use, doesn't have `wxHAS_HUGE_FILES` resulting in 32bit timestamps.

So that explains why it only happens in certain cases (if we generate things at 0.999 seconds, and kicad-cli runs at 1.000 seocnds) and certain systems (slow SSD's with slow filesystems) will not trigger this bug either, because of a 'natural' delay.

There's a little more to this story, but that'll be in the kicad bug report, because I think their implementation could be better to avoid this all together (even when not using `wxHAS_HUGE_FILES` as that's a host limitation).

P.S. I must say I'm suprised my CI wasn't having issues, because my CI host is a shared VM, so would expect it to be quite slow. But since I run 3 views, thus run the job 3 times, I have 3 times more chance of triggering the error.

After merging #817 we should be able to close this issue, or hope for KiCad 9.0.3, though that may not happen for a long time, and #817 probably should be backported to other release branches too ...

Thanks @set-soft for being pro-active and patient :D

oliv3r, 2025-05-15:
Your work-around, by copying the files if kicad failed to copy them also works. It's not nice, and hopefully kicad merges the patch upstream.

oliv3r, 2025-05-16:
@set-soft with https://gitlab.com/kicad/code/kicad/-/merge_requests/2245 being merged, the issue should resolve itself (probably not before kicad 10 :( so your workaround helps and is needed until then.

set-soft, 2025-05-16:
> [@set-soft](https://github.com/set-soft) with https://gitlab.com/kicad/code/kicad/-/merge_requests/2245 being merged, the issue should resolve itself (probably not before kicad 10 :( so your workaround helps and is needed until then.

It was merged and milestone is 9.0.3.

> Any ETA on 1.8.5? :)

Use `dev` images, I'm currently working on `kicad-cli`, to make it as functional as the Python API. 
So 1.8.5 is just maintenance patches without any critical one yet.

oliv3r, 2025-05-16:
> > [@set-soft](https://github.com/set-soft) with https://gitlab.com/kicad/code/kicad/-/merge_requests/2245 being merged, the issue should resolve itself (probably not before kicad 10 :( so your workaround helps and is needed until then.
> 
> It was merged and milestone is 9.0.3.
💌 
> 
> > Any ETA on 1.8.5? :)
> 
> Use `dev` images, I'm currently working on `kicad-cli`, to make it as functional as the Python API. So 1.8.5 is just maintenance patches without any critical one yet.

I am using the `-dev` image, but I can't offer the `-dev` image of course with kici to anybody else, as it's a moving target. I was just looking at history and saw a month between 1.8.2, 1.8.3 and 1.8.4 so was hoping 1.8.5 would be released this month some time.

But if not, I'll just use the current hash (again, `-dev` is a moving target.

commit bb8754ac (2025-05-14): [VRML][Added][KiCad 9] Woraround for missing 3D models

See https://gitlab.com/kicad/code/kicad/-/issues/20877
Closes #804