INTI-CMNB/KiBot #807: [BUG] kibot 1.8.4 fails while kibot 1.8.3 passes with relative path for pcb drawing sheet template

stefan894, 2025-04-16:
**Describe the bug**
set the drawing sheet path relative on a windows pc:

```
..\..\..\kicad_template\xxx_A3_pcb.kicad_wks
```

run kibot in a docker container via gitlab-runner

With kibot v1.8.3 it passes, with kibot 1.8.4 it fails.

**To Reproduce**
I can not share the repository.

.gitlab-ci.yml:
```
### Gitlab CI/CD example for KiCad

stages:
  - gen_variants

image:
  #name: ghcr.io/inti-cmnb/kicad8_auto_full:1.8.1 # ok
  #name: ghcr.io/inti-cmnb/kicad8_auto_full:1.8.2 # fails
  name: ghcr.io/inti-cmnb/kicad8_auto_full:1.8.3 # ok
  #name: ghcr.io/inti-cmnb/kicad_auto_full:dev_k8 # fails
  name: ghcr.io/inti-cmnb/kicad8_auto_full:1.8.4 # fails

var_default:
  stage: gen_variants
  tags:
    - bozzio
    - docker
  before_script:
    - git remote prune origin
  script:
    #- cd .. && rm -rf kicad_template && git clone git@gitlab.com:xxx/kicad_template.git
    #- cd Testing
    - kibot --log Generated/kibot.log -c AC_Pedal_Adapter/V2.0/Assembly_DNP.kibot.yaml -b AC_Pedal_Adapter/V2.0/AC_Pedal_Adapter_V2.0.kicad_pcb
  rules:
    - changes:
      - 'AC_Pedal_Adapter/V2.0/*.kicad_sch'
      - 'AC_Pedal_Adapter/V2.0/*.kicad_pro'
      - 'AC_Pedal_Adapter/V2.0/AC_Pedal_Adapter_V2.0.kicad_pcb'
      - 'AC_Pedal_Adapter/V2.0/*.kibot.yaml'
      - 'Makefile'
      - ".gitlab-ci.yml"
      - "versionbump.txt"
      when: on_success
  artifacts:
    when: always
    paths:
      - Generated/
```

Assembly_DNP.kibot.yaml:
```
kibot:
    version: 1

globals:
    dnp_cross_top_layer: 'User.2'
    dnp_cross_bottom_layer: 'User.3' 
    
outputs:     
  - name: print_pdf
    comment: "PDF for the PCB"
    type: pcb_print
    dir: Generated
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      dnf_filter: _kibom_dnf_Config
      scaling: 1.0
      pages:
        - layers:
            - layer: 'F.Silkscreen'
            - layer: 'F.Fab'
            - layer: 'User.2'
              color: '#ff0000'
          sheet: 'Assembly Top'
        - layers:
            - layer: 'B.Silkscreen'
            - layer: 'B.Fab'
            - layer: 'User.3'
              color: '#ff0000'            
          sheet: 'Assembly Bottom'
          mirror: true          
```

**Expected behavior**
With kibot V1.8.4 I get  this error and it fails_

```
ERROR:Missing page layout file: /builds/xxx_hw/testing/..\..\..\kicad_template\xxx_A3.kicad_wks (kibot.gs - gs.py:899)
```

while kibot v1.8.3 just passes.

**Screenshots**

Kibot 1.8.3:
```
Running with gitlab-runner 16.9.1 (782c6ecb)
  on asti-docker t3_4bAyJq, system ID: s_0ce977d89d45
Preparing the "docker" executor
00:03
Using Docker executor with image ghcr.io/inti-cmnb/kicad8_auto_full:1.8.3 ...
Pulling docker image ghcr.io/inti-cmnb/kicad8_auto_full:1.8.3 ...
Using docker image sha256:574e217413b1f26b69da9961b91449d5425594ea037c70fe541f08ef83399af9 for ghcr.io/inti-cmnb/kicad8_auto_full:1.8.3 with digest ghcr.io/inti-cmnb/kicad8_auto_full@sha256:885c03d0fca9906c22a74e00f7e83e740c5d6df9af8d8b05f35aa1343fb49643 ...
Preparing environment
00:00
Running on runner-t34bayjq-project-56284103-concurrent-0 via asti...
Getting source from Git repository
00:02
Fetching changes with git depth set to 20...
Reinitialized existing Git repository in /builds/xxx_hw/testing/.git/
Checking out 82cd0483 as detached HEAD (ref is main)...
Removing Generated/
Skipping Git submodules setup
Executing "step_script" stage of the job script
00:05
Using docker image sha256:574e217413b1f26b69da9961b91449d5425594ea037c70fe541f08ef83399af9 for ghcr.io/inti-cmnb/kicad8_auto_full:1.8.3 with digest ghcr.io/inti-cmnb/kicad8_auto_full@sha256:885c03d0fca9906c22a74e00f7e83e740c5d6df9af8d8b05f35aa1343fb49643 ...
$ kibot --log Generated/kibot.log -c AC_Pedal_Adapter/V2.0/Assembly_DNP.kibot.yaml -b AC_Pedal_Adapter/V2.0/AC_Pedal_Adapter_V2.0.kicad_pcb
- 'PDF for the PCB' (print_pdf) [pcb_print]
ERROR:Unable to expand `xxx_DATASHEETS` in `${xxx_DATASHEETS}\TI\OPAMP\tlv2371-q1.pdf` (kibot - config.py:130)
WARNING:(W013) `REF**` component in board, but not in schematic (kibot - kiplot.py:411)
WARNING:(W147) Not including component `REF**` in filters because it has a malformed reference @ PCB (168.15, 51.5) mm (kibot - kiplot.py:342)
Found 2 unique warning/s (5 total, 3 filtered)
Uploading artifacts for successful job
00:02
Uploading artifacts...
Generated/: found 3 matching artifact files and directories 
Uploading artifacts as "archive" to coordinator... 201 Created  id=9737933893 responseStatus=201 Created token=glcbt-66
Cleaning up project directory and file based variables
00:00
Job succeeded
```

Kibot 1.8.4:
```
Running with gitlab-runner 16.9.1 (782c6ecb)
  on asti-docker t3_4bAyJq, system ID: s_0ce977d89d45
Preparing the "docker" executor
00:02
Using Docker executor with image ghcr.io/inti-cmnb/kicad8_auto_full:1.8.4 ...
Pulling docker image ghcr.io/inti-cmnb/kicad8_auto_full:1.8.4 ...
Using docker image sha256:89a5eee99a529c71f64a95b73aff828debc86fab22b69d7e83c0c6000146569a for ghcr.io/inti-cmnb/kicad8_auto_full:1.8.4 with digest ghcr.io/inti-cmnb/kicad8_auto_full@sha256:89420c8b826bc391c93f02da3dd2d2f21a5c1f68de26109cf4ae62c5977a2cde ...
Preparing environment
00:00
Running on runner-t34bayjq-project-56284103-concurrent-0 via asti...
Getting source from Git repository
00:02
Fetching changes with git depth set to 20...
Reinitialized existing Git repository in /builds/xxx_hw/testing/.git/
Checking out eafc6ec3 as detached HEAD (ref is main)...
Removing Generated/
Skipping Git submodules setup
Executing "step_script" stage of the job script
00:04
Using docker image sha256:89a5eee99a529c71f64a95b73aff828debc86fab22b69d7e83c0c6000146569a for ghcr.io/inti-cmnb/kicad8_auto_full:1.8.4 with digest ghcr.io/inti-cmnb/kicad8_auto_full@sha256:89420c8b826bc391c93f02da3dd2d2f21a5c1f68de26109cf4ae62c5977a2cde ...
$ git remote prune origin
$ kibot --log Generated/kibot.log -c AC_Pedal_Adapter/V2.0/Assembly_DNP.kibot.yaml -b AC_Pedal_Adapter/V2.0/AC_Pedal_Adapter_V2.0.kicad_pcb
- 'PDF for the PCB' (print_pdf) [pcb_print]
ERROR:Unable to expand `xxx_DATASHEETS` in `${xxx_DATASHEETS}\TI\OPAMP\tlv2371-q1.pdf` (kibot - config.py:130)
WARNING:(W013) `REF**` component in board, but not in schematic (kibot - kiplot.py:416)
WARNING:(W147) Not including component `REF**` in filters because it has a malformed reference @ PCB (168.15, 51.5) mm (kibot - kiplot.py:347)
ERROR:Missing page layout file: /builds/xxx_hw/testing/..\..\..\kicad_template\xxx_A3.kicad_wks (kibot.gs - gs.py:899)
Uploading artifacts for failed job
00:02
Uploading artifacts...
Generated/: found 2 matching artifact files and directories 
Uploading artifacts as "archive" to coordinator... 201 Created  id=9737952280 responseStatus=201 Created token=glcbt-66
Cleaning up project directory and file based variables
00:00
ERROR: Job failed: exit code 1
```

**Environment (please complete the following information):**
- In a CI/CD workflow on `GitLab`
- Using a docker image:  ghcr.io/inti-cmnb/kicad8_auto_full:1.8.3 and ghcr.io/inti-cmnb/kicad8_auto_full:1.8.4

stefan894, 2025-04-16:
I have just seen in the log of kibot 1.8.3:

```
DEBUG:Using KICAD8_TEMPLATE_DIR="/usr/share/kicad/template" (guessed) (kibot - config.py:433)
```

stefan894, 2025-04-16:
I an using locally on a windows 10 kicad 8.0.9 while kibot 1.8.3 seems to use kicad 8.0.8+1 (not sure if that's the same) as 8.0.9

stefan894, 2025-04-16:
I have another project which was running fine on the 17.02.2025 with:

```
Using docker image sha256:47fa7607cfeaf5d2a10ff19f8ce06100dd8439800f90323cbe40ec363e67223a for ghcr.io/inti-cmnb/kicad_auto_full:dev_k8 with digest ghcr.io/inti-cmnb/kicad_auto_full@sha256:1f010bad62aaf8097fc272dd70ae6ce1f1d66213deb2b3ef03a6eb2386631c9b ...
```

and this uses the custom drawing sheet template.

now with :

```
Using docker image sha256:0e41353901b1d5315806f2cc959899cd307ddef391b31f9afc32eabeef980323 for ghcr.io/inti-cmnb/kicad_auto_full:dev_k8 with digest ghcr.io/inti-cmnb/kicad_auto_full@sha256:230168762e7b44c370b14f5e7f350aaaff6b0b98f387c828e67551efbca49fe0 ...
```

it fails as well..

So I imagine something changes between this dev_k8 release and v1.8.3/1.8.4

set-soft, 2025-04-16:
Hi @stefan894 !

I think what changed is that 1.8.4 is stopping instead of silently going on after the error was found. Which I think is the correct behavior (stop). After all you didn't realize your worksheet wasn't working until you migrated to 1.8.4.

Now about the problem, without a repo I can just see things from the log, and it says:

```
Fetching changes with git depth set to 20...
Reinitialized existing Git repository in /builds/xxx_hw/testing/.git/
```

So I assume the root of your repo is stored in `/builds/xxx_hw/testing/`, now you get an error about `/builds/xxx_hw/testing/..\..\..\kicad_template\xxx_A3.kicad_wks`. This is clearly outside your repo. Even worst, this isn't a valid file name because `\` is an escape character on Unix, not a directory separator. But KiBot is smart enough to solve such an error. Assuming it was a valid file name this will resolve to `/kicad_template/xxx_A3.kicad_wks` which is not in your CI/CD environment.

Then you mention something about `KICAD8_TEMPLATE_DIR` which makes me think you have `xxx_A3.kicad_wks` installed in the KiCad templates directory. In this case there is no way I can be accessed from the CI/CD.

The solution is quite simple: copy the worksheet to your project, use a relative path to the file contained in your project.

BTW: try using `/` instead of `\`

stefan894, 2025-04-16:
thanks for your feedback

> Which I think is the correct behavior (stop)

agree

>After all you didn't realize your worksheet wasn't working until you migrated to 1.8.4.

it's working sometimes

>This is clearly outside your repo.
> The solution is quite simple: copy the worksheet to your project, use a relative path to the file contained in your project.

yep, I think I have to change this, to have a reliable setup

>BTW: try using / instead of \

I already tried, kicad changes it back..

set-soft, 2025-04-16:
> > This is clearly outside your repo.
> > The solution is quite simple: copy the worksheet to your project, use a relative path to the file contained in your project.
> 
> yep, I think I have to change this, to have a reliable setup

If you are using the same worksheet in various project I'll suggest using a git submodule. Is annoying but will allow you to propagate fixes in the worksheet to all projects with less effort. 

> > BTW: try using / instead of \
> 
> I already tried, kicad changes it back..

Ugh! this looks like a bug to me. The OS itself supports both `\` and `/`, is just the GUI that uses `\`.

closed 2025-04-16 by set-soft

stefan894, 2025-04-16:
> suggest using a git submodule

yep, that's what I had in mind.

>Ugh! this looks like a bug to me

possible, but it's kicad 8.0.x so it goes into the category of "won't fix, pleas try with kicad 9.0.x" which I am not yet interested, I prefere giving others time to do some bug reporting ;-)